letscontrolit/ESPEasy #2608: MPR121 Plugin - various sensitivities

gitgrube, 2019-09-11:
Hello!
When I use the MPR121 module on the Arduino (with MPR121test from the sample folder), with a custom library with higher sensitivity (to capture something through plastic), everything works as desired. However, once I run the whole thing on the ESP with the same library then a touch can no longer be detected. Unless I touch the electrode directly.
Is there a certain difference between the example sketch and the plugin? Unfortunately, I can't get any further with my knowledge at this point.

TD-er, 2019-09-11:
Can you give direct links to whatever sketch you tried?

gitgrube, 2019-09-11:
example sketch: [https://github.com/adafruit/Adafruit_MPR121/blob/master/examples/MPR121test/MPR121test.ino](url)

Adapted values of the library (Adafruit_MPR121.cpp) (as described here: [https://forums.adafruit.com/viewtopic.php?f=19&t=72025&start=15#wrap](url) (fourth answer on the second page)

-before:  `writeRegister(MPR121_CONFIG1, 0x10); // default, 16uA charge current` -> after: `writeRegister(MPR121_CONFIG1, 0x3F);`
-before: `writeRegister(MPR121_CONFIG2, 0x20); // 0.5uS encoding, 1ms period` -> after: `writeRegister(MPR121_CONFIG2, 0x3A);`

Domosapiens, 2019-09-11:
@gitgrube 

> Unless I touch the electrode directly.

Arduino ..... 5V
ESP8266 .....3.3V
Could that make the sensitivity difference?

gitgrube, 2019-09-11:
> @Domosapiens 
> Arduino ..... 5V
> ESP8266 .....3.3V
> Could that make the sensitivity difference?

Since the MPR121 chip works with 3.3V, this should not be the reason. Nevertheless I operated the breakout board with 5V as well as with 3.3V from the Arduino, each with the same result. 

By the way: I use this module: [MPR121 Breakout](https://learn.adafruit.com/adafruit-mpr121-12-key-capacitive-touch-sensor-breakout-tutorial)