letscontrolit/ESPEasy #2609: deepsleep,0 is not working

treki1, 2019-09-11:
Installed is Git Build: mega-20190803
In the rules deepsleep,0 makes nothing (no deepsleep).
Values above 1 and higher work.
Is this a bug?

Budman1758, 2019-09-11:
Are you trying to make it sleep for zero seconds?

treki1, 2019-09-11:
See https://www.letscontrolit.com/wiki/index.php?title=SleepMode

- Wake up manually
- If you specify a Sleep Delay of 0, you will have to wake the device yourself by pulling RST to GND.

As a rule, I do not want to enable the deepsleep-timer.

TD-er, 2019-09-11:
The 0 as parameter should indeed (according to the wiki at least) be considered as no automatic wake.
Just like you're not connecting the RST and GPIO-16 and go to deep sleep.

I can imagine some changes in the core lib may have changed the behavior of this special value for deep sleep.

Micha-he, 2020-03-24:
'deepsleep,0' still doesn't work. Tested with source from yesterday.

Deepsleep with values of 1 or greater, need too much energy for longlife-battery-powered devices.
(~13mA after sleeptime instead of 14µA while)
Discussion in the forum: [https://letscontrolit.com/forum/viewtopic.php?p=42065#p42065](url)

TD-er, 2020-03-24:
See also [this post](https://www.letscontrolit.com/forum/posting.php?mode=reply&f=6&t=7389#pr42181) of differences between deep sleep from command and regular deep sleep

Micha-he, 2020-03-24:
#2966

uzi18, 2020-03-24:
did you measure current before and after your modification?

Micha-he, 2020-03-24:
Yes. Current is 14µA, shortly after the 'deepsleep,0' Commando and that over hours.

This commit didn't change the current, it only enable a value 0 for the deepsleep-command. I need it to sleep until the ESP is reseted external.

uzi18, 2020-03-25:
@TD-er we can add this small change but this should be definitely documented
so @Micha-he please add this also into documentation please

Micha-he, 2020-03-25:
Can you give me a hint, where I modify the documentation? There are some ESPEasy-docs...

uzi18, 2020-03-25:
https://github.com/letscontrolit/ESPEasy/tree/mega/docs/source

Micha-he, 2020-03-25:
I need some time to find, that a 'command' is a plugin ;-)
Now it's done...

tonhuisman, 2023-09-02:
This seems to be solved, can be closed.

closed 2023-09-02 by TD-er