letscontrolit/ESPEasy #2610: Make B/G default

jimmys01, 2019-09-12:
Despite what I have previously said, I can see that B/G does have a slight better connectivity on my nodes that are in the enge of my network. Since 3-4 months now I have all my nodes on B/G and I believe that this setting can be made to be the default and not an experimental option.

uzi18, 2019-09-12:
@clumsy-stefan @TD-er what do you think?
@jimmys01 maybe just option to set this mode automatically if rssi is bellow some point?

clumsy-stefan, 2019-09-12:
I have all my nodes fix on B/G since long, after all the trouble with the rekeeying in N mode... Since then I have no unreachable nodes anymore...
However I did not try to go back to N mode since there have been changes in the WiFi core...
Also I think in B/G mode the RSSI is slightly better..
IMHO

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er