letscontrolit/ESPEasy #2615: Use of IR RX in rules

jimmys01, 2019-09-13:
The RX plugin when it receives a signal it converts it into a decimal and sents it to the rules engine.
Now that we can have the more meaningfull "string" representation of the IR signal there is a problem. How to keep the rules working for IR TX.

One solution is to drop the RX rules functionality
Another is to "split" (@TD-er is this possible?) the data path: one represesntation of the signal goes to rules as a decimal value as always has been and another goes to the controllers as Sting value
Another is to support string values in the rules.

Any thoughts?

TD-er, 2019-09-13:
Well support of string values in rules would be great. But I'm afraid it may take quite some big changes to make it work.
If it is something which can be converted without much trouble into a float/int (like "1234") then it is very simple to do.
Have to look for it.

Also the addition of string support is in a very early stage, since I needed it for the serial proxy plugin.

tonhuisman, 2022-04-18:
This is probably solved by #3653

TD-er, 2022-04-18:
If that didn't solve it, I think #4004 may also allow for what is mentioned here.

closed 2022-04-18 by TD-er