letscontrolit/ESPEasy #2616: ESP Easy and Senseair S8 abnormal deep sleep issue

sobolkz, 2019-09-14:
- Build mega-20190830 test core 260 sdk3 16M at Wemos D1 mini Pro
- Discussed at the forum https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=7019
- EspEasy with Senseair S8 active plugin abnormally going to deep sleep after about 16 sec from wake up
- No any problems with power quality (separate powering was checked)
- Different PINS for SWserial and HWserial (with disabled serial log) already checked - same issue
- Board can not be fully loaded if S8 was removed from scheme
- Board can be loaded only after factory reset
- Another one board also checked - same issue (hardware independed)
- Plantower PMS7003 was checked with same PINs (no issues with SWSerial or HWserial plugins)

TD-er, 2019-09-14:
Just to make sure, the plugin does work fine when deep sleep is not used?

sobolkz, 2019-09-14:
Yes. Plugin works good until deep sleep was activated. Also (may be it isn't applicable for one time cycle) board can be loaded successfully from manual command "deepsleep,60".

TD-er, 2019-09-14:
I've taken a look at the code and at least there can be made some improvements in handling timeout and no data.
There is a timeout in the ModbusRTU code, but it just considers it as an invalid packet.
Reading the device description alone already takes about 6 readings.
So that does take more than a second when no device is connected.
Meaning we should find a way to send a command the device must act on to detect its presence.
For now I have not found anything in the Modbus protocol description which **must** be implemented in all devices.
So that leaves a device specific command (and thus not in the modbus code) to check the presence and act on it.

I have not yet seen code in the SenseAir plugin which may explain why the ESP keeps rebooting when no device is detected or not responding in due time.
But I will look further.

@uzi18 I will make a PR to add the "timeout" and "no data" state to the ModbusRTU code.

sobolkz, 2019-09-28:
It seems, that issue was really fixed in mega-20190926, as reported:
[Modbus RTU] Add "timeout" and "no data" as error
**### Checked with connected Senseair S8:**
INIT : Booting version: mega-20190928 (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-d                                                                                                                                                             ev(38a443e), LWIP: 2.1.2 PUYA support)
82 : INIT : Free RAM:30344
83 : INIT : Warm boot #1 Last Task: Background Task - Restart Reason: External S                                                                                                                                                             ystem
85 : FS   : Mounting...
110 : FS   : Mount successful, used 76053 bytes of 957314
578 : CRC  : program checksum       ...OK
589 : CRC  : SecuritySettings CRC   ...OK
693 : INIT : Free RAM:27184
694 : INIT : I2C
695 : INIT : SPI not enabled
3411 : S/N: 72b56eb - Type: 10e -
8034 : INFO : Plugins: 79 [Normal] [Testing] [Development] (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
8035 : EVENT: System#Wake
8208 : WIFI : Connecting access_point8 attempt #0
8224 : EVENT: System#Boot
9301 : BMx280 : Detected BME280
9891 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20448 WiFiStatus 6
(88:DC:96:26:7F:0A) Ch: 9 Duration: 3783 ms
12163 : EVENT: WiFi#ChangedAccesspoint
12171 : EVENT: WiFi#ChangedWiFichannel
12176 : WIFI : DHCP IP: 192.168.3.3 (ESP-Easy-0) GW: 192.168.3.1 SN: 255.255.255.0   duration: 18 ms
12220 : NTP  : NTP replied: delay 40 mSec Accuracy increased by 0.458 seconds
12221 : Time adjusted by -1569670925280.80 msec. Wander: -436019701.47 msec/second
12222 : Current Time Zone: STD time start: 2019-10-27 03:00:00 offset: 360 min
12224 : EVENT: Time#Initialized
12234 : EVENT: WiFi#Connected
12245 : Webserver: start
12246 : firstLoopConnectionsEstablished
12252 : EVENT: Clock#Time=Sat,17:42
13017 : BME280: dew point 10.12C
13034 : BME280 : Address: 0x76
13035 : BME280 : Temperature: 26.65
13035 : BME280 : Humidity: 35.46
13035 : BME280 : Barometric Pressure: 1006.93
13036 : EVENT: BME280#Temperature=26.65
13070 : EVENT: BME280#Humidity=35.46
13076 : EVENT: BME280#Pressure=1006.93
13083 : ACT  : TaskValueSet,12,1,1007*0.75
13094 : Command: taskvalueset
13161 : ACT  : Publish ESP_Easy/P_Hg/P_Hg,
13171 : Command: publish
15767 : Senseair: co2 = 0
15768 : EVENT: S8#CO2=0.00
pm open,type:2 0
                22982 : ADC  : Analog value: 88 = 88.000
22986 : EVENT: VBat#Vbat=0.36
23644 : Dummy: value 1: 755.25
23645 : EVENT: P_Hg#P_Hg=755.25
**### Connected but disabled:**
INIT : Booting version: mega-20190928 (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
83 : INIT : Free RAM:30344
84 : INIT : Rebooted from deepsleep #19 Last Task: Const Interval timer, id: 1 - Restart Reason: Deep-Sleep Wake
86 : FS   : Mounting...
110 : FS   : Mount successful, used 76053 bytes of 957314
175 : FS   : Success garbage collection
641 : CRC  : program checksum       ...OK
655 : CRC  : SecuritySettings CRC   ...OK
758 : INIT : Free RAM:27072
760 : INIT : I2C
760 : INIT : SPI not enabled
761 : EVENT: System#NoSleep=60
913 : INFO : Plugins: 79 [Normal] [Testing] [Development] (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
915 : EVENT: System#Wake
1090 : WIFI : Connecting access_point8 attempt #0
1105 : EVENT: System#Boot
1478 : ADC  : Analog value: 0 = 0.000
1482 : EVENT: VBat#Vbat=0.00
[... 54 lines cut ...]
INIT : Booting version: mega-20190928 (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
83 : INIT : Free RAM:30344
84 : INIT : Rebooted from deepsleep #25 Last Task: Background Task - Restart Reason: Deep-Sleep Wake
86 : FS   : Mounting...
111 : FS   : Mount successful, used 76053 bytes of 957314
169 : FS   : Success garbage collection
234 : FS   : Success garbage collection
297 : FS   : Success garbage collection
756 : CRC  : program checksum       ...OK
767 : CRC  : SecuritySettings CRC   ...OK
870 : INIT : Free RAM:27016
872 : INIT : I2C
872 : INIT : SPI not enabled
873 : EVENT: System#NoSleep=60
3142 : S/N: 0 - Type: 0 -
5520 : INFO : Plugins: 79 [Normal] [Testing] [Development] (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
5522 : EVENT: System#Wake
5694 : WIFI : Connecting access_point8 attempt #0
5709 : EVENT: System#Boot
6274 : Senseair:
6276 : EVENT: S8#CO2=3766.00
6557 : ADC  : Analog value: 0 = 0.000
6560 : EVENT: VBat#Vbat=0.00
7440 : Dummy: value 1: 755.25
Thank's to TD-er for fastest fix.

closed 2019-09-28 by sobolkz

TD-er, 2019-09-28:
> Thank's to TD-er for fastest fix.

You're welcome