letscontrolit/ESPEasy #2619: Settings archive: timeout updating config.dat

chemmex, 2019-09-17:
Getting acquainted with settings archive. Small rules1.txt (48b) downloaded properly, while config.dat stuck at timeout.
Tried on precompiled ESP_Easy_mega-20190903_hard_core_252_SONOFF_POW_4M1M.bin

192.168.110.61 - - [17/Sep/2019:12:38:17 +0000] "GET /rules1.txt HTTP/1.1" 200 48 "-" "ESP8266HTTPClient"                                                
192.168.110.61 - - [17/Sep/2019:12:41:33 +0000] "GET /config.dat HTTP/1.1" 200 65536 "-" "ESP8266HTTPClient"     

[image]
[image]

TD-er, 2019-09-17:
How are both ends (the one you're downloading to and 192.168.110.71) connected to the network?

Do both nodes react quickly on loading of the web page served on both? (assuming you are trying to clone the settings from another node, not downloading from some webserver)
Have you tried without "delete first" checked?
It could be the config.dat file is not actually deleted so the file open for write call can be causing the timeout?

chemmex, 2019-09-17:
The environment is close to ideal, everything is very fast. Webserver is connected directly to a WiFi router.

Actually, I am trying to copy the settings from the webserver, not from another node,  I assume this is valid scenario.

Just tried fresh config without "delete first" checked,  same story:

220020: downloadFile: URL: http://192.168.110.71/config.dat decoded: 192.168.110.71:80/config.dat
222052: Timeout

TD-er, 2019-09-17:
How does the webserver serving the files? With chunked transfer? I am not sure that is supported.

Can you look at the headers the webserver sends when serving the file? (from your browser inspect window)

chemmex, 2019-09-18:
You are right, apache sends the file as chunked, but that's what the ESP requests:

Frame 4: 205 bytes on wire (1640 bits), 205 bytes captured (1640 bits)
Ethernet II, Src: 80:7d:3a:6e:9f:93 (80:7d:3a:6e:9f:93), Dst: 02:42:38:0b:b8:86 (02:42:38:0b:b8:86)
Internet Protocol Version 4, Src: 192.168.110.61, Dst: 192.168.110.71
    GET /config.dat HTTP/1.1\r\n
    Host: 192.168.110.71\r\n
    User-Agent: ESP8266HTTPClient\r\n
    Connection: close\r\n
    Accept-Encoding: identity;**q=1,chunked**;q=0.1,*;q=0\r\n
    \r\n

Will try to explicitly disable chunked transfer and see what happens

TD-er, 2019-09-18:
OK, good to know "we" also allow it as an option.
Maybe because we do send using chunked transfers too?

TD-er, 2019-09-18:
Can you try the PR made by @uzi18  in #2623 ?

chemmex, 2019-09-19:
Tried. The client changes from HTTP 1.1 to 1.0, but it seems it's not the issue, the transfer still ends with timeout. 

In a tcpdump log I can see that ESP requests config.dat from the server, apache responds with 1.0 header and starts thansfer. Then, after receiving about 33K of the file, the ESP sends several RST packets and transfer finishes.

uzi18, 2019-09-19:
so it is a part of problem, is it possible for you to get debug log durring download?

chemmex, 2019-09-19:
that's all it says:

3306553: downloadFile: URL: http://192.168.110.72/config.dat decoded: 192.168.110.72:80/config.dat
3308585: Timeout

uzi18, 2019-09-19:
filesystem/spiffs informations (free space,etc.) ?

chemmex, 2019-09-19:
Storage
--
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 28 daily / 28 boot
Sketch Size: | 876 kB (2192 kB free)
SPIFFS Size: | 934 kB (876 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 116
Maximum open files: | 5
Maximum path length: | 32

Free Mem: | 19440 (14776 - sendContentBlocking)
Free Stack: | 3616 (1344 - sendContentBlocking)
Heap Max Free Block: | 13536
Heap Fragmentation: | 29%
Boot: | Manual reboot (2)
Reset Reason: | External System
Last Task: | Const Interval timer, id: 1
SW WD count: | 0

chemmex, 2019-09-19:
Filesystem:

[image]

uzi18, 2019-09-19:
ok try also with this change: https://github.com/letscontrolit/ESPEasy/pull/2623/commits/033ddd057c94ef58af57e11f6fa1fcfba7569429

chemmex, 2019-09-19:
Now it's ok, and the filesystem looks like this:

[image]

TD-er, 2019-09-19:
Hmm, I've seen other issues with respect to appending to a file on SPIFFS.
That's also the reason the "cache controller" is only using files upto 24k in size.
That appeared to be some kind of magical size limit (3 blocks).
When crossing that size, the time needed to find a new block to write to was becoming a real issue.

SPIFFS is rather tricky with its administration.
Maybe we should pre-allocate the size of the file and then not use append, but just fill it.

uzi18, 2019-09-19:
@TD-er maybe sometimes writes takes more time, my idea was to refresh timeout if stream is received even slowly, this explanation for last commit

chemmex, 2019-09-19:
reverted back to HTTP 1.1 transfer - it works too

uzi18, 2019-09-19:
ok so will remove set to 1.0 from source in #2623

uzi18, 2019-09-19:
@TD-er ready to merge ;)

closed 2019-09-29 by chemmex