letscontrolit/ESPEasy #2622: Add INA226

ronnythomas, 2019-09-18:
The INA219 is already include as Energy (DC) and I'm happy with it but this sensor is limited for some use.
So is it possible to add the INA226 ? with Shunt 2mOhm (10A Load)

The library can be found here : https://github.com/SV-Zanshin/INA226

Thx

TD-er, 2019-09-18:
I guess it could perhaps be combined with this issue: https://github.com/letscontrolit/ESPEasy/issues/2602 ?

RedDev85, 2020-04-07:
has the ina226 already been immigrated?  otherwise I would join the request.

TD-er, 2020-04-07:
Nope not yet. (I don't even have one here to test)

Oleg-Sob, 2021-11-15:
I subscribe to the request. I would very much like the INA 226 to be added to the Energy (DC) build. This sensor works better. Help with the solution of the issue.

tonhuisman, 2021-11-15:
The INA library has been renamed/moved to https://github.com/Zanduino/INA
We should probably look at switching to this library, so we have support for all INA chips supported there.

Oleg-Sob, 2021-11-15:
I am not an expert in programming, if the developers included these libraries in the Energy (DC) assembly, that would be very good.

Oleg-Sob, 2021-11-17:
Tell me, is it possible to add these libraries https://github.com/Zanduino/INA to the Tasmota firmware? Or where else to turn?

Oleg-Sob, 2021-11-17:
I really need firmware for INA 226.

TD-er, 2021-11-17:
> Tell me, is it possible to add these libraries https://github.com/Zanduino/INA to the Tasmota firmware? Or where else to turn?

I have no idea how the Tasmota firmware does handle libraries.

Oleg-Sob, 2021-11-17:
I was wrong not in Tasmota but in EspEasy.

Oleg-Sob, 2021-11-17:
> > Tell me, is it possible to add these libraries https://github.com/Zanduino/INA to the Tasmota firmware? Or where else to turn?
> 
> I have no idea how the Tasmota firmware does handle libraries.

I was wrong not in Tasmota but in EspEasy.

TD-er, 2021-11-17:
> I subscribe to the request. I would very much like the INA 226 to be added to the Energy (DC) build. This sensor works better. Help with the solution of the issue.

What is the main difference between the INA 226 and the one that is supported?
It doesn't add more inputs, so in that sense it doesn't look like a lot has to be changed to use the library linked by Ton.
Still it is very well possible some function names may be different, so I don't think it is just a matter of replacing or adding a library to make it work.

I will have a look in my drawer with sensors to implement (which is a large drawer with lots of sensors) to see if I already have it.
Can't promise it, but if I have it here, I will try to free up some time this week to see what has to be done to implement it.

Oleg-Sob, 2021-11-17:
The current measures correctly, the voltage does not measure correctly, it is greatly underestimated.

Oleg-Sob, 2021-11-17:
I connected the INA 226 to EspEasy using the library provided by the INA 219
[image]
.

TD-er, 2021-11-17:
OK, so we're now probably reading the values from the wrong registers, and/or apply the wrong formula for calculating the values.
That's clear. 

But what I asked was, what makes the INA226 so much better than the INA216?

Oleg-Sob, 2021-11-17:
INA 226 measures voltages up to 36V and connects to the negative bus, INA 219 measures voltages up to 26V and connects to the positive bus. And yet, maybe it's just me in 219 the measurements jump strongly, in 226 it measures stably.

TD-er, 2021-11-17:
Interesting.
Just take a look at both reference applications.
[image]

[image]

The INA226 also allows to measure the shunt on the high side.
But given that it does have the shunt on the low side, it may use a higher gain (or lower attenuation) to measure both Vin voltages, thus you're then using a larger part of the full range of the ADC to measure the voltage drop over the shunt.

But that doesn't explain why your voltage readings are so off (other than my assumption that we're reading the wrong register or setting some gain at an incorrect value)
One of the things that should improve between low side shunt and high side shunt is that the current measurement should be better.

Anyway, we should have a look at the linked library and/or dive into both datasheets to be sure what it is you're seeing here.

Oleg-Sob, 2021-11-17:
I have already tried several times to connect INA 219, the voltage measures normally, but the current readings jump a lot, for example, if there is a constant consumption of 1A, it measures 5A once, the second time is -3A, in INA 226 it shows stably 1A. I don’t know why, maybe the high frequency inverter is interfering. But the voltage in INA 226 does not show correctly, there is no linearity, for example, if the voltage shows 20v and in fact 20v, if in fact 18v INA 226 shows 17v, and if in fact 22v INA 226 shows 23v. There is no linearity. And the readings of INA 226 are not correct, instead of 20v it shows 10v, I already multiply this by 2 with the formula.

TD-er, 2021-11-17:
If the current measurement is varying a lot, while the current drawn from the device doesn't, then you can add a small capacitor over the shunt, like this:

[image]

Oleg-Sob, 2021-11-17:
I tried experiments with capacitors and filters, it did not work out to achieve stability, I think this is due to the positive bus. For this I put INA 226, but I am not satisfied with the nonlinearity of the voltage in 226.

Oleg-Sob, 2021-11-18:
If it helps, here is the firmware https://home-garden.org/forum/index.php?action=dlattach;topic=42.0;attach=98. The INA 226 sensor itself is working correctly, but something is wrong with the firmware, the rules do not work and the commands from the command line do not pass. Link to page https://home-garden.org/forum/index.php?topic=42.0. Can you fix this and integrate it into the ESPEasy firmware itself?

TD-er, 2021-11-18:
Apparently someone already had made it working for the INA226, but I don't see the code for it. (I hope Google translate didn't mess up the post you linked)

Oleg-Sob, 2021-11-18:
The first link is a BIN file, can you download it?

Oleg-Sob, 2021-11-18:
I post it here 

[ESPEasy.ina226.bin.zip](https://github.com/letscontrolit/ESPEasy/files/7561900/ESPEasy.ina226.bin.zip)

knrdl, 2022-12-13:
Hey there, I want to follow up this issue.  I don't know the firmware provided by @Oleg-Sob. 
But there is another source: https://github.com/wollewald/INA226_WE
Examples and explanation can be found in the repo. Would it be possible to include this library in order to provide support for INA226?
Thanks for your effort!