letscontrolit/ESPEasy #2626: Dummy Device error

AxelMilb, 2019-09-25:
It's about the ESP_Easy_mega-20190925_test_ESP8266_4M_VCC.bin
Generic - Dummy Device is now korekt adjustable Singel, Dual, Driple, Quad but if I need two Dummy Device it is only possible to use all in the last setting (Example Driple)
So it is not possible to use 1x Dual Dummy Device and 1x Quad Dummy Device at the same time.

Build Time: ⋄ Sep 25 2019 02:14:14
Binary Filename: ⋄ ESP_Easy_mega-20190925_normal_core_242_ESP8266_1M.bin
Flash Chip ID: Vendor: 0x85 (PUYA, supported) Device: 0x6014
Flash Chip Real Size: 1024 kB
Flash IDE Size: 1024 kB
Flash IDE Speed: 40 MHz
Flash IDE Mode: QIO
Flash Writes: 95 daily / 95 boot
Sketch Size: 820 kB (52 kB free)
SPIFFS Size: 110 kB (26 kB free)

uzi18, 2019-09-25:
We know about this limit

AxelMilb, 2019-09-25:
> We know about this limitation

Not helpful

uzi18, 2019-09-25:
We have some design limitation.
I see some kind of solution, like dummy plugin per type, but it will be not backward compatible.
@TD-er ideas?

TD-er, 2019-09-25:
> So it is not possible to use 1x Dual Dummy Device and 1x Quad Dummy Device at the same time.

Hmm that's a strange limitation indeed.
It is a side-effect I did not realize when implementing it.

Not sure yet how this can be changed without changing a lot of code.
I'll try to explain as simple as possible what's going on here.

We have:
- plugins (the code to support some kind of sensor)
- Tasks (the list shown on the Devices tab in the web interface)

The number of values a task may output is defined by the plugin.
For example the BME280 has 3 values (temp/hum/pressure).
I made it possible to select the number of outputs on a few plugins, like Dummy and Systeminfo.
What this selection does, is temporary change the number of outputs of the plugin, not the task.

If we want to change that, then every single call in the code checking for the number of output values should be changed to some call which may check if this value has a local override in the task.
Simply stated, we should ask the task the number of outputs and not the plugin.

I think this will involve more than half the number of files we have in our project, not including the plugin files.
That's a major redesign of the code and not something that we can do very easily.

uzi18, 2019-09-25:
I think about simply add extra plugins with datatype in name like:
Dummy - Single
Dummy - Dual
Dummy - Triple
Dummy - Quad

AxelMilb, 2019-09-25:
Thanks for your support

TD-er, 2019-09-25:
That's going to be a long list then, since we also have types as LONG, etc.
Also what to do with other plugins allowing to change the output values?
The SenseAir plugin is not very likely to have multiple instances of the same plugin. (although it is modbus, so it is possible)
I don't think having multiple of these is the way to go, since it will get messy very soon.

uzi18, 2019-09-25:
is it possible to add vtype and vcount to task settings?

TD-er, 2019-09-25:
As long as all fit in 32 bit, a lot is possible.
But you also have to make sure not to increase the "todo" list too much.

Maybe also practical to discuss this in a new issue?

uzi18, 2019-09-25:
please add this task to your "list with things to resolve later, but not to forget", this issue is a good place not need another one =)

Jedrunia, 2019-10-18:
I don't know that this is the same issue, but even when I use only one Dummy Device, and set device type other than "single", after hardware reset plugin is set back to "single".
I use this plugin to send data to Thingspeak, so I receive only first value.

I use last release from 2019-10-03 4Mbit version

TD-er, 2019-10-18:
Do you really save the plugin after changing the output type? (also try moving to another page, dummy device again and submit again)

Jedrunia, 2019-10-20:
Sorry for delayed answer, I needed some time to sit back for this issue.
The difference to post I wrote before is that I use 1M ESP version, ant the settings is changed not after Reset, but after wakeup from Deep Sleep mode.
Answer for your questions:

1.	Yes, I save settings, and each time go to devices page and chose once again this dummy device to check that all parameters are OK.
2.	Yes, I change “slot” and the problem still exist

My test environment is:
1.	ESP01S, with additional connection to enable Deep Sleep
2.	Firmware: ESP_Easy_mega-20191003_normal_ESP8266_1M_VCC.bin

What I do:
1.	Set dummy device, chose output data type to Quad, and fill all values names
2.	Save
3.	On Devices page I can see all 4 variables on Values column
4.	Enable deep sleep mode (40sec sleep, 30sec wakeup)
5.	Wait device to go deep sleep mode
6.	Wait for “blink” of LED and I start to login into ESP on browser
7.	Delete 4th value name, save, change type to Triple, save -> on devices page I can see only 3 variables
9.	Wait 1 cycle of deep sleep
10.	And...
11.	Device type change to single, but I can see 4 values on device settings page, and 4 on all devices page. (should be only one!)

I flashed ESP with 1M blank file, after that I flashed device with firmware mentioned above. 
I chose different "slot" that I not tested earlier and still the same effect.

PR #3265 (2020-09-24): Use task specific valueCount and VType instead of shared in plugin
Lots of reported issues regarding strange issues when using multiple instances of the same plugin which allows to change its output variables and value count.

Possible affected plugins:

- P026 Sysinfo
- P033 Dummy
- P052 SenseAir
- P078 Eastron
- P082 GPS
- P085 AcuDC243

N.B. this PR is not yet finished.
Still need to have a look at usage of `sensorTypeHelper_loadOutputSelector` and related functions.

Fixes: #3184 
Fixes: #2675
Fixes: #2626
Fixes: #2600
Fixes: #3045 
Fixes: #3061
closed 2020-09-24 by TD-er