letscontrolit/ESPEasy #2632: PN532 prone to freezing

ghtester, 2019-09-26:
I am just testing the latest firmware custom build (from official sources 20190926)  compiled through Vagrant using attached Custom.h file.
[Custom.h.txt](https://github.com/letscontrolit/ESPEasy/files/3658500/Custom.h.txt)
Everything looks OK so far except PN532 plugin which is not so reliable as with earlier (20190413) firmwares - after first or several readings it hangs (continuous PN532 Read Errors) and the PN532 device can't be reinitialized without powering off/on (reboot is not enough).  So the question is - are there recent changes on i2c bus timing etc. which could lead to this issue? Could anyone else confirm the same behavior?

Currently I have connected and enabled BH1750 on i2c bus, disconnected and disabled LCD2004 device, disconnected and disabled SSD1306 device (see the attached screenshot).

Firmware:
Build:⋄ | 20104  - Mega
System Libraries:⋄ | ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Sep 26 201908:20:00
Plugins:⋄ | 34 [Normal]
Build Md5: | e6d1ef3eb972c31b6659c853501e48
Md5 check: | passed.
Build Time:⋄ | Sep 26 2019 08:09:00
Binary Filename:⋄ | ESP_Easy_20190926_vagrant_custom_ESP8266_4M1M.bin

101697: EVENT: Clock#Time=Thu,16:56
122029: Dummy: value 1: 0.00
122030: EVENT: Time#1=0.00
122058: EVENT: Time#2=0.00
122088: EVENT: Time#3=0.00
122121: EVENT: Time#4=0.00
122259: WD   : Uptime 2 ConnectFailures 0 FreeMem 14968 WiFiStatus 3
129212: BH1750 Address: 0x23 Mode: 0x1 : Light intensity: 26.67
129213: EVENT: Light#Lux=26.67
141796: PN532: Tag: 3286420891 1
141803: EVENT: RFID#Tag=3286420891
142032: PN532: Read error: 1
142410: PN532: Tag: 3286420891 2
142411: EVENT: RFID#Tag=3286420891
142696: PN532: Tag: 3286420891 3
142697: EVENT: RFID#Tag=3286420891
152299: WD   : Uptime 3 ConnectFailures 0 FreeMem 14944 WiFiStatus 3
159211: BH1750 Address: 0x23 Mode: 0x1 : Light intensity: 36.67
159212: EVENT: Light#Lux=36.67
160996: PN532: Tag: 3286420891 4
161002: EVENT: RFID#Tag=3286420891
161310: PN532: Tag: 3286420891 5
161311: EVENT: RFID#Tag=3286420891
161610: PN532: Tag: 3286420891 6
161611: EVENT: RFID#Tag=3286420891
161697: EVENT: Clock#Time=Thu,16:57
161896: PN532: Tag: 3286420891 7
161897: EVENT: RFID#Tag=3286420891
173296: PN532: Tag: 3286420891 8
173297: EVENT: RFID#Tag=3286420891
173533: PN532: Read error: 1
173911: PN532: Tag: 3286420891 9
173912: EVENT: RFID#Tag=3286420891
174196: PN532: Tag: 3286420891 10
174197: EVENT: RFID#Tag=3286420891
179596: PN532: Tag: 3286420891 11
179597: EVENT: RFID#Tag=3286420891
179833: PN532: Read error: 1
180133: PN532: Read error: 2
180433: PN532: Read error: 3
180733: PN532: Read error: 4
181033: PN532: Read error: 5
181333: PN532: Read error: 6
181633: PN532: Read error: 7
181933: PN532: Read error: 8
182029: Dummy: value 1: 0.00
182033: EVENT: Time#1=0.00
182063: EVENT: Time#2=0.00
182092: EVENT: Time#3=0.00
182123: EVENT: Time#4=0.00
182252: WD   : Uptime 3 ConnectFailures 0 FreeMem 14928 WiFiStatus 3
182332: PN532: Read error: 9
182632: PN532: Read error: 10
182932: PN532: Read error: 11
[... 243 lines cut ...]
88694: PN532: Read error: 107
88994: PN532: Read error: 108
89294: PN532: Read error: 109
89594: PN532: Read error: 110
89894: PN532: Read error: 111
90194: PN532: Read error: 112
90494: PN532: Read error: 113
90794: PN532: Read error: 114
91094: PN532: Read error: 115
91394: PN532: Read error: 116
91694: PN532: Read error: 117
91812: BH1750 Address: 0x23 Mode: 0x1 : Light intensity: 33.33
91813: EVENT: Light#Lux=33.33
91994: PN532: Read error: 118
92252: WD   : Uptime 2 ConnectFailures 0 FreeMem 14896 WiFiStatus 3
92294: PN532: Read error: 119
92594: PN532: Read error: 120
92894: PN532: Read error: 121
93194: PN532: Read error: 122
93494: PN532: Read error: 123
93794: PN532: Read error: 124
94094: PN532: Read error: 125

BTW. it looks when there are firmware changes / missing plugins, it has some impact to device configuration - I don't know how it works / should work but in configuration my IRTX device has changed to Input switch (otherwise I must be crazy... :-) ) when I updated the firmware to custom version without IRTX plugin.
[image]

uzi18, 2019-09-26:
@TD-er little OffTopic we can now use string as output for card ID, so no 32 bit limit anymore, hmm?

TD-er, 2019-09-26:
@uzi18 
String as output is highly experimental, so not yet.

And about the existing plugin config with a firmware not having that plugin included....
I think we should detect it and not show up as "Switch" plugin.
So no, you're not loosing your mind, only functionality of that plugin with the firmware build :)

One of the things you changed in your build is running core 2.6.0 instead of core 2.5.2
It is possible something has changed in the core library regarding I2C.
Maybe you could also test with core 2.5.2?

ghtester, 2019-09-27:
Thanks for the hint, I'll try when I find where to change the core release :-) So far I (hopefully) kept the defaults, just added the Custom.h file with uncommented plugin / controller definitions I need. 
Still need to learn the Vagrant environment, I am happy I was even able to self compile. ;-) Many thanks for preparing this tool, it's a great support for BFUs like me so we are now able to customize the ESP Easy firmware easily.

ghtester, 2020-01-10:
Well, as I don't encounter this issue anymore in latest ESP Easy firmware releases, I think it can be closed. Nevertheless, the PN532's Reset pin should be used for a serious application.

closed 2020-01-10 by ghtester