letscontrolit/ESPEasy #2636: Wrong P052 Senseair plugin configuration

sobolkz, 2019-09-28:
Detected in mega-20190928.
After new device (052 "Gases - CO2 Senseair) was added and submitted - new information was displayed at it's configuration page:
[image]
Wrong value was displayed in entry fields.
If you are trying to change any settings of plugin new error was displayed:
[image]
Sorry for russian, it's mean " Value shoud be more or equal 2".
It can be changed manually for submitting, but it can't be saved for next time.
Also - bad sensor data was displayed even if sensor was disconnected then plugin is active:
INIT : Booting version: mega-20190928 (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
84 : INIT : Free RAM:30344
85 : INIT : Rebooted from deepsleep #29 Last Task: Const Interval timer, id: 1 - Restart Reason: Deep-Sleep Wake
87 : FS : Mounting...
111 : FS : Mount successful, used 76053 bytes of 957314
580 : CRC : program checksum ...OK
591 : CRC : SecuritySettings CRC ...OK
694 : INIT : Free RAM:27168
696 : INIT : I2C
696 : INIT : SPI not enabled
697 : EVENT: System#NoSleep=60
2966 : S/N: 0 - Type: 0 - **### - sensor wasn't detected, no it's S\N, perfect.**
5344 : INFO : Plugins: 79 [Normal] [Testing] [Development] (ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
5345 : EVENT: System#Wake
5517 : WIFI : Connecting access_point8 attempt #0
5532 : EVENT: System#Boot
6096 : Senseair:
6099 : EVENT: S8#CO2=3766.00  ### **-CO2 level from undetected device.**
6379 : ADC : Analog value: 0 = 0.000
6382 : EVENT: VBat#Vbat=0.00
7260 : Dummy: value 1: 755.25
7262 : EVENT: P_Hg#P_Hg=755.25
7298 : EVENT: P_Hg#=0.00
7307 : EVENT: P_Hg#=0.00
7316 : EVENT: P_Hg#=0.00
7936 : BMx280 : Detected BME280
8064 : WD : Uptime 0 ConnectFailures 0 FreeMem 19728 WiFiStatus 6

sobolkz, 2019-09-28:
P.S. - it can't be fixed after plugin removal, saving settings or (and) new device with same type adding after it - same issue at plugin page.

TD-er, 2019-09-28:
Just tested it with the latest nightly 'normal' build:

SenseAir S11 (Sunrise) sensor:
[image]

SenseAir S8 sensor:
[image]

But both do show a believable CO2 concentration of about 670 ppm here on my desk.
I do agree the bogus values for S8 should not be visible when the sensor does not support it.

TD-er, 2019-09-28:
Ah, I already see where things have changed.
The last changes I made to the Modbus functions, also changed the return types of the functions.
The SenseAir plugin does check explicitly if the value is not -1, to hide some functions.

I will update these checks to again hide the unsupported values.

TD-er, 2019-09-28:
I do have a fix ready.
Since it is not handling the Modbus error codes well, I have made it a priority fix.
I don't know if continuous sending unsupported commands and register writes to the sensor may damage it.
Better safe than sorry, so this will be included in tomorrow's build.

sobolkz, 2019-09-28:
> I do have a fix ready.
> Since it is not handling the Modbus error codes well, I have made it a priority fix.
> I don't know if continuous sending unsupported commands and register writes to the sensor may damage it or corrupt any settings or calibration.
> Better safe than sorry, so this will be included in tomorrow's build.
As I understand S8 internal measuring interval - 4 sec. But I'm also don't know how often it can be requested if two side connection was enabled.
Ok, I can try to check results for this sensor with native sketch and you fixed version of plugin. Strange, that native sketch returns realistic values from first start, but P052 was not (as you see before). Problem with deep sleep stops my work with it, but I can start it again and see what is it.

TD-er, 2019-09-28:
Like I said, there was definitely some bug introduced by changing the return values of the read functions in the modbus library.
So you should consider the SenseAir plugin of the last nightly build broken.

It would be nice if you could test this PR I made.

I have it running now on 2 nodes for testing, but both do not use deep sleep.

closed 2019-10-02 by TD-er

sobolkz, 2019-10-04:
> Like I said, there was definitely some bug introduced by changing the return values of the read functions in the modbus library.
> So you should consider the SenseAir plugin of the last nightly build [...]

Hi!
It seem's to all done. Deep sleep AND Senseair plugin now are working without problems. Thank you again!

fluppie, 2019-12-09:
@TD-er  I see you have a Sunrise S11 as well. I just received mine from Grovkillen's shop.
Do you have info on the wiring, I see an EN pin and nRDY. Do you use serial or I2C?

TD-er, 2019-12-09:
I use serial. The plugin does not have support (yet) for the I2C.

The EN pin should not left floating.
Enable (active high).
Drive this pin over 1.2V to turn on the sensor and low (below 0.4V) to put the sensor in shutdown mode.
Connect to VBB if not used.

nRDY is an output signaling when the measurement is ready. The plugin does not use that pin.

The only pins you need are:
GND
VBB  - Sensor supply voltage
VDDIO - supply voltage for TX/SCL/nRDY

COMSEL - select UART/I2C => Can be left floating due to internal pull-up which is HIGH = UART.
EN - Connect to VBB

RX  
TX  (100k pull up to VDDIO)

As you can see, you have to make sure not to connect the TX of the sensor to some pin that may need a specific state during boot.
I used mine with this config and it works just fine for S8 as well as S11:
[image]

fluppie, 2019-12-09:
Nice, thanks for the info. Let me update to the latest build and rewire :)

TD-er, 2019-12-09:
Just keep in mind that the extra settings you will see with the S11 may not all be tested very well.
Some are also not really that well documented in the public SenseAir documentation :)
So better leave them at their defaults.
You can however use CO2 and temp if you like.

fluppie, 2019-12-09:
Got to be kidding me... Soldering iron stops working after 4 pins...

TD-er, 2019-12-09:
Hmm flaky soldering iron is something that must be taken very serious.
Before you know it, you have either mains power on the tip or a tip that gets way too hot. 
I've once seen a soldering iron glow, which later appeared to be a defective temperature control.

fluppie, 2019-12-09:
Luckily this one stays cold and I didn’t feel mains power when touching the tip :-D. I’ve had this one for about 20 years, got it when I was 12 :-). So many things I’ve soldered with this one...

fluppie, 2019-12-10:
[image]

:) :) :)

fluppie, 2019-12-10:
And the second one works too :)

[image]

Thanks for your help!

Grovkillen, 2019-12-10:
Thanks for the feedback :)

PR #2638 (2019-10-02): Fix SenseAir plugin incorrect Modbus error handling
Fixes: #2636
Fixes: #2639 
Fixes: #2642