letscontrolit/ESPEasy #2637: Update page showing 2 upload fields in core 2.6.0

TD-er, 2019-09-28:
The page on /update does now show 2 fields:
[image]

I guess this is some addition in the latest stage branch of esp8266/Arduino.
This is rather confusing, so we have to see how we can tackle this.

Not sure what updating the filesystem does with existing files.
It may be about a filesystem image, or a set of files, so do not use the option.

Grovkillen, 2019-09-28:
Yeah I was thinking that my GUI would have it's own upload page. So my suggestion is that we then hide the core enabled page.

TD-er, 2019-09-28:
That's easier said than done.
We now just register the url to call some service.
All that's running when calling /update is handled by some service in the core libs.

Grovkillen, 2019-09-28:
Ok, that's something we might suggest them handle some other way. Core wise I mean. 😬

TD-er, 2019-09-28:
I was thinking of finally wrapping that service in our own page.

Grovkillen, 2019-10-01:
Just let it be an endpoint for my GUI to post the file to and the SPA will take care of the rest.