letscontrolit/ESPEasy #2640: Support Clock#Time< or > in rules

TD-er, 2019-09-30:
Currently the rules only support a match on an exact clock event.
But for some use cases it would be great to have a proper initial state based on some time interval when powering on the ESP.
For example when using time based events which are taking a while to happen like daily or weekly events.

It may need some new interval definition, to make it clear what happens.

For example:
```
on Clock#Time>All,**:30 
```
When should this be considered true?
Between xx:31 and xx:59? (including those values)

giig1967g, 2019-10-01:
Nice idea.

giig1967g, 2019-10-01:
I currently do this to properly set my house lights after a reboot:
```

on Time#Initialized do
 timerSet,2,10
endon

On Rules#Timer=2 do
 if %systm_hm%>%sunset+1h%
  do something
 elseif %systm_hm%>%sunset% 
  do something
 elseif %systm_hm%<%sunrise%
  do something
 elseif %systm_hm%<%sunrise-1h%
  do something
 else
  do something
 endif	
endon
```

giig1967g, 2019-10-01:
So it could become:
```
on Clock#Time>All,%sunset+1h% do
  do something
endon

on Clock#Time>All,%sunset% do
  do something
endon

on Clock#Time>All,%sunrise% do
  do something
endon

on Clock#Time>All,%sunrise-1h% do
  do something
endon

```

TD-er, 2019-10-01:
The (new) problem then is that your task should be able to be executed more than once.
I think your if...elseif solution would be better suitable to be executed right after boot (or some scheduled delay after boot)

giig1967g, 2019-10-01:
that's true. 
A more complex could be:

`on Clock#Time>All,%sunrise% do //allow repeat of the command`

`on Clock#TimeOnce>All,%sunrise% do //execute the command only once per day`

TD-er, 2019-10-01:
I think the syntax should then be like `on ... do once` or something like that.
Functionality like that can already be achieved by setting a variable when it has been done.