letscontrolit/ESPEasy #2641: Command "LogEntry" should output the text "LogEntry," to the log line/output

Grovkillen, 2019-09-30:
Currently using the "LogEntry" command to send info to the log does also output the text `LogEntry,` which isn't the way it should be.

This isn't wanted:
```
182840 : ACT  : LogEntry,LED turned "OFF" (0)
182843 : Command: logentry
182843 : LogEntry,LED turned "OFF" (0)
```

This is wanted:
```
182840 : ACT  : LogEntry,LED turned "OFF" (0)
182843 : Command: logentry
182843 : LED turned "OFF" (0)
```

TD-er, 2019-09-30:
I can imagine people may now grep in the logs when sending it to a logserver?

Grovkillen, 2019-09-30:
I guess so, but you could still add that string of text if you need it.

giig1967g, 2019-10-01:
Probably to retain the log style, it should look as follows:
182843 : LogEntry: LED turned "OFF" (0)
(change the comma to a colon after the word LogEntry)

TD-er, 2019-10-01:
The thing is, it is now just echo'ing the command.
Of course we can strip the arguments, but I think it would make more sense to also look into optional parameters so you can decide where to direct the log (info/error/debug)
We now have about 3 entries indicating a logentry command has been given.
Not sure what loglevel was active in Jimmy's test?

Grovkillen, 2019-10-01:
Yeah the logentry will report on info. So begging y able to set type of log output.

And yeah, three lines in the log isn't what we first intended with the logentry command.

TD-er, 2019-11-04:
It has been changed in the last build.
Also the parameter parsing has improved, so later we can add an optional parameter to select the log level.
So I close this one.

closed 2019-11-04 by TD-er