letscontrolit/ESPEasy #2647: Add setlevel in JSON.

khseal, 2019-10-03:
Please add  setlevel for Regulator - Level Control!

TD-er, 2019-10-03:
Can you elaborate a bit more what you expect?
What has JSON to do with it?
What should be implemented to fulfill this feature request?

khseal, 2019-10-03:
I confused setlevel with getLevel. In JSON no information on the level control. 
Sometimes this information is needed.

TD-er, 2019-10-03:
Again, can you be a bit more specific?
Where should this JSON formatted info be given?
Should it be a specific command for the level plugin?
Or should it be outputted at the /json url?
That last one is not in line with other information being shown since it only lists output values, not settings.

So please give some clear examples of what should be shown where and how to obtain it.

khseal, 2019-10-03:
Sorry me per little information.
> Or should it be outputted at the /json url?
Output to the file information about some settings in
http://ip/ json would be helpful.
Example
"TaskSettings": [
{"Level": xx,
"Hysteresis:xx"
}],
Get level using [<taskname>#getLevel]

TD-er, 2019-10-04:
Hmm, let me think about it. Like I said, the JSON outputs data which is generic for any plugin.
Including tasksettings also needs plugin specific information to interpret the tasksettings, so then to generate the json, it needs to query all (enabled) tasks, to give descriptive information on the task settings.
It is then looping over all enabled tasks again for just another function for each plugin to interpret its settings.

Although, we now have the `PLUGIN_GET_CONFIG`, which is only implemented in this level plugin, but it is requiring a command `getLevel` (case sensitive!!) to get it.
This command can already be executed from anywhere it accepts our rules syntax using:
```
[<taskname>#getLevel]
```

This `PLUGIN_GET_CONFIG` does only return a single string formatted float value and it has to be called using a specific string ("getLevel") to return it.
This may take a lot of iterations, so to optimize it, the `Devices` struct should have a boolean flag indicating it may have extra config information and then add a new function (or see if we have one already) to hand over some formatted label + value to describe the settings.
The only interaction we now have with plugins is to give a function (e.g. `PLUGIN_GET_CONFIG`) and then return a string.
Then this string must be decoded and converted into JSON.
Also should we call this for one single setting, or receive all settings at once? How to decode it without wasting a lot of resources?
Or should we add a function to let the plugin generate JSON? That's going to be a lot of work when we later decide to re-structure the JSON formatting (all plugins should be changed)

And there's a dozen other issues I can think of to implement it with their drawbacks.

khseal, 2019-10-04:
I realized that the implementation of the data from the settings is not easy. =(

closed 2019-10-04 by khseal