letscontrolit/ESPEasy #2649: Rotary encoder feature requst. Instead of counting up and down, communicate intent up or intent down.

Alacritous, 2019-10-06:
The rotary encoder feature works as advertised, but is limited in its application. 

As it stands now, the Rotary Encoder device communicates a value with a set range to the controller and based on the movement of the attached rotary encoder it increments or decrements that value inside that range.  I'm using Node-red to attach that value to a sound volume for example. 

If something else changes that value in Node-red and then the rotary encoder is engaged it jumps to the new value supplied by the encoder. Which can be jarring when that value is linked to a sound volume. 

So I'm requesting an option to have it simply communicate "encoder up" or "encoder down" intent to the controller.   A "+1" or "-1" or whatever that turns out to be.

TD-er, 2019-10-06:
I think you may want to have a look at the formula field.
This formula field can compare a value to the previous one.
Only thing is, you may need to take into account what to do when wrapping on max/min value.

Maybe you can do it already in rules by storing the raw value in a dummy and determine whether it is moving up or down.

I think it can be made as an output option and would be a useful addition.
However, I also have limited time for it right now, so it may take a while.

Alacritous, 2019-10-06:
Take your time. I appreciate your efforts.  In the meantime, I'll explore your suggestions.