letscontrolit/ESPEasy #2650: AP Mode not working when WIFISSID is set

martament, 2019-10-07:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [X] This issue describes what components are affected (e.g. name of plugin/controller)
- [X] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
https://github.com/letscontrolit/ESPEasy/issues/2567

- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
ESP_Easy_mega-20191003_normal_ESP8266_4M1M.bin and others

- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X] Tested previous/other build (mention which one already tested)
- [X] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

AP Mode is working if no WIFISSID is set. IF a WIFISSID is set and not reachable AP Mode is started, but not usable.

### Expected behavior

AP Mode should be usable, if WIFISSID is not reachable. 

### Actual behavior

I can see the AP Mode and connect to it. But in the background, as i can see on console, the device tries to connect to the not reachable WIFISSID and the device is unusable.
Sometimes i can reach http://192.168.4.1 or http://192.168.4.1/setup . But it is not usable.

### Steps to reproduce

Type A to reproduce:

1. Flash image with no WIFISSID set
2. AP Mode is started and WIFI can be configured
3. Save settings and restart device
3. If configured SSID is not reachable, or SSID is changed to a non existing one AP Mode is started
4. I can connect to AP of the device, but i cant use it.

Type B:
1. Flash image with no WIFISSID set
2. AP Mode is started and usable
3. set WIFI SSID via console: `WIFISSID XXX` and `Save` and `reboot`
4. I can connect to AP of the device, but i cant use it.

Fix issue of Type A or B:
1. set WIFISSID to ssid,save and reboot
2. I can connect to the device and use it

### System configuration

Hardware:
WEMOS D1 MINI

ESP Easy version: ESP_Easy_mega-20191003_normal_ESP8266_4M1M.bin

ESP Easy settings/screenshots: 

### Rules or log data

[... 168 lines cut ...]

Ok
589645 : WIFI : No valid wifi settings
589646 : WIFI : Could not prepare WiFi!
602437 : WD   : Uptime 10 ConnectFailures 0 FreeMem 20856 WiFiStatus 6
609566 : AP Mode: Client disconnected: A4:50:46:6A:11:CD Connected devices: 0
609788 : WIFI : No valid wifi settings
609790 : WIFI : Could not prepare WiFi!
613020 : AP Mode: Client connected: A4:50:46:6A:11:CD Connected devices: 1
629922 : WIFI : No valid wifi settings
629923 : WIFI : Could not prepare WiFi!
>settings
822202 : Command: settings

System Info
  IP Address    : (IP unset)
  Build         : 20104
  Name          : ESP_Easy
  Unit          : 0
  WifiSSID      : ssid
  WifiKey       : wpakey
  WifiSSID2     : 
  WifiKey2      : 
  Free mem      : 19896
```

TD-er, 2019-10-07:
Ah thank you for the analysis.
I noticed similar behavior, but that was on nodes that could not be read from the serial port.
So I could not see what was happening.

I also had it once that the unit does react to ping, but it looked like it was probably not aware it was connected. AP was active, but indeed not possible to connect to (and stay connected or receive an IP) and it was not sending data via the controllers.

martament, 2019-10-24:
One usecase to consider:
- Using this device as a **Standalone-Device**

- If no WIFI is reachable this bug allows to use the device without wifi. You can connect to the AP and Query the Webpage. This is a useful feature even it is not very stable
- If this bug is fixed and AP mode is recognized correct, then you can't use it standalone anymore. Because in correct AP mode all request are redirected to the WIFI-Manager .
- This is not always wanted behaviour.

- I know, that there my be security reasons for not allowing using the device without wifi and without security.

Possible solutions:
a) If device is started and no wifi is reachable. AP mode is started and recognized, make the other pages reachable **for a certain amount of time**. (2-3 Minutes schould be enough). 
b) If device is started and no wifi is reachable. AP mode is started and recognized, make the **dashboard pages reachable** for a certain amount of time. (2-3 Minutes schould be enough).
c)   If device is started and no wifi is reachable. AP mode is started and recognized, make the dashboard pages reachable if **a admin password** is set
d) If device is started and no wifi is reachable. AP mode is started and recognized, make **dashboards (*.esp) reachable for unlimited amount of time.**

My favorit is solution d).

martament, 2019-10-24:
I understand...
 My opion:

> Don't let the connect command do the scanning, but scan for yourself and keep the records for when to make a connection. (this data needs a time to live administration)

I would take that option. 
Reasons:

- scanning without somenone to chose a wifi from the result is not necessary. (Correct?)

- If someone is connected to ap we have to choices in my opinion.

1. He knows SSID and Password and he doesnt even want to use a scan or the ssid is hidden. So we don't need to scan. 

1. He wants to scan, so he can do the scan himself. But after the scan we stop it and he can make his choice .

TD-er, 2019-10-24:
The SSID and password are set by the user.
To make the actual connection, you must know 2 completely different parameters:
- BSSID
- Channel

Meaning, when you call the `WiFi.begin()` function, you can also give the BSSID and channel and then the core library code does not need to scan for WiFi networks. It will either connect successful or not.
If you do not supply the BSSID and channel information, the WiFi code does start a scan and looks for the matching SSID in the scan results.

If you do the scanning yourself, you can also take out the known SSID's and filter on RSSI values to take the strongest as first attempt (repeaters use the same SSID, but have different BSSID).
If that doesn't work, you can try the next one etc. but no longer have to wait for the entire scan to complete.

So performing the scanning yourself has several positive side effects:
- less time wasted
- Able to take out the strongest
- Base your decision on extra knowledge like multiple AP's with the same SSID (e.g. repeaters)

It also has a negative side effect. This will not work for "hidden SSID" configurations.
But for those we can store the known BSSID as soon as we make a normal WiFi connection without providing the BSSID/channel.

This is all very nice, but the key issue we're facing here can also be solved without taking the scanning in our own hands.
Just don't try to reconnect as long as someone is connected to the AP. Or at least not in the first N seconds/minutes after the client connected to the AP.

The current implementation can already be used as a stand-alone setup, which also does work just fine if there is no WiFi connection present. Even if it isn't configured or just unreachable.
The internal rules engine is just working as it should.

TD-er, 2019-10-25:
First part of this issue should be fixed.
The fix prevents an automatic reconnect attempt as long as someone is connected to the AP mode of the ESP.

Still need to address the logic when to use the last known BSSID/channel.
Later I will also look into manual performing a WiFi scan and using the best RSSI value first.

PR #2667 (2019-10-27): [Rules] Improve rules matching (#2660)
Detecting time compare and value compare could give false matches in some cases.
This PR also incorporates a number of other changes:
- Adds option to change the remote unit on remote fed plugins via p2p. (#2679)
- Cleanup of use of taskIndex, DeviceIndex (#2676)
- Disable and mark configured plugins not included in the build (e.g. when switching builds with other plugins included) (related to #2676)
- Fix build issues with PlatformIO using Python3
- Perform first read of ADC at init, before starting WiFi (#2646)

Fixes: #2683 - Duplicated section in `pla [... cut]
closed 2019-10-27 by TD-er