letscontrolit/ESPEasy #2651: MQTT not working under Wroom2 20191003

fraeggle, 2019-10-07:
I am using the prebuild file in the zip 
"ESP_Easy_mega-20191003_normal_core_252_WROOM02_2M256.bin"
When i use this one on most of my ESP's suddenly MQTT does not work (not updating or switch)
If I go back to the version 20190928  MQTT works immediately again.

Only one thing i have not tested until now. Factory Reset and load 20191003 again, because hard to get a hand on the Modules. I will try tomorrow if i can get some time.

Maybe someone has the same issue.

TD-er, 2019-10-08:
How do you use MQTT?
Via a controller (which one?) or via rules?
A bit more information would be nice.

What does "not work" anymore?
No connection to the broker?
Connection lost?

fraeggle, 2019-10-08:
I am really sorry............
I had everything prepared, but then I had to leave urgently.
I use the IOBroker with MQTT as broker.
The ESP is configured with Home Assistant (openHAB) MQTT.
I use MQTT Import and Rules in the ESP to control the IO ports via IOBroker. With 20190928 the MQTT import from broker to ESP works.
Same ESP (occurs at 5 ESP), same configuration---> 20191003 Firmware: MQTT Import makes no change.
Another strange behavior at 2 ESP where the 20191003 goes: Suddenly the outputs start to switch in half-second intervals. Like when MQTT is used to suddenly transmit the states incorrectly.
If I then play the 20190928 in all again, everything works again without disturbances.
The connection to the broker himself looks like it would always be there. Switching states which I carry out directly at the ESP are transmitted cleanly to the broker. Only with MQTT Import I have problems.

Regards Peter

TD-er, 2019-10-08:
Can you post the rules?

The mayor changes in the last build is about the rules parsing.
It is mainly about reduction of resources used when processing rules and thus also make the rules parsing faster.
I also fixed a few things along the way and added a few features like make it possible to match a rule by comparing with 2 character compares (<= , >=, <>)

So please show me the rules and then I can see which of these changes could be related to your specific rules.
Maybe you found a bug, maybe something that was working by accident has now been broken due to a fix I made.

fraeggle, 2019-10-08:
On the one which really doesn't work it is very Simple.

on MQTT#Connected do
  Publish %sysname%/status/IP,%ip%
EndOn

// SPG_OUT	GPIO-4

//  MQTT Server (muss angelegt werden): ESP_TAB
//  ESP_TAB.SPG_OUT.status

// Switch MQTT IMPORT Device 11:  switch#Wertxx
// %sysname%.switch.Wertx
// 1 SPG_OUT   io4

// dummy Device 12:  status#xxxx  ohne MQTT
// 1 SPG_OUT   io4

On switch#Wert1 Do
  If [switch#Wert1]=1
   GPIO,4,1
   TaskValueSet,12,1,1
   Publish,%sysname%/SPG_OUT/status,1
  Else
   GPIO,4,0
   TaskValueSet,12,1,0
   Publish,%sysname%/SPG_OUT/status,0
   EndIf
EndOn

[image]

on the one who suddenly startet switching every second (Light LED Kitchen  IO12):
[image]

on MQTT#Connected do
  Publish %sysname%/status/IP,%ip%
endon

// MQTT Server (muss angelegt werden):
// ESP_1OG.Licht.status
// ESP_1OG.LED_Kueche.status
// ESP_1OG.LED_Treppe.status

// Automatik MQTT Import Device 10: sniff#Wertx
// %sysname%.Steuerung.Wert1  Treppe_Zeit
// %sysname%.Steuerung.Wert2  LED_Kueche_Manuell
// %sysname%.Steuerung.Wert3  Treppe_Manuell
// %sysname%.Steuerung.Wert4  LED_Kueche_LUX

// Switch MQTT IMPORT Device 11:  switch#ioxx
// %sysname%.switch.Wertx
//1 LICHT Kueche  io4
//2 LED Kueche    io12
//3 LED Treppe    io14
//4 Reset

// dummy Device 12:  status#ioxx
//1 LICHT Kueche io4
//2 LED Kueche   io12
//3 LED Treppe   io14

On Licht_Kueche_Taster#Taster_Licht_Kueche=0 Do
  If [status#io4]=1
    event,LICHTKUECHE_AUS
  Else
    event,LICHTKUECHE_AN
  EndIf
EndOn

On switch#Wert1 Do
  If [switch#Wert1]=1 and [status#io4]=0
   event,LICHTKUECHE_AN
  EndIf
  If [switch#Wert1]=0 and [status#io4]=1
    event,LICHTKUECHE_AUS
  EndIf
EndOn

on LICHTKUECHE_AN do
  TaskValueSet,12,1,1
  GPIO,4,1
  Publish %sysname%/Licht/status,1
EndOn

on LICHTKUECHE_AUS do
  TaskValueSet,12,1,0
  GPIO,4,0
  Publish %sysname%/Licht/status,0
EndOn

on RF_Kueche#RF_Kueche do
 event,LEDKUECHE
Endon

on LEDKUECHE do
 if [RF_Kueche#RF_Kueche]=1 and [status#io12]=0
   if [sniff#Wert4]=0 and [sniff#Wert2]=0
     PWM,12,1023,1000
     TaskValueSet,12,2,1
     Publish,%sysname%/LED_Kueche/status,1
   endif
   if [sniff#Wert4]=1 and [sniff#Wert2]=0
     PWM,12,512,500
     TaskValueSet,12,2,1
     Publish,%sysname%/LED_Kueche/status,1
   endif
 endif
 if [sniff#Wert4]=1 and [sniff#Wert2]=0
   PWM,12,512,500
   TaskValueSet,12,2,1
   Publish,%sysname%/LED_Kueche/status,1
 endif
 if [RF_Kueche#RF_Kueche]=0 and [sniff#Wert2]=0
   PWM,12,0,1000
   TaskValueSet,12,2,0
   Publish,%sysname%/LED_Kueche/status,0
 endif
endon

On switch#Wert2 Do
  If [switch#Wert2]>0 and [status#io12]=0
   PWM,12,1023,1000
   TaskValueSet,12,2,1
   Publish,%sysname%/LED_Kueche/status,1
  endif
  If [switch#Wert2]<1
   PWM,12,0,1000
   TaskValueSet,12,2,0
   Publish,%sysname%/LED_Kueche/status,0
   EndIf
EndOn

on RF_Treppe#RF_Treppe do
 if [sniff#Wert3]=0
  event,AUTOSPERRE
 endif
endon

on AUTOSPERRE do
 if [sniff#Wert1]=0 and [status#io14]=0 and [RF_Treppe#RF_Treppe]=1
  PWM,14,1023,1000
  TaskValueSet,12,3,1
  Publish,%sysname%/LED_Treppe/status,1
 endif
 if [status#io14]=1 and [RF_Treppe#RF_Treppe]=0
  PWM,14,0,1000
  TaskValueSet,12,3,0
  Publish,%sysname%/LED_Treppe/status,0
 endif
endon

On switch#Wert3 Do
  If [switch#Wert3]>0 and [status#io14]=0
   PWM,14,1023,1000
   TaskValueSet,12,3,1
   Publish,%sysname%/LED_Treppe/status,1
  endif
  If [switch#Wert3]<1
   PWM,14,0,1000
   TaskValueSet,12,3,0
   Publish,%sysname%/LED_Treppe/status,0
   EndIf
EndOn

TD-er, 2019-10-08:
I added block quote markup to make your rules a bit more readable.

TD-er, 2019-10-08:
Just as a first idea, can you change all keywords like if, else, etc. to lower case?
Maybe I made a mistake there...

fraeggle, 2019-10-08:
I'll try tonight.

TD-er, 2019-10-08:
Just curious.
On the first rule set (not working you said), you trigger a rule on changing the value of a variable in MQTT import.
`On switch#Wert1 Do`

Can you give an example of such a MQTT message and how it is imported?
I did also change something in parsing integer/float values.
Maybe that could have introduced an unintended change?

fraeggle, 2019-10-08:
found the issue: In the handling of the values in the script at the Broker:

After i Changed following in the script it works
Getting Stauus of 5 Volt Output:.
....
  spg_tab13 = getState("mqtt.0.ESP_TAB.SPG_OUT.status").val;  ---> left unchanged
.....

  if (akku_tab13 <= 25 && spg_tab13 == 0) {   ---> changed from ....spg_tab13 === 0 to ...spg_tab13 == 0
    setState("mqtt.0.ESP_TAB.switch.Wert1", 1);
  }
  if (akku_tab13 >= 95 && spg_tab13 == 1) {   ---> changed from ...spg_tab13 ===1 to ...spg_tab13 == 1
    setState("mqtt.0.ESP_TAB.switch.Wert1", 0);
  }

Did you changed something about the type to be transferred with MQTT?

Found following:
---> What is difference between == and === in JS?
In one word, main difference between "==" and "===" operator is that formerly 
compares variable by making type correction e.g. if you compare a number with
a string with numeric literal, == allows that, but === doesn't allow that, 
because it not only checks the value but also type of two variable, if two variables are ...

TD-er, 2019-10-08:
What I changed is to accept leading spaces in a numerical string.
Also there are several functions here in use.
- To check if something is a numerical (that may have changed in behavior)
- To convert a string to a numerical.

The last one is also done in several ways in the code:
- Use the C++ function `atof`, which did already accept leading spaces and just returns 0 if it is no valid float.
- First use my check and convert to numerical (may have changed behavior now)
- Get a numerical string part from another string (may have changed behavior now)

Especially the last one may change the type of a returned value, since it may not include the trailing dot if it is exclusively is checking for an integer type and not a float.
So for JavaScript it may indeed matter if it is checking against `===` or `==`.

fraeggle, 2019-10-08:
Ok really looks like this is the goal.
Made a "Quik" Test. So no Problem for me. I took a look at the once who worked... guess what? in this scripts i only use "=="....

Unfortunately i don't see the type the ESP sends to the Broker. Anyway ,I changed my scripts. I will have a try for a few days  :-))

Here the log from a ESP with "old" Firmware and the new one.
Thanks for your help.

mega-20190830: "===" value in JS

122264: WD : Uptime 2 ConnectFailures 0 FreeMem 17208 WiFiStatus 3
145526: IMPT : [switch#Wert1] : 1.00
145528: EVENT: switch#Wert1=1.00
145555: ACT : GPIO,4,1
145559: SW : GPIO 4 Set to 1
145563: ACT : TaskValueSet,12,1,1
145572: Command: taskvalueset
145581: ACT : Publish,ESP_Odys_1/SPG_OUT/status,1
145591: Command: publish
150934: EVENT: Clock#Time=Tue,17:06
152239: WD : Uptime 3 ConnectFailures 0 FreeMem 19416 WiFiStatus 3
181979: Dummy: value 1: 1.00
181980: Dummy: value 2: 0.00
181980: Dummy: value 3: 0.00
181981: Dummy: value 4: 0.00
181985: EVENT: Status#io4=1.00
182035: EVENT: Status#Wert2=0.00
182072: EVENT: Status#Wert3=0.00
182107: EVENT: Status#Wert4=0.00
182238: WD : Uptime 3 ConnectFailures 0 FreeMem 19384 WiFiStatus 3
210934: EVENT: Clock#Time=Tue,17:07
212239: WD : Uptime 4 ConnectFailures 0 FreeMem 19368 WiFiStatus 3
241954: Dummy: value 1: 1.00
241955: Dummy: value 2: 0.00
241956: Dummy: value 3: 0.00
241956: Dummy: value 4: 0.00
241960: EVENT: Status#io4=1.00
241996: EVENT: Status#Wert2=0.00
242031: EVENT: Status#Wert3=0.00
242068: EVENT: Status#Wert4=0.00
242238: WD : Uptime 4 ConnectFailures 0 FreeMem 19368 WiFiStatus 3
270934: EVENT: Clock#Time=Tue,17:08

mega-20191003:
	
NO IMP --> In JS still "==="
43864: WD : Uptime 1 ConnectFailures 0 FreeMem 21344 WiFiStatus 3
65565: EVENT: Clock#Time=Tue,17:12
73625: Dummy: value 1: 0.00
73625: Dummy: value 2: 0.00
73626: Dummy: value 3: 0.00
73627: Dummy: value 4: 0.00
73630: EVENT: Status#io4=0.00
73673: EVENT: Status#Wert2=0.00
73703: EVENT: Status#Wert3=0.00
73732: EVENT: Status#Wert4=0.00

Changed in JS from "===" to "=="
125568: EVENT: Clock#Time=Tue,17:13
133597: Dummy: value 1: 0.00
133598: Dummy: value 2: 0.00
133599: Dummy: value 3: 0.00
133600: Dummy: value 4: 0.00
133603: EVENT: Status#io4=0.00
133633: EVENT: Status#Wert2=0.00
133662: EVENT: Status#Wert3=0.00
133690: EVENT: Status#Wert4=0.00
133867: WD : Uptime 2 ConnectFailures 0 FreeMem 20648 WiFiStatus 3
146854: IMPT : [switch#Wert1] : 1.00
146855: EVENT: switch#Wert1=1.00
146881: ACT : GPIO,4,1
146886: SW : GPIO 4 Set to 1
146890: ACT : TaskValueSet,12,1,1
146900: Command: taskvalueset
146908: ACT : Publish,ESP_Odys_1/SPG_OUT/status,1
146920: Command: publish
73867: WD : Uptime 1 ConnectFailures 0 FreeMem 20912 WiFiStatus 3

TD-er, 2019-10-10:
Can we conclude it is not a bug on ESPeasy side?
Is there something to be done regarding the "undocumented" aspect?
Can this issue be closed?

fraeggle, 2019-10-10:
Sure...
"Is there something to be done regarding the "undocumented" aspect?" Don't know what you mean.
Anyway... after changing the Script it is working now...
Thanks for your help.....

closed 2019-10-10 by fraeggle