letscontrolit/ESPEasy #2652: Generic HTTP Advanced Controller is missing in UI

mrdc, 2019-10-08:
I'm using ESP_Easy_mega-20190903_normal_core_241_ESP8266_4M1M.bin with NodeMCU and Generic HTTP Advanced is missing in UI. Was it removed? I don't see it in changes for releases.
[image]

TD-er, 2019-10-09:
Controller 11 is in the "test" build, not the "normal" build.

See:
```c++
#ifdef CONTROLLER_SET_TESTING
    #define USES_C011   // Generic HTTP Advanced
    #define USES_C012   // Blynk HTTP
    #define USES_C014   // homie 3 & 4dev MQTT
    //#define USES_C015   // Blynk
    #define USES_C017   // Zabbix
    // #define USES_C018 // TTN RN2483
#endif
```

mrdc, 2019-10-09:
Thanks! I've searched info about 

Will be a good idea to make the remark concerning test build here, as I've searched everywhere and wasn't able to find that I need the test build:
https://www.letscontrolit.com/wiki/index.php/EasyProtocols#Nettemp_HTTP_Api_.28via_Generic_HTTP_Advanced.29

Like it's done for plugins: Normal, Testing, Development
https://www.letscontrolit.com/wiki/index.php/Official_plugin_list

closed 2019-10-09 by mrdc

TD-er, 2019-10-09:
The information on the Wiki is a bit outdated, since we're moving the docs to [ReadTheDocs](https://espeasy.readthedocs.io/en/latest/Controller/_Controller.html)

So we're somewhat inbetween then regarding documentation.