letscontrolit/ESPEasy #2653: PWM does not respond to frequency changes

tomatos-st, 2019-10-10:
Hi
My ESP does not respond to frequency changes
I try:
 PWM, < GPIO > , < duty > , < duration > , < frequency >
and no effects
I checked 
ESP_Easy_mega-20191003_normal_core_241_ESP8266_4M1M
ESP_Easy_mega-20191003_normal_core_260_sdk222_alpha_ESP8266_4M1M
ESP_Easy_mega-20191003_test_core_260_sdk222_alpha_ESP8266_4M1M
ESP_Easy_mega-20191003_test_core_260_sdk3_alpha_ESP8266_4M1M
and nothing

There are some binary that support PWM frequency?

TD-er, 2019-10-10:
What is the literal command you use? (including spaces etc.)

Where do you set these commands? In rules, or via HTTP-GET, or another method?

tomatos-st, 2019-10-10:
via  HTTP-GET
http://192.168.100.132/control?cmd=PWM,2,300,0,3000
works all (duty and duration) except the frequency
I followed the instructions https://espeasy.readthedocs.io/en/latest/Plugin/P001.html

kr0815, 2019-11-10:
Any news on this topic?
I additionally have the problems that status gpio, x only returns zero instead of the actual pwm value
I guess this is also related to this problem?

Does anybody know which was the latest working version?

TD-er, 2019-11-10:
I have not done anything on this issue.

Gruffnco, 2019-11-21:
any chance this could be fixed. I am driving a motor which works well, but it sings at the default frequency of 1Khz. 
Thanks

uzi18, 2019-11-21:
is it possible for you to create debug log for us, at time  before and after execution of this command?

uzi18, 2019-11-21:
@TD-er what do you think?

```
--- a/src/_P001_Switch.ino
+++ b/src/_P001_Switch.ino
@@ -820,6 +820,12 @@ boolean Plugin_001(byte function, struct EventStruct *event, String& string)
           pinMode(event->Par1, OUTPUT);
             #endif // if defined(ESP8266)
 
+          if (event->Par4 > 0 && event->Par4 <= 40000 ){
+            #if defined(ESP8266) //TODO: ESP32 not supported, https://github.com/ERROPiX/ESP32_AnalogWrite
+            analogWriteFreq(event->Par4);
+            #endif // if defined(ESP8266)
+          }
+
           if (event->Par3 != 0)
           {
             const byte prev_mode = tempStatus.mode;

```

TD-er, 2019-11-21:
Is that code missing?
I know for a fact that I have tested it in the past, since that remark for max. frequency is mine.
I did test it on my scope.

uzi18, 2019-11-21:
this code was never introduced here before

TD-er, 2019-11-22:
Very strange.
Well it does look good to me.
Maybe have the #ifdef part into the analogWriteFreq function, since it should be TODO'ed there?

uzi18, 2019-11-22:
this lib is reported as working good for esp32 https://github.com/ERROPiX/ESP32_AnalogWrite

uzi18, 2019-11-22:
@tomatos-st @kr0815 @Gruffnco please try to test this build: 
[normal_ESP8266_4M1M.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3880870/normal_ESP8266_4M1M.bin.zip)

tomatos-st, 2019-11-22:
It works :)
I tested from 100Hz to 30KHz and looks ok.

Gruffnco, 2019-11-24:
Just tested the new bin file, set the frequency to 15khz and the motor was nice and quiet.
Many thanks for the effort in fixing this for us.
Will this fix be in the next release?

TD-er, 2019-11-24:
@Gruffnco It should be part of mega-20191123.

closed 2019-11-24 by TD-er