letscontrolit/ESPEasy #2658: SHT3X plugin sometimes can't read data when the interval value is less than 3s

hamed-ta, 2019-10-12:
I'm using the the esp32test_1M8_partition platform on ESP32 which have a SSD1306 OLED and SHT31 connected on I2C bus, the problem is when I set the interval value of SHT31 less than 3S some time I just got nan as temp and humidity values while I don't have any problem with bmp180 or SHT11 connected to the same device so it seems there is a problem in SHT3X plugin, here is the Log of the deive, 

```
22199965: BMP : Temperature: 29.98
22199965: BMP : Barometric Pressure: 1007.98
22199984: SYS : 370.00,213244.00,43.00,-1.00
22200037: SHT3x: Temperature: 30.38
22200038: SHT3x: Humidity: 24.78
22201024: SHT3x: Temperature: nan
22201024: SHT3x: Humidity: nan
22201899: ADC : Analog value: 698 = 698.000
22202038: SHT3x: Temperature: 30.32
22202038: SHT3x: Humidity: 24.87
22202937: BMP : Temperature: 29.98
22202937: BMP : Barometric Pressure: 1008.08
22202959: SYS : 370.00,213224.00,43.00,-1.00
22203038: SHT3x: Temperature: 30.38
22203039: SHT3x: Humidity: 24.81
22203315: WD : Uptime 370 ConnectFailures 0 FreeMem 213108 WiFiStatus 3
22204025: SHT3x: Temperature: nan
22204025: SHT3x: Humidity: nan
22204900: ADC : Analog value: 688 = 688.000
22205039: SHT3x: Temperature: 30.35
22205039: SHT3x: Humidity: 24.78
22205935: BMP : Temperature: 29.98
22205935: BMP : Barometric Pressure: 1008.03
22205957: SYS : 370.00,213200.00,44.80,-1.00
22206041: SHT3x: Temperature: 30.38
22206041: SHT3x: Humidity: 24.79
22207025: SHT3x: Temperature: nan
22207025: SHT3x: Humidity: nan
```

you can see there are almost one nan valuse on every two read values,

here are some picture of the settings 
[image]
[image]

TD-er, 2019-10-13:
I have to look at the code to know for sure, but my first impression is that the separate values are being read one after another and not in one cycle.
This may take >1 second to complete and sometimes the scheduled operations are not done exactly on their scheduled interval. (for example if some other operation takes more than desired)
When this happens, the `PLUGIN_READ` may be called before the cycle has finished.
There should be a check for such conditions, but given your observed symptoms there may not be any.

hamed-ta, 2019-10-14:
I'm still getting the same nan values with the interval of 3s.

TD-er, 2019-10-14:
Just curious. What is the use case where you need to check temp/hum values so often?
There must be quite some forces at hand to make noticeable changes in such a short interval where action must be taken in short time.

hamed-ta, 2019-10-14:
for now I just for testing stuff use this interval but there isn't any problem in reading other I2C sensores and also with SHT11 within 1s interval so there should be some optimization issue in the code.

TD-er, 2019-10-14:
I'm not saying the plugin is working correct, I was just curious what use case there could be to really need such frequent updates :)

tonhuisman, 2023-09-03:
This issue is no longer relevant, and can be closed.

TD-er, 2023-09-03:
Not 100% sure this is actually being fixed, but please let me know if it is still an issue.
Will close for now.

closed 2023-09-03 by TD-er