letscontrolit/ESPEasy #2659: [BUG] TM1637 stops working every couple of days

raqua, 2019-10-13:
Build:⋄
20104  - Mega
System Libraries:⋄
ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20191003
Plugins:⋄
46 [Normal]
Build Time:⋄
Oct  3 2019 02:15:29
Binary Filename:⋄
ESP_Easy_mega-20191003_normal_ESP8266_4M1M.bin

### Summarize of the problem/feature request
I am using this with TM1637 as a clock. Plugin name is 'Display - 7-segment display', I guess. It works fine, as expected except it (presumably) crashes every once. Usually it takes couple of days for this to happen. I am running on Wemos D1 mini. 
What happens is that it runs fine and then the display goes dark. But the rest of the system works fine, I can still access ESP Easy GUI and when I reboot the machine, it works fine for some time again. So wiring and HW should be fine, I suspect that the time code is failing, but I do not know how to confirm this. I would be happy to provide more info, when instructed how to do so. 

### Expected behavior
The display should keep showing the current time. 

### Actual behavior
Display goes dark after some time.

### Steps to reproduce
1. Run TM1637, I have no other way to reproduce.

Well, it works for some time and then happens again.

### System configuration
Hardware:
TM1637 and Wemos D1 mini

ESP Easy version: 
2.0 mega-20191003

ESP Easy settings/screenshots: 
[image]

mackowiakp, 2019-10-17:
For what You use 3rd GPIO? Is not necessary for TM1637.

raqua, 2019-10-17:
I know, but I also have MAX7219 and have tested it with this one. It is not possible to remove. Another bug I guess, when I select None from the dropdown and submit, it reverts back to D6.

mackowiakp, 2019-10-18:
I had similar problem with dummy sensor. Such sensor set originally to "single" and then changed to "quad", after reboot revert itself to "single". So parameters intended to appear at position 2,3 and 4 where lost.
As temporary workaround, I cleared flash and configured device manually from beginning (not from config backup file!). Only rules restoring was done from files but via copy/paste., not by downloading..
I fell (not know!) that Your original problem and wrong GUI config are interrelated, because in many cases, clearing flash resolved for me some unusual problems. Including problems with LED displays.
Since that, I can display time/date/temp/hum/any text or numbers concurrently without problems. And the unit does not hangs and even reboots itself. As You can see on attached photo.

[image]

But still it is workaround not bug fixing.

raqua, 2019-10-18:
I guess it is worth a try, thank you.

TD-er, 2019-10-18:
Just as a remark, not really on-topic ;)

The output selector of the dummy is indeed flawed. It does not appear to be working as intended when you have multiple instances of that sensor active.

mackowiakp, 2019-10-18:
Nevertheless, flash clearing helps in many situations. Especially if we are dealing with bugs. This is probably why "blank" firmware was invented, I think....

raqua, 2019-10-25:
I did factory reset and set up everything again from scratch and it did not solve my issue.

mackowiakp, 2019-10-26:
Factory reset or installing "blank" firmware ? Factory reset from ESP menu could not resolve the problem.

raqua, 2019-10-26:
Factory reset from menu. I thought it is the same. 
I can try blank firmware. Is there a difference between flashing a blank firmware and erasin firmware using esptool?  
esptool.py --port /dev/ttyUSB0 erase_flash

mackowiakp, 2019-10-26:
I use (for 4M HW). esptool is a available on my Linux repo directly.
esptool.py --port /dev/ttyUSB1 write_flash --flash_size 4MB --flash_mode dout 0x00000 blank_4MB.bin

If it does not help, I will describe You the environment I use. As I wrote it works without problems, even without self reboots. I use custom FW. The changes are:
- proper display of temperature in range -1C to + 1C
- dim leading zero in time and date.
Tested only with 4 digit display!!! If You want to try, just use attached file.
Disadvantages are:
- time is display from rule, not automatically
- colon is not blinking (lights all the time if "time" is display) 
[firmware.zip](https://github.com/letscontrolit/ESPEasy/files/3774739/firmware.zip)

because display is set to "manual"

TD-er, 2019-10-26:
> Factory reset or installing "blank" firmware ? Factory reset from ESP menu could not resolve the problem.

Well there is a slight difference between the complete flash wipe and the factory reset from ESPeasy.
Factory reset does:
- Perform format of flash filesystem
- Create new settings
- Depending on factory reset settings it may restore some of the old settings still active. (plugin data is not one of them)

These new settings should be created by resetting some of the settings in memory, but I cannot guarantee this will be the same as when starting from scratch.

When wiping the flash, you not only make sure there is no setting left in memory which is not reset, but also you wipe some settings used by the core library.
Most of the time those are WiFi related settings, but I don't really know what else is stored on the flash outside the filesystem space.

The difference between erasing with esptool.py and writing the blank image is that the esptool does use a special command to let the flash chip erase a block.
Writing 0xFF to it (what flashing a blank image does) is also indirectly calling for an erase, but I'm not sure what the state of a block is when it has 0xFF written to it.
It could be this block is then marked as being written on (no longer marked as erased) and thus effectively has slower write access when flashing the firmware to it.

My preferred way would be to use esptool.py for erasing the flash.

raqua, 2019-10-26:
@mackowiakp: I mean, before I flash custom FW on ESP I erase its flash using the command I posted above. Then I use  same command as you do to flash the actual firmware. I was wondering if there is difference between erasing with esptool command I posted or erasing with flashing blank firmware. 
Anyway, I will try both and thank you, if that won't help I will try your custom FW. 

@TD-er: Thanks for explanation, I will try esptool erase first.

mackowiakp, 2019-10-26:
One more important remark. This aspect is often overlooked in DiY projects. nodeMCU is very sensitive to power quality. That's why I always power nodeMCU from 9V, 12V and even 24V.
Of course, there must be a DC step-down voltage converter in series of supply, set to 5,2-5,3V on output (cost less than 1 Euro in a store in my Country).
In addition, I add two 470uF capacitors at the input and output of such a converter.
The next thing that is causing strange behaviours of various devices and not just nodeMCU (and believe me, I have been dealing with this professionally since 1987) is temporary suspension of switching power supplies (e.g. for half a second).
This can occur several times a day or once a week. Because the quality of such power supplies is getting worse (and not better as it might seem),
I have encountered it many times. Also in the case of RPi or nodeMCU.

raqua, 2019-10-26:
I run this on Wemos D1 mini, not NodeMCU. 
But yes, this is something I understand and I use almost exclusively the official Raspberry Pi power source which should be of decent quality. I never had power source issue with these, having at least 12 of them at home running. This one is also on such a PS. 
The negative of that is that the power source costs usualy 2-3x more than the device it is powering .... If anyone has info about good and cheap PS, I am all ears.

mackowiakp, 2019-10-26:
> The negative of that is that the power source costs usualy 2-3x more than the device it is powering 

This - unfortunately - is true. And when I order a housing in the form of a 3D print, the total cost is actually the cost of the housing. And if this housing is to be resistant to UV radiation, then the cost is a real horror.

Grovkillen, 2019-10-26:
A battery is by itself a good PS given that you charge it inbetween. The battery then act as a filter and it will output a really stable voltage.

mackowiakp, 2019-10-26:
Yes it's true. Only as Murphy's Law states: "The battery is always depleted when it is needed most." Well, but it's a more philosophical argument.

raqua, 2019-11-01:
I have erased flash with esptool and then setup all from scratch. Still the issue persists. :(

TD-er, 2019-11-01:
Could you try with [this test build](https://www.dropbox.com/s/7cbfjvg5ytlqczf/ESPEasy_mega-20191028-21-PR_2698.zip?dl=0) ?
It is not specifically addressing issues with regards to this issue, but it does address a number of other generic issues in ESPeasy.
Please let me know if it does improve behavior of this plugin, and/or if any other issues may arise.

raqua, 2019-11-01:
Ok, I upgraded to your build and let you know how it goes.

raqua, 2019-11-02:
It just died again, so no, your build did not solve the issue.

TD-er, 2019-11-02:
OK, that was a bit to be expected, so the problem is very likely in the plugin itself and not the "ESPEasy core".

mackowiakp, 2019-11-02:
This works for me without problems, using FW I previously place in this topic.
Set display to "Manual" mode and use rule below.
Disadvantage - colon is not blinking
Advantage - it possible to toggle between displaying different parameters using push-on button.

```
On System#Boot do
  7dn,----
EndOn

On Clock#Time=All,**:** Do
  Event,UpdateDisplay
Endon

On UpdateDisplay Do
 if %sysyear% > 2000 // check if NTP synced
   7dst,%syshour%,%sysmin%
 Endif
EndOn
```

raqua, 2019-11-02:
mackowiakp: Thanks, but I like my blinking dots! :D 

For now I just made a rule that reboots the machine every hour. It is not a big deal, but if there is something I can do to try to debug this I am open to experiments. I have not really looked at the code, but I do Java coding for living .. I wonder whether there is a way to see some debug logs ... I expect that the plugin itself just crashes, because the rest works ok, so I would expect that there is some log where I can see some sort of exception.

mackowiakp, 2019-11-02:
OK, you like it blinking. But for the test it's worth a try. Little effort.

raqua, 2019-11-02:
Of course I can try that. I have set it up now and will let you know.

raqua, 2019-11-20:
I have uptime of almost 18 days using the setup suggested @mackowiakp, to this one is probably working fine. So the bug would be in the automatic time code I guess.

mackowiakp, 2019-11-20:
No bugs, no software.....

tonhuisman, 2023-09-03:
This seems to have been resolved, so can be closed.

closed 2023-09-03 by TD-er