letscontrolit/ESPEasy #2662: MQTT sending to Openhab broken

waspie, 2019-10-14:
Using the latest (10/03) firmware sending sensor values or publishing via rules (numbers) is broken in openhab
No changes in rules or sensors was done from September 03 release (working) to October 03 release (broken, possibly before this one).

Sending a dallas sensor results in an error in openhab:
2019-10-11 09:15:32.180 [WARN ] [ab.binding.mqtt.generic.ChannelState] - Incoming payload '0 ' not supported by type 'NumberValue'

There seems to be extra space/blank characters after the number that weren't present previously.

Again, this happens with sensors sending temperatures or rules publishing a number to openhab.

TD-er, 2019-10-14:
Possible duplicates: 
- #2660 
- #2657

aleksey-ru52, 2019-10-16:
i've found similar issue
i suppose problem with template in src/_C005.ino:
    case CPLUGIN_PROTOCOL_TEMPLATE:
      {
        event->String1 = F("%sysname%/#");
        event->String2 = F("%sysname%/%tskname%/%valname%");
        break;
      }

this should be preceded with slash:
/%sysname%/#
/%sysname%/%tskname%/%valname%
without slash ESPEasy do not even try to send something to mosquitto, just silent, nothing in debug log either

if i'm right, can i correct it?
i'm new to gitlab and don't know how things should be done right here

TD-er, 2019-10-16:
Nope, that's not related to this one (as far as I know)
Having a leading slash in MQTT topics is a bad idea.
It will create an extra (unneeded) layer in your topics.

If your current setup is based on having a leading slash, then please append one in your settings.
The code you quote is the template or in other words the "default", but nothing is stopping you to add a slash in front.

aleksey-ru52, 2019-10-16:
ok, understand

TD-er, 2019-10-16:
Hmm, you (now?) state it will not even send anything if the leading slash is omitted.
That does sound like a bug.

aleksey-ru52, 2019-10-16:
well, i have mosquitto for testing purposes on OpenWrt, it's a bit old release, i suppose
then i connect to it with this command:
mosquitto_sub -h 192.168.21.253 -v -t /#
it have slash in topic, i can't specify -t # because it generates error in mosquitto_sub
my openhub setup generates and listening channels with leading slash
how can i log/watch command send from ESPEasy? with log level "debug dev" in serial console i see nothing related to mosquitto
ohh... i'm trying to Publish through serial console - nothing happends, jast "command ok", and when i try to execute through web /tools i've got error in serial log:
[E][vfs_api.cpp:27] open(): TmplStd.htm does not start with /
[E][vfs_api.cpp:27] open(): esp.css does not start with /
[E][vfs_api.cpp:27] open(): esp.png does not start with /
2295891 : Command: publish[E][vfs_api.cpp:27] open(): TmplStd.htm does not start with /

TD-er, 2019-10-16:
What is the exact command you gave in ESPeasy?

aleksey-ru52, 2019-10-16:
Publish 2,1 - silent in mosquitto and error in serial console if send from web
Publish /2,1 - Ok

aleksey-ru52, 2019-10-16:
well, a was wrong, sorry
mosquitto_sub -h 192.168.21.253 -v -t ESPEasy32/#
shows readings from thermo sensor
problem with my _own_ openhab setup, i will get it in mind from this time

aleksey-ru52, 2019-10-16:
mosquitto_sub -h 192.168.21.253 -v -t 2/#
serial and web Publish works well too
Publish 2,1 - generates error i mentioned above, but it sends message
from serial console this command executed without error

TD-er, 2019-10-27:
Please test this against tomorrow's build, or try this [test build](https://www.dropbox.com/s/xczig7tzr9mlds1/ESPEasy_mega-20191016-19-PR_2667.zip?dl=0)

If it appears to be fixed, please close the issue.

waspie, 2019-10-28:
> Please test this against tomorrow's build, or try this [test build](https://www.dropbox.com/s/xczig7tzr9mlds1/ESPEasy_mega-20191016-19-PR_2667.zip?dl=0)
> 
> If it appears to be fixed, please close the issue.

This release (10/28) has resolved this issue for me

waspie, 2019-10-28:
I spoke too soon... I wonder if you'll see this if it is closed....
The sensors sending data are fine but the rules part is still broken.

Here's a rule:
`
on ping do
Publish /%sysname%/wd/ur,1
endon
`

The result in mosquitto is:
'
19-10-28 13:20:24.030 /utilityRoom/cmd Event,ping
2019-10-28 13:20:24.578 Publish /utilityRoom/wd/ur   <----------------
'

It is sending the entire line including "Publish" ??????

TD-er, 2019-10-28:
Hmm that's probably a bug I introduced.
What happens if you place them on separate lines like this:

```
on ping do 
  Publish /%sysname%/wd/ur,1 
endon
```

waspie, 2019-10-28:
It is on separate lines the message just didn't come out right

TD-er, 2019-10-30:
Can you try this test build: [ESPEasy_mega-20191028-21-PR_2698.zip](https://www.dropbox.com/s/7cbfjvg5ytlqczf/ESPEasy_mega-20191028-21-PR_2698.zip?dl=0)
Please let me know if it does work and it would be great if you could do some more thorough testing so I can merge it for tomorrow's build.

waspie, 2019-10-30:
I'll make sure the rules work right this time before reporting back.

waspie, 2019-10-30:
I send sensor data via MQTT and it is working.
I contact the node via MQTT (not import) and a rule catches it and send a reply and it is working
Sensors gather temperature data and a rule runs a formula on it and saves it to a dummy item (rule)
and that rule sends that dummy data via MQTT and it is working. 
In all my use cases it is working.

closed 2019-10-30 by waspie

TD-er, 2019-10-30:
Thanks for testing and letting me know.