letscontrolit/ESPEasy #2663: in Dust.ino plugin which is related to Sharp GP2Y10 dust sensor, the output value is 25 times bigger than actual adc value

hamed-ta, 2019-10-14:
here is the Read code part of the plugin, why sums analogRead 25 times ? for averaging ? so 
doesn't need value /=25; after loop ?
```
    case PLUGIN_READ:
      {
        Plugin_GP2Y10_LED_Pin = CONFIG_PIN1;
        noInterrupts();
        byte x;
        int value;
        value = 0;
        for (x = 0; x < 25; x++)
        {
          digitalWrite(Plugin_GP2Y10_LED_Pin, LOW);
          delayMicroseconds(280);
          value = value + analogRead(A0);
          delayMicroseconds(40);
          digitalWrite(Plugin_GP2Y10_LED_Pin, HIGH);
          delayMicroseconds(9680);
        }
        interrupts();
        UserVar[event->BaseVarIndex] = (float)value;
        String log = F("GPY  : Dust value: ");
        log += value;
        addLog(LOG_LEVEL_INFO, log);
        success = true;
        break;
      }
```

TD-er, 2019-10-14:
I have to take a proper look at it after a bit more sleep.
https://github.com/letscontrolit/ESPEasy/blob/b6507784828d2a92785f2027658e6d5df277ead2/src/_P018_Dust.ino#L65-L88

I do agree it is outputting a value which does look oddly high at least.
But I don't know what the relation should be between the analog value and the output value.
The analogRead function does output a value from 0 ... 1023 for ESP8266 and 0 ... 4095 for ESP32.

Anyway even if the value should be 25x as high, it should be the same for ESP32 and ESP8266.

Another thing which I don't like of this code is that it is disabling interrupts for over 25 msec.
That's a big no-no for me.

hamed-ta, 2019-10-16:
acording to this article and as esp32 adc sample rate is about 2 Msps I would suggest something like this only for ESP32

`
case PLUGIN_READ: 
   { 
     Plugin_GP2Y10_LED_Pin = CONFIG_PIN1; 
     noInterrupts(); 
     byte x; 
     int value; 
     value = 0; 

     digitalWrite(Plugin_GP2Y10_LED_Pin, LOW); 
     delayMicroseconds(268); //280us - (25 * 0.5us)

     for (x = 0; x < 25; x++) 
           value = value + analogRead(A0); 

      digitalWrite(Plugin_GP2Y10_LED_Pin, HIGH); 

     interrupts(); 

     value /= 25;
     UserVar[event->BaseVarIndex] = (float)value; 
     String log = F("GPY  : Dust value: "); 
     log += value; 
     addLog(LOG_LEVEL_INFO, log); 
     success = true; 
     break; 
   } 

`

it's fast enough while still uses avraging.

TD-er, 2019-10-16:
What time is taken between disabling and enabling the interrupts?
Also what purpose has the delay which is apparently derived from 25*0.5 us?
If it is based on the same 25 as the number of samples, shouldn't it be in the loop itself?
And taking an average over N samples in such a short interval, what is it good for?

I don't know the Sharp sensor, so I don't know why it does need to take a sum or average over N samples.
The original loop does trigger a GPIO pin followed by an analog read, but your change no longer does this.
Also it did use a quite specific delay between reads, which is also missing in your proposed code.

hamed-ta, 2019-10-16:
> What time is taken between disabling and enabling the interrupts?

maybe something near 300us

> Also what purpose has the delay which is apparently derived from 25*0.5 us?

the sensor output will be in peak amplitude at 280us after turning on the inside LED so as 25 adc samples take maximun 25us, so 12.5us befor 280us would be the best time to start sampling and it will lasted to about 12.5 us after that peak so in this way not only we can do avraging but also there wouldn't be that much delay between disabling and enabling interrupts

> If it is based on the same 25 as the number of samples, shouldn't it be in the loop itself?

I used 25 samples just for averaging.

> And taking an average over N samples in such a short interval, what is it good for?

averaging

> I don't know the Sharp sensor, so I don't know why it does need to take a sum or average over N samples.

> The original loop does trigger a GPIO pin followed by an analog read, but your change no longer does this.

the original approach could be better if we add value /= 25 after loop but as you said it's taking time just too much so I offer this approach and will test on my device.  

> Also it did use a quite specific delay between reads, which is also missing in your proposed code.

it's needed if we want to do averaging with different sensor measurement but this code do averaging on one sensore measurement

TD-er, 2019-10-16:
If you're looking for the peak, is averaging the best option then?

commit 99469018 (2019-11-02): [MQTT] Fix Publish command (#2662)

The parsing was using the wrong indices to determine topic and value.
Fixes: #2663
closed 2019-11-02 by TD-er