letscontrolit/ESPEasy #2666: PN25F16 flash erase and write error

joychase, 2019-10-15:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. ESP8266 with PN25F16 flash
2. erase with esptool
3. 

### System configuration

Hardware:
ESP8266 with PN25F16 flash

ESP Easy version: 
ESPEasy_mega-20190928

ESP Easy settings/screenshots: 

### Rules or log data
######2019-10-15######
#######0.04.007#######
######FLASH INFO######
BIN file: blank_2MB.bin
COM port: (COM3) Prolific USB-to-Serial Comm Port (Port_#0002.Hub_#0002)
Baud rate: 115200
######POST FLASH######
No post flash information entered...
######FLASH LOG######
[esptool.exe -vv -cd nodemcu -cb 115200 -cp COM3 -ca 0x00000 -cf "F:\ESPEasy_mega-20190928\bin\blank_2MB.bin"]
[2019/10/15 12:25:43] esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
[2019/10/15 12:25:43] 	setting board to nodemcu
[2019/10/15 12:25:43] 	setting baudrate from 115200 to 115200
[2019/10/15 12:25:43] 	setting port from  to COM3
[2019/10/15 12:25:43] 	setting address from 0x00000000 to 0x00000000
[2019/10/15 12:25:43] 	espcomm_upload_file
[2019/10/15 12:25:43] 	espcomm_upload_mem
[2019/10/15 12:25:43] 	setting serial port timeouts to 1000 ms
[2019/10/15 12:25:43] opening bootloader
[... 152 lines cut ...]
[2019/10/15 12:25:43] setting serial port timeouts to 1000 ms
[2019/10/15 12:25:43] flush complete
[2019/10/15 12:25:43] espcomm_send_command: sending command header
[2019/10/15 12:25:43] espcomm_send_command: sending command payload
[2019/10/15 12:25:43] read 0, requested 1
[2019/10/15 12:25:43] trying to connect
[2019/10/15 12:25:43] flush start
[2019/10/15 12:25:43] setting serial port timeouts to 1 ms
[2019/10/15 12:25:43] setting serial port timeouts to 1000 ms
[2019/10/15 12:25:43] flush complete
[2019/10/15 12:25:43] espcomm_send_command: sending command header
[2019/10/15 12:25:43] espcomm_send_command: sending command payload
[2019/10/15 12:25:43] read 0, requested 1
[2019/10/15 12:25:43] trying to connect
[2019/10/15 12:25:43] flush start
[2019/10/15 12:25:43] setting serial port timeouts to 1 ms
[2019/10/15 12:25:43] setting serial port timeouts to 1000 ms
[2019/10/15 12:25:43] flush complete
[2019/10/15 12:25:43] espcomm_send_command: sending command header
[2019/10/15 12:25:43] espcomm_send_command: sending command payload
[2019/10/15 12:25:43] read 0, requested 1
[2019/10/15 12:25:43] warning: espcomm_sync failed
[2019/10/15 12:25:43] error: espcomm_open failed
[2019/10/15 12:25:43] error: espcomm_upload_mem failed
[2019-10-15 12:25:43] STOPPED due to 2 errors! (try reset on the unit, then start a new flash attempt)

uzi18, 2019-10-15:
try force dio 40mhz spi mode

TD-er, 2019-10-15:
Instead of writing the blank image, you could also try to call the erase function from esptool.py.

And indeed, what @uzi18 suggested, lower the SPI frequency.
The datasheet mentions:

- Normal Read (Serial): 50MHz clock rate
- Fast Read (Serial): 108MHz clock rate
- Dual/Quad (Multi-I/O) Read: 108MHz clock rate

Not sure what the difference is between "normal" and "fast" or what esptool.py does try to use.

joychase, 2019-10-27:
> Instead of writing the blank image, you could also try to call the erase function from esptool.py.
>
> And indeed, what @uzi18 suggested, lower the SPI frequency.
> The datasheet mentions:
>
> * Normal Read [...]

Thanks for replying. I solved the problem by replacing the SPI chip with a compatible Winbond chip.

TD-er, 2019-10-27:
Hmm that's also a proper fix, although not one everyone could pull off :)
So I guess it is considered to be solved.

closed 2019-10-27 by TD-er