letscontrolit/ESPEasy #2671: Feature request: Energy Meter SDM630 add reading more registers

seamoor, 2019-10-19:
Hi, 

I'm creating this issue at the request of @TD-er from this closed issue: [link](https://github.com/letscontrolit/ESPEasy/issues/205)

SDM_Energy_Meter plugin supports only 1 phase power meters like SDM120, 220 etc
I personally have SDM630, this is 3 phase power meter. Connected up and running.
On ESP view:

[image]

We can choose only generic options like Voltage, Current etc but according to SDM.h by Reaper7 this power meter has 3 Voltage registers, 3 Current registers etc:

[image]

Please add other registers that we can select them and get data from them.

Sasch600xt, 2019-10-19:
hello :)
may i ask where i find this plugin ?
Is ist available in "normal" build ?

Thank you

seamoor, 2019-11-09:
Hi,

[image]

I believe so, i don't remember anything about uploading any custom version.

Just normal 4M or 1M version.

Maikel-K, 2020-11-14:
I tested the plugin and it's running fine.
I have the SDM630 so i selected it from the dropdown, but i cant choose anything other then the 1 phase values.

Any news regarding this?

TD-er, 2020-11-14:
Can you select  a specific phase?
If so, then you can have multiple instances of the plugin and they will share the same interface to the Eastron meter.

Maikel-K, 2020-11-14:
No you can only select 1 phase[image]

TD-er, 2020-11-14:
Hmm apart from the missing selections, I also notice you do have lots of messages with failed checksum.
How is the meter connected?
SW serial?

Maikel-K, 2020-11-14:
I did not yet connect the resistor as I don't have one at the moment. The cable is 2 meters at the moment. I use a RS485 to TTL adapter connected to the RX/TX inputs of a Wemos D1 mini. I also had the data acquisitions at 1 second.

Nikolay-Ch, 2020-11-19:
Could it be more usefull if we have a modbus-read-register-plugin to read any Value from any Modbus Device?
We need to preset:
DeviceId - Modbus device Id
FunctionName - 3 or 4 (byte) - to read different registers
ParameterName - "Voltage", "Current", etc...
MeasuredUnits - "Volts", "Amps", etc
LengthOfData - value length (count of registers, or bytes)
RegisterAddress - address of the register in device
ValueFormat - Float, Hex, UnsignedInt, etc...

Then we can read this registers and format values depending on ValueFormat value.
I created similar .Net component and use it in my private-project...

tomcatxx, 2021-01-30:
Hi I added all Registers from Eastron SDM120C SDM120/SDM120CT/220/230/630/72D & also DDM18SD to the Plugin. Actually I'm testing it with a D1 Mini NodeMcu ESP8266-12F. Its working so far. Yesterday I started with getting 4 Values from my SDM630 pushing them to mqtt with an intervall of 1 second. I dont know if this is to fast but the ESP8266 restartes from alone after some time. Now I'm testing it with 2 Second Intervall. Test with an ESP32 will follow..
If anyone want to join me with testing just contact me.

TD-er, 2021-01-30:
Please check the timing stats page.
Currently the Modbus RTU implementation is not really optimal as it is waiting for the reply.
Not sure how quickly the Eastron meters react, but I've seen some other Modbus devices take roughly 200 msec to read a register.
So for now you may want to add a call to `delay(1)` after each register read as my assumption is that it may restart due to a watchdog intervention.

tomcatxx, 2021-01-30:
[image]
The highest delay I saw was 150ms.
I also tryed using the whole new code from here https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.cpp but it ends up in 350ms and more so I switched back....
So for now I added the delay(1) and reduced the MAX_MILLIS_TO_WAIT to 200 for testing.

I hope I did it right cause my c++ skills are nearly 0 and I dont unterstand all code..

    case PLUGIN_READ:
      {
        if (Plugin_078_init)
        {
          int model = P078_MODEL;
          byte dev_id = P078_DEV_ID;
          UserVar[event->BaseVarIndex]     = p078_readVal(P078_QUERY1, dev_id, model);
          delay(1);
          UserVar[event->BaseVarIndex + 1] = p078_readVal(P078_QUERY2, dev_id, model);
          delay(1);
          UserVar[event->BaseVarIndex + 2] = p078_readVal(P078_QUERY3, dev_id, model);
          delay(1);
          UserVar[event->BaseVarIndex + 3] = p078_readVal(P078_QUERY4, dev_id, model);
          delay(1);
          success = true;
          break;
        }
        break;
      }

TD-er, 2021-01-30:
Isn't it easier to just add this `delay` in the `p078_readVal` function? ;)

tomcatxx, 2021-01-30:
You are right :D

tomcatxx, 2021-01-31:
Well the delay(1) does not realy help. After I changed the Baud Rate to 38400 I had no reboots for 15h. But now I have strange behavior. The reaction time of loop, whatever it means, goes high sometimes op to 15000ms most time ~6000ms. And it stays high even after I disables all controlers and devices. Any idea what could be the reason for this?

TD-er, 2021-01-31:
Such high loop times are often an indication of some network reconnect issue.
So the process is waiting and retrying, but at least it is calling a delay, or else the watchdog would have kicked in.

tomcatxx, 2021-02-01:
Ok I understand. Well it runs now stable for 1,5 days on the ESP8266 . So I started to test ESP32. I used RX0 on GPIO3 and TX0 on GPIO1 and selected HW Serial 0 cause software serial is missing. But I dont get any connection to the RS-485 MAX485 TTL 5V UART 3,3V Level Konverter. Any suggestions ?

TD-er, 2021-02-01:
Those are the same pins as the standard (Serial 0) port.
Did you uncheck "Enable Serial port" on Tools -> Advanced?

tomcatxx, 2021-02-01:
I think so yes. 
https://microcontrollerslab.com/wp-content/uploads/2019/02/ESP32-pinout-mapping.png
Ifs also defined in ESPeasySerial.h
#ifndef ESP32_SER0_TX
  # define ESP32_SER0_TX 1
#endif // ifndef ESP32_SER0_TX
#ifndef ESP32_SER0_RX
  # define ESP32_SER0_RX 3
#endif // ifndef ESP32_SER0_RX
Enable Serial Port is unchecked yes. If it is checked the two pins are blocked to use under devices.

TD-er, 2021-02-01:
You you also need a pin for DE/RE?
What pin do you use there?

Can you show a picture of how you configured it in ESPEasy? (screenshot of the serial config of the task)

tomcatxx, 2021-02-01:
[image]
I just need RX and TX with this Konverter: https://www.ebay.de/itm/RS-485-MAX485-TTL-5V-UART-3-3V-Level-Konverter-auto-Flusskontrolle-Flow-Control/284124664509?ssPageName=STRK%3AMEBIDX%3AIT&var=585614016895&_trksid=p2060353.m2749.l2649
On the ESP8266 with SoftwareSerial it worked.

TD-er, 2021-02-01:
Have you tried setting the serial port to HW Serial1?
On ESP32 you can set the pins you like. (don't use GPIO 1 and 3)
Just to be sure Serial0 isn't used for something else.

tomcatxx, 2021-02-01:
I tryed 18&19. Then I get some broken communication.
One value after changing settings and then nothing.
[image]
[image]

TD-er, 2021-02-01:
Baud rate is correct?

tomcatxx, 2021-02-01:
Yes baut rate is correct.
Modbusadress too.
TX from ESP32 goes to RX on the Konverter and RX from ESP32 to TX from the Konverter.
I have an 120 Ohm Resistor between A and B on the SDM630 side.
With this configuration and the ES8266 it was working.

TD-er, 2021-02-02:
Do you have SPI enabled in the settings?
See [ESP32 pinout reference GPIO](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/)

[image]

tomcatxx, 2021-02-02:
I tried 16 and 17.
[image]
I get one respond and after this nothing like with 18&19. SPI was/is disabled.

TD-er, 2021-02-02:
Just to be sure, are you running the existing code, or have you changed code?

tomcatxx, 2021-02-02:
I first tryed my modified plugin but after this also the one compiled by you. It doesnt matter. With ESP32 its not working. If I switch back to ES8266 its working again.

TD-er, 2021-02-02:
Do you happen to know what other chip is present on that MAX485 board you have?
Is it an isolator? 
How does it toggle the DE/RE pin?

Can you check the exact MAX485 model number used? Not all MAX485 chips are meant for 3.3V, but should work... most of the time, but with a very tight margin.

tomcatxx, 2021-02-02:
It is this converter Board: 
[image]
On the small ship in the middle is printed MAX485 ESA +2010
On the bigger below 74HC04D XBLIC46 UNG1946G
According to the discription it can be used with 3,3V (what im doing and 5V).
https://www.ebay.de/itm/RS-485-MAX485-TTL-5V-UART-3-3V-Level-Konverter-auto-Flusskontrolle-Flow-Control/284124664509?ssPageName=STRK%3AMEBIDX%3AIT&var=585614016895&_trksid=p2060353.m2749.l2649

For comprehension its working with these settings on ESP8266:
[image]
[image]
[image]

TD-er, 2021-02-02:
I think you probably have slightly lower "3V3" on this ESP32 board.
The HEX inverter chip also allows to be powered via 5V, but I have to check the datasheet to see whether the HEX inverter outputs levels based on the supplied voltage, or expects a pull-up resistor for the "high" level.

So what you can try first, is to attach a pull-up resistor between 3V3 and the "TX" of the MAX485 board (the RX on the ESP) to see if that makes a difference.
The same for the other pin as the HEX inverter may also be picky here.

If that doesn't help, I suggest to connect the board to the 5V for powering it.
But maybe you should also consider using a level converter when powering it at 5V.
I know the ESP8266 is not officially 5V tolerant, but a lot of reports state it is. Just don't connect it to pins that are also connected to some other circuitry like the USB to UART chip as that one may not be 5V tolerant.
I don't know if the ESP32 is 5V tolerant, so to be sure, you should consider a level converter.
Also the "high" level of the ESP32 may not be high enough for the HEX inverter chip to be seen as high when it is powered at 5V.

tomcatxx, 2021-02-02:
I put a 10k Ohm resistor between 3V3 and the TX of the MAX485. Does not work. Than I added a second 10k Ohm resisto between 3V3 and the RX of the MAX485. Still does not work.

TD-er, 2021-02-02:
If you connect the MAX485 to 5V, then I should not use pull-up resistors to 3v3.
What you can do, is create some protection for the ESP32 by making a voltage divider using 4k7 and 10k to at least protect the RX of the ESP.
But like I said, not sure if the high of the ESP's TX is high enough for the board to consider it a high level.

Another thing I just realize...
The HEX inverter chip is meant to protect the circuits on both ends and also brush up the signal a bit.
However it does invert the logic levels.
The chip used has enough inverter parts to double invert it, but then that should be done.
So can you see on the traces how this 74HC04D is wired? Is one output fed into an input again?
I guess it should not matter as the ESP8266 is working well with it, but just something to check here.

uzi18, 2021-02-02:
If it work on ESP8266 it should also on ESP32

tomcatxx, 2021-02-02:
Pin 4 and 5 of the 74HC04D are connected to each other and 4 is connected to RX.
However the MAX485 board seams to be broken so I ordered a new one and also a level converter.

TD-er, 2021-02-02:
> If it work on ESP8266 it should also on ESP32, without voodoo magic

Well the voltage (3v3) may be a bit tricky with a MAX485, so I guess that may be the main issue here.
That may already be an issue among different ESP8266 boards.

uzi18, 2021-02-02:
Should try external 3V3 for converter

uzi18, 2021-02-02:
It says - supply voltage 5V but RX/TX are 3/5V tollerant

tomcatxx, 2021-02-02:
I will try external 3V when I get new MAX485 board :)

uzi18, 2021-02-02:
you got there also LEDs for indication if there is any communication, does they work?

tomcatxx, 2021-02-02:
With ES8266 they worked. With ESP32 they was flickering sometime but not the same as with ES8266.
Now after the borad is broken they shine both all the time if GND and 3.3V is connected.

tomcatxx, 2021-02-04:
Well new MAX485 board arrived. I also bought secound ESP32 to be sure. But the behavior is the same even with extern 3.3V.
ESP32 not working getting one value as shown on screens above. Nodemcu ESP8266 is working.

ilexpl01, 2021-09-12:
> No you can only select 1 phase[image]

Hi. Got the same problem. SDM630 meter. Newest ESP_Easy_mega_20210802_energy_ESP8266_4M1M loaded on D1 mini.

Unable to select phase number. The connection to the meter is running fine, Can anyone give a hint ?

TD-er, 2021-09-12:
Have you checked this pending PR? https://github.com/letscontrolit/ESPEasy/pull/3504

njordan77, 2022-01-02:
hi, can someone tell me how to get ESPeasy to provide parameters per phase (1,2,3). I see here screenshots showing that its implemented a while ago, but flashing the latest 24.12.21 binarry of "energy" does again not provide e.g. W per phase.

Any idea what i'm doing wrong? Thanks Norbert

tonhuisman, 2022-01-03:
I think the screenshots are made using the code from the Pull request that is linked here, but that's not merged, so it is not (yet) available in the release.
You could download a build from the [Github Actions page for the last commit](https://github.com/letscontrolit/ESPEasy/actions/runs/1373710511) to see if it fits.

helbgd, 2022-07-06:
@TD-er can you please check if you can push your updates to mainstream letscontrolit release ?
addidtionally if it is anyhow possible add the following missing register 
#define SDM_NET_KWH                                   0x018C
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L202

Thanks for your effort's meanwhile I did manually compile and integrate your changes into latest release for esp32
it is working like expected.

Thanks for your efforts and the work you did with this.

tonhuisman, 2023-09-01:
This has been fixed recently, so can be closed.

closed 2023-09-01 by TD-er