letscontrolit/ESPEasy #2673: Invalid framelength for PMS7003 in logs

mrdc, 2019-10-19:
Hello,

I've recently connected PMS7003 to my NodeMCU with ESP Easy mega-20191003.
And I see in logs `PMSx003 : invalid framelength`:

> 3362516: PMSx003 : invalid framelength - 156
> 3363061: WD : Uptime 56 ConnectFailures 0 FreeMem 19496 WiFiStatus 3
> 3393061: WD : Uptime 57 ConnectFailures 0 FreeMem 19496 WiFiStatus 3
> 3395125: PMSx003 : invalid framelength - 32772
> 3423061: WD : Uptime 57 ConnectFailures 0 FreeMem 19496 WiFiStatus 3
> 3430117: PMSx003 : invalid framelength - 142
> 3453061: WD : Uptime 58 ConnectFailures 0 FreeMem 19472 WiFiStatus 3
Measurements are updating.

Is it normal behaviour?

TD-er, 2019-10-19:
I have not such a module running here, but if it is happening every now and then then I would not be too concerned about it.
If it does cause faulty readings and happening in a significant number of samples, then it does sound like either a bug or too much noise on the line between sensor and ESP.

Do you use short cables?
You could test to enable a pull-up resistor for the RX pin on the ESP side, to make it less susceptible to noise.

mrdc, 2019-10-19:
> Do you use short cables?

Not so short.

> You could test to enable a pull-up resistor for the RX pin on the ESP side, to make it less susceptible to noise.

Will try 10K tomorrow.

TD-er, 2019-10-19:
ESPEasy also can set the internal pull-up resistors (not all , GPIO-15 and 16 may not work like that)
What pins do you use for this plugin?
See [Documentation - Best pins to use on ESP8266](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)

chemmex, 2019-10-20:
I would vote for improper pin selection. Or a faulty sensor. My preference is to connect the PMS to the HW UART (provided it is disabled in the Advanced settings). If your board allows access to the HW UART, of course.

mrdc, 2019-10-20:
> ESPEasy also can set the internal pull-up resistors

Do you mean GPIO boot states?
[image]

> I would vote for improper pin selection.

I'm using Software Serial:
[image]

> (provided it is disabled for logging in the Advanced settings)

[image]

TD-er, 2019-10-20:
Serial port should be disabled if you're using either _HW Serial 0_ or _HW Serial0 swapped_. (not SW serial, or any of the pins of HW serial))
The serial port is not only used for logging. Anything sent to it is also being interpreted as commands.

The pins you are using already have a pull-up resistor present, or else the board would not even boot.  So no need to add extra pull-up resistors then.  Most likely the resistors present are 12k.

How long are the cables from the sensor to your board
How often are these warnings reported? Any single measurement?
Does the sensor update its value to a believable value every now and then?

mrdc, 2019-10-20:
> How long are the cables from the sensor to your board

30cm

> How often are these warnings reported? Any single measurement?

Periodically, not every single measurement.

> Does the sensor update its value to a believable value every now and then?

Yes, readings look pretty adequate: no random values etc.

TD-er, 2019-10-21:
30 cm  is not really that long. But you may want to shield the wires. Either by twisting them (not only rx/tx, but also the GND) or maybe add some ferrite clamp or aluminium foil.
Just make sure you don't make things worse by either causing a short circuit or limiting the WiFi sensitivity.
An occasional read error is not really a problem. As long as it doesn't happen too often (what is too often depends on your requirements)

mrdc, 2019-10-22:
> An occasional read error is not really a problem. As long as it doesn't happen too often (what is too often depends on your requirements)

Ok, I will keep an eye on it.

mrdc, 2019-10-23:
@chemmex 

> I would vote for improper pin selection. Or a faulty sensor. My preference is to connect the PMS to the HW UART

Using HW Serial solves the issue.
[image]

TD-er, 2019-10-23:
More like a timing issue. Software Serial is not very stable on receiving lots of data, or on nodes that also have a lot of other blocking tasks. (high values in the timing stats page)

Just use it on short transfers (or blocking transfers) and only use it on plugins that are allowed to miss a sample every now and then.

mrdc, 2019-10-23:
> More like a timing issue. Software Serial is not very stable on receiving lots of data, or on nodes that also have a lot of other blocking tasks. (high values in the timing stats page)

Thanks, it's clear now.

closed 2019-10-23 by mrdc