letscontrolit/ESPEasy #2674: Only one bin-file in ESPEasy-mega\dist\bin?!

Sinic98, 2019-10-21:
Why is there only one bin-File plus four blank bin files? The readme says there should be many different bin-files for different flash sizes.

Grovkillen, 2019-10-21:
Those are the blank files + the two way OTA. These aren't needed to be compiled on each release thus they are already present.

When we make a new release the bin files are compiled and together with these four files they are packed into a release zip file and updated to our release page.

You can download them from there.

closed 2019-10-21 by Grovkillen

Sinic98, 2019-10-21:
So where exactly can I find the binary files which I really need?

Grovkillen, 2019-10-21:
https://github.com/letscontrolit/ESPEasy/releases

Sinic98, 2019-10-21:
Sorry, I must be too stupid. I know that page but there I could only find these binarys after I unzipped.

[image]

Grovkillen, 2019-10-21:
Only download the ones that have a blue colored title. Sometimes our automatic release robot fail, this is the case of those release with no release page/blue title.

Grovkillen, 2019-10-21:
[image]

Sinic98, 2019-10-21:
OMG, how could I be so blind... Thank you so much!