letscontrolit/ESPEasy #2675: Changing Dummy Device Output Data Type changes all Dummy Devices

Wiki591, 2019-10-21:
I have a device with 5 Dummy devices set. If I change one of them to a different outpu data type, all of the dummy devices are changed to the same output data devices, idependently from their former settings.

PPatou, 2019-10-26:
Hello,
Same problem when using global sync over UDP if you enable the task in the receiving unit the data disappears

TD-er, 2019-10-26:
Yep, the output type of the dummy is a bit buggy and also shared among all instances of the plugin.
That was an unforeseen consequence.
Not sure yet how it can be done differently, since the plugin definition is changed in memory to achieve this.
It does have an effect on a lot of places in the code if it will be changed how to look for a plugin definition (nr. of output values, sensor type, etc)

In short: No idea yet how to fix it other than just keeping it the same for all instances of the plugin.
When syncing via p2p, this limitation has even bigger impact. (multiple nodes with the same plugin should have the same output type)

Wiki591, 2019-10-27:
> 
> When syncing via p2p, this limitation has even bigger impact. (multiple nodes with the same plugin should have the same output type)

Worse: If you need a dummy device with two nr. of values (i.e. Temp/Hum) for publishing data to Domoticz, you have to configure it at least to double, otherwise with this bug you won't get the data into Domoticz. Consequently you are limited with that specific node to a nr. of two output values over all dummy devices. In this case the current releases are almost unusable, I went back to ESP_Easy_mega-20190225_normal_ESP8266_4M.bin and its working as I need it.

TD-er, 2019-10-27:
But then you cannot set the number of outputs to the Dummy device, right?
Domoticz is a bit strange in handling the data, but as far as I know, you could send more data than Domoticz expects.
On the other hand, Domoticz is rather specific about the order of the values it receives.
For example with a configured temp/baro/hum sensor, it does expect the humidity on the 3rd position, but when it is configured as a temp/hum sensor, it needs the humidity on the 2nd position.

If you set the Dummy device to 4 output values and do not occupy the unused fields, it can still receive the data just fine and you do not limit the Dummy device for other instances of the same plugin.

Misiu, 2019-11-05:
I configured the dummy device to output a single value, but after rebooting the device shows 4 values, when I edit the device and hit submit 3 unneeded values disappear:
[image]

Misiu, 2019-11-15:
Any updates on this? Maybe the new UI will fix this? If yes, then I can wait :)

TD-er, 2019-11-15:
Nope, no update yet and it must be fixed via another change.
It has nothing to do with the UI.
The real issue is in how the "changeable outputs" are implemented.
I change the plugin definition in memory, but it should have been a change on task definition.
That does require quite some changes to get it fixed, so it may take some time to get it fixed.

To be honest, given what also has to be done, I don't think I will have it fixed before January.

Misiu, 2019-11-15:
@TD-er thanks for the honest reply :)

Misiu, 2020-07-03:
@TD-er any updates on this? I know this is an old one.

TD-er, 2020-07-03:
Nope not yet.
As you may have seen, I've been working on other parts of the code lately.
Mainly on instability caused by running out of memory.

Like I said, a proper fix for this may require quite a big code change.

Misiu, 2020-07-03:
@TD-er I'm monitoring the changes and the great job you are doing. Optimizations are great, especially with memory, wifi and of course move from ino to cpp/h is a big move forward 🎉💪

TD-er, 2020-07-03:
Yes, those moves from .ino to .cpp/h are really hard needed.
The problem with the .ino files is that you don't have a guarantee the used defines in the code are actually set when compiling it.
This because the .ino files are concatenated into one .cpp file and thus it may depend on the build in what order things are compiled.

So I really want to make things to always fail the same way.... ehhh lead to a predictable build :)

PR #3265 (2020-09-24): Use task specific valueCount and VType instead of shared in plugin
Lots of reported issues regarding strange issues when using multiple instances of the same plugin which allows to change its output variables and value count.

Possible affected plugins:

- P026 Sysinfo
- P033 Dummy
- P052 SenseAir
- P078 Eastron
- P082 GPS
- P085 AcuDC243

N.B. this PR is not yet finished.
Still need to have a look at usage of `sensorTypeHelper_loadOutputSelector` and related functions.

Fixes: #3184 
Fixes: #2675
Fixes: #2626
Fixes: #2600
Fixes: #3045 
Fixes: #3061
closed 2020-09-24 by TD-er