letscontrolit/ESPEasy #2676: Bug in determine index of Plugin_ptr

TD-er, 2019-10-21:
Found a very, very serious bug in the __Plugin.ino code and it looks like I introduced it around 20180705. See: https://github.com/letscontrolit/ESPEasy/commit/2fde3d28a8af4ff044ae123ca820d312b7e59a90

Problem is in the handling of some functions like PLUGIN_xxx (WRITE/READ/etc)
It is then using the wrong index to do stuff, which may yield strange results on builds with not all plugins included (mainly test and custom)

Impact is a bit hard to predict, but I guess it may lead to strange crashes.
That's also around the time when WDT reboots started to be reported.
So when fixed, it may hopefully fix a number of issues.

Plugin_ptr will only use indices based on the number of plugins included in the build.
So it must be addressed using an index based on the DeviceIndex (not sure if that's the right index) and not on the plugin ID number (e.g. 004 for Dallas sensor)

ghtester, 2019-10-22:
So perhaps that's why I experienced a device change from IRTX to Switch recently after replacing the ESP_Easy firmware with custom build without IRTX plugin and keeping the config... :-)

TD-er, 2019-10-22:
The bug appears to be quite old, so not sure what big jump you made in versions?

ghtester, 2019-10-22:
Well, the config was very old and a many firmware upgrades performed during last 2 years on that node... In past both IRRX and IRTX were in std or test official builds but it's not the case yet so I had to start with custom builds using Vagrant }many thanks for preparing that great environment!). As I need a many plugins and the current IRRX and IRTX plugins are so big, I disabled the IRTX as IRRX is more important for me but the config remained the same when I uploaded the customized firmware to node. Then the IRTX (device 12) changed to Switch.

TD-er, 2019-10-23:
Yep, if a plugin cannot be found, it will set it to the first one.
That's also a bug, which I may be able to fix with these changes.

The size of IR plugins is what made me decide to move them to a separate build.
But like you also found out, you cannot have a 1-build-fits-all, so that's why I was looking into the Vagrant setup.

TD-er, 2019-10-23:
Also fixing a long lasting annoyance along the way.
If a task is set to a pluginID not included in the current build, it was showing "Switch" as plugin.
Also it was not really clear what would happen if you then edit the page.

This is how it will look after this has been fixed:

[image]

The plugin ID will be shown and the name. Enabled state will be shown as disabled and the rest will be left empty.
While typing this, I realize the red button should show "Add" or maybe some other text indicating you will create a new entry when pressing the button.

PR #2667 (2019-10-27): [Rules] Improve rules matching (#2660)
Detecting time compare and value compare could give false matches in some cases.
This PR also incorporates a number of other changes:
- Adds option to change the remote unit on remote fed plugins via p2p. (#2679)
- Cleanup of use of taskIndex, DeviceIndex (#2676)
- Disable and mark configured plugins not included in the build (e.g. when switching builds with other plugins included) (related to #2676)
- Fix build issues with PlatformIO using Python3
- Perform first read of ADC at init, before starting WiFi (#2646)

Fixes: #2683 - Duplicated section in `pla [... cut]
closed 2019-10-27 by TD-er