letscontrolit/ESPEasy #2678: Ability to reset settings to default or enter AP mode using a physical button

Misiu, 2019-10-22:
This is a feature request more than a bug.
I'm preparing four temperature sensors that I want to mount in my parent's house. I've set up everything at my place and now I want to move sensors to their place.
When setting everything up I've entered my SSID and password, now when I attach the devices in my parent's house they won't connect (different SSID).
I've searched the forum and Github and I didn't find a way to manually enter AP mode (to be able to connect via my phone and change WiFi settings) or to do a factory reset.
If this is currently possible please let me know how to do that.
My hardware is: Wemos D1 mini and DS18B20 as sensors.

chemmex, 2019-10-22:
Espeasy will automatically enter AP mode if it haven't found the known wifi network. And you can configure two networks, your own and your parents', in the Config menu

ghtester, 2019-10-22:
It's already there, of course you should have configured it properly before and a button connected to some GPIO... See Hardware tab - Reset Pin. Then you can press the button for about 10 sec to reset ESP_Easy to Factory Defaults.

Misiu, 2019-10-22:
@chemmex I didn't know that ESPEasy will enter AP mode if it won't find WiFi.
Thank You for the hint!
@ghtester should I connect a button between let's say D7 and GND or D7 and 3.3V? I saw multiple schemas on the internet and I don't want to damage my devices.

ghtester, 2019-10-22:
I don't know if the Reset Pin is active LOW or HIGH - perhaps it's mentioned somewhere but I did not use it so far.  I don't recommend to connect a button directly to 3,3V or to ground.  If the GPIO is not configured properly (i.e as output instead of input), direct connection to 3,3V could damage it. To be safe, I would put a 1k resistor in series to GPIO. In addition, there should be a pull-up (to 3,3V) or pull-down (to ground) resistor about 10k to GPIO and a button connecting 1k resistor (opposite end than to GPIO) to ground or to 3,3V.

TD-er, 2019-10-23:
There is also an option to reset to factory default by connecting RX and TX to each other  at boot.
This may not work if serial was disabled by the way.

Misiu, 2019-10-23:
@ghtester I'll try connecting a button using 
I have couple of button shields (https://wiki.wemos.cc/products:d1_mini_shields:1-button_shield), I don't see such resistors there. It is connected to D3. Do I must configure D3 as input and then in Hardware Tab select D3 as Reset Pin? 
I found [this article](https://www.instructables.com/id/Control-LED-Using-PushButton-With-NodeMCU/) and it suggest using 10K Ohms and 5V. Same [schema here](https://programmingelectronics.com/tutorial-19-debouncing-a-button-with-arduino-old-version/).

My question is: how should I connect the button? To 3.3V or to 5V? and how should I configure ESPEasy, only the reset pin section? Or should I also configure that pin as input (pull up, pull down or none)?

Misiu, 2019-10-23:
> There is also an option to reset to factory default by connecting RX and TX to each other at boot.
> This may not work if serial was disabled by the way.

Didn't know about that, thank You for the hint :)

TD-er, 2019-10-23:
The old Arduino was 5V. ESP processors are 3V3, so I would use that one.
There is still some debate whether the GPIO pins on the ESP are 5V tolerant, so better safe than sorry.

The resistor you need for the switch is not really precisely defined.
It must be low enough to pull the voltage to a proper level and high enough not to cause a short or excessive current when pushing the button.
You could also set the internal pull-up resistor of a GPIO pin and just connect the pin via a switch to GND. Better make sure you have a little resistor included then to prevent high current when you push the button and have the GPIO pin set high as an output accidentally. (470 Ohm would be enough).

tonhuisman, 2023-09-03:
This issue is no longer relevant, and can be closed.

TD-er, 2023-09-03:
This is not yet fixed/added.