letscontrolit/ESPEasy #2679: Improve usability of ESPEasy p2p protocol

TD-er, 2019-10-23:
See also:
- [ESPEasy P2P] Add port nr to NodeStruct #2252
- ESPEasy P2P Networking unclear #2522

It is not really intuitive to use the feature of sharing data from remote plugins using our p2p protocol.

The current implementation does require a lot of automatic configuration which is not really intuitive nor insightful.

One of the first things to improve this, is to at least make the unit number of the remote feed configurable like this:
[image]

Things to consider:
- Let the receiving end subscribe to a p2p feed (incl. setting the receiving task nr)
- Let the sender add a plugin to a selected unit
- Use a topic ID instead of a unit number.  (use IDX value on the sending controller as topic ID)

clumsy-stefan, 2019-10-24:
My comment would be, that you don't make any dependencies to MQTT as probably not everyone uses it and therefore excludes it in the build.

> Let the receiving end subscribe to a p2p feed (incl. setting the receiving task nr)

This would be the most important to me, that I can select the unit nr (by nr, IP, FQDN or whatever) AND the Task-Nr, as I don't necessarily have the same numbers empty on all nodes and probably want data from two different nodes which use the same task!

But I love the idea to enhance this. Would make some homeautomation things a lot easier not requiring a server in between for certain things (like connecting a switch to a lamp)!

TD-er, 2019-10-24:
> My comment would be, that you don't make any dependencies to MQTT as probably not everyone uses it and therefore excludes it in the build.

ESPeasy p2p has nothing to do with MQTT, but that doesn't mean we can't use concepts of MQTT (like a topic to subscribe to) in it.
That's (in my opinion) the only relation it should have with MQTT.