letscontrolit/ESPEasy #2683: Duplicated section in `platformio.ini`

ivankravets, 2019-10-25:
```
Error: Invalid '/Users/ikravets/Downloads/ESPEasy-mega/platformio.ini' (project configuration file): 'While reading from '/Users/ikravets/Downloads/ESPEasy-mega/platformio.ini' [line 501]: section 'env:esp32test_1M8_partition' already exists'
```

TD-er, 2019-10-25:
That's already being worked on for the next nightly build.
See https://github.com/letscontrolit/ESPEasy/pull/2667/commits/d42091f20e073756d11ce86a649c0dba3973ee78

Problem is that we had more issues than only that duplicate and as soon as I merge something, it will trigger a nightly build.

ivankravets, 2019-10-25:
Great! Also, a lot of non-valid commented values, such as "env_defaults". I tried to uncomment `esp32dev` and PIO says no env :)

TD-er, 2019-10-25:
The `esp32dev` is an rather old env definition, so I am not really surprised it is using an old syntax.
I will look into the ini file to see if I can clean up the old configs or fix the ones that may some day be useful again.

PR #2667 (2019-10-27): [Rules] Improve rules matching (#2660)
Detecting time compare and value compare could give false matches in some cases.
This PR also incorporates a number of other changes:
- Adds option to change the remote unit on remote fed plugins via p2p. (#2679)
- Cleanup of use of taskIndex, DeviceIndex (#2676)
- Disable and mark configured plugins not included in the build (e.g. when switching builds with other plugins included) (related to #2676)
- Fix build issues with PlatformIO using Python3
- Perform first read of ADC at init, before starting WiFi (#2646)

Fixes: #2683 - Duplicated section in `pla [... cut]
closed 2019-10-27 by TD-er