letscontrolit/ESPEasy #2696: DoubleClick- and Longpress-Issue (tested with Sonoff Touch T1)

toxic-tonic, 2019-10-29:
Hi everybody,

I face an issue with Sonoff Touch T1 ESPEasy (tried mega-20191028 and mega-20191003) with the doubleclick-/longpress-function of the switch-plugin. When trying to use that function, it always triggers both funtions shortpress and doubleclick AND longpress (?) on doubleclick or only shortpress and longpress on longpress! The log looks like this:

doubleclick:
......
207621: SW : GPIO=0 State=1 Output value=1
207624: EVENT: Touch#State=1.00
207662: ACT : gpio,12,0
207666: SW : GPIO 12 Set to 0
207709: SW : GPIO=12 State=0 Output value=0
207717: EVENT: Relay#State=0.00
207744: ACT : gpio,13,0
207749: SW : GPIO 13 Set to 0
207823: SW : GPIO=0 State=1 Doubleclick=3
207825: EVENT: Touch#State=3.00
208920: SW : LongPress: GPIO= 0 State=1 Output value=11
208922: EVENT: Touch#State=11.00
......

longpress:
......
115896: SW : GPIO=0 State=1 Output value=0
115898: EVENT: Touch#State=0.00
115942: ACT : gpio,12,1
115947: SW : GPIO 12 Set to 1
115983: SW : GPIO=12 State=1 Output value=1
115991: EVENT: Relay#State=1.00
116013: ACT : gpio,13,1
116017: SW : GPIO 13 Set to 1
116898: SW : LongPress: GPIO= 0 State=1 Output value=11
116909: EVENT: Touch#State=11.00
......

The related article in the forum is here: https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=7162

I tested different configurations like active high and active on low, several debounce-times and timeouts, with no effect.

I'm not shure waht additional informations you need, just let me know if you need something...

Thanks in advance

Tobias

TD-er, 2019-10-29:
I added the needsinfo tag, since I don't know if this behavior you're seeing is intended.
So it's not about you not supplying enough info, but I hope the author of the double-/long-press will shed some light on this.

toxic-tonic, 2019-10-29:
Found the same Problem discribed here: https://www.letscontrolit.com/forum/viewtopic.php?t=6152

toxic-tonic, 2019-11-04:
Hi,

Sorry for the late reply, was busy the last days! And Thank you for your explanation. I understand your point and it is more complicate than expected. In my opinion the easiest way would be to separate the classical switch and the push button.  So we leave the double-click / longpress just in the touch button and the activation of either would result in a delay, what would be logic for everyone! So the workflow would look like this:

Config: 	Button Active Low
		Doubleclick within 1000ms active on low
		Logpress longer then 1000ms active on low

You press the button (gpio low), and it starts a timer for 1000ms
-	Opt 1 you release the button a counter is set to 1
	o	no further clicks, after 1000ms (1Sec) the action for single click starts
	o	1 (or more?) further clicks happens and the counter increases and after 1000ms the action for double (or triple or what ever) click starts
-	Opt 2  you do not release the button and the button is still low after 1000ms
	o	The action for longpress starts

Do I miss something? If someone don’t want the delay he could either disable (or not enable) the double-click or increase the delay for it.

Best regards

Tobias

toxic-tonic, 2019-11-22:
Am I any near a possible solution? ;-)

Thanks

Tobias

TD-er, 2019-11-22:
@giig1967g is working on the GPIO handling for switch plugins.
Maybe he is also the best one to answer this question?

giig1967g, 2019-11-22:
sorry, but I am working in the PLUGIN_WRITE section of the plugin (commands)
The PLUGIN_TEN_PER_SECOND (longpress, etc.) will be my next work...

Deggle, 2020-05-16:
Hi All,

I've had the same issue - I'd like three separate MQTT events for single, double and long - however the standard behaviour will, as described above, will always immediately send the =1 event on initial stage change, with the double=3/long=11 (in my case) events.

I have a solution that has worked for me - using a rule and timer. Essentially, I've disabled the switch directly being sent to the controller and handled this with the rule. The 'single' event isn't sent until 1 second has passed to be sure it's not actually a double or long event (in which case the time is stopped).

It may not work for all your use cases, but I've seen at least someone wanting it for controlling 3 hue lights etc, which this will probably work fine for (though you'll need to manage the toggling of state elsewhere - this solution just splits the three events).

Rules:

```// Switch 1

on Switch1#switch1=1 do
 timerSet,1,1
EndOn

On Rules#Timer=1 do
 Publish sometopic/switch1,single
 timerSet,1,0
endon

on Switch1#switch1=3 do
 Publish sometopic/switch1,double
 timerSet,1,0
EndOn

on Switch1#switch1=11 do
 Publish sometopic/switch1,long
 timerSet,1,0
endon
```

Obviously you'll need to amend based on your switch names/values, MQTT topics and if you're using other timers. If you don't want MQTT I'm sure other actions/rules could be worked into this.

Hope this is helpful for someone!

giig1967g, 2020-05-16:
I am aware of the problem, and I think that the best practice (that I will develop as soon as the GPIO commands in the CORE will be merged) is to operate the SINGLE and LONG press events based on the RELEASE action of the button.

Maikel-K, 2020-07-06:
Has this been worked out?
I want to use single press to turn on the light and long press for dim up

tonhuisman, 2023-09-02:
No replies here for a long time, most likely can be closed

iz8mbw, 2023-10-28:
@TD-er @tonhuisman 
It seems this issue is still present, I have the same issue, see:

https://www.letscontrolit.com/forum/viewtopic.php?t=9881

https://www.letscontrolit.com/forum/viewtopic.php?t=6152