letscontrolit/ESPEasy #2700: SD Card Value Logger Enable

Droscovi, 2019-10-31:
How do I enable this in ESPEasy-mega-20191028? I have an SD connected up to NodeMCU SPI but the "SD Card Value Logger" is not available (not even visible). Any help would be greatly appreciated.

_Originally posted by @Droscovi in https://github.com/letscontrolit/ESPEasy/pull/429#issuecomment-548222739_

TD-er, 2019-10-31:
You have to enable it in the code by defining `FEATURE_SD`
But the feature has been left disabled for a long time and now it won't compile if you enable it.
This was also mentioned on the forum a few times, so I am working on repairing it, but I had to change/update a lot more code than anticipated, so not finished yet.

I hope to have it able to compile again in a few days.

Droscovi, 2019-10-31:
> You have to enable it in the code by defining `FEATURE_SD`
> But the feature has been left disabled for a long time and now it won't compile if you enable it.
> This was also mentioned on the forum a [...]

Thanks for the response.

 I had found some of the SD card lines in the code (and tried some of the earlier file changes) and  tried to compile a few times without any success.  

Looking forward to testing out the fix! I'm working on a NodeMCU/SDS011 mass deployment to solve some air quality issues where wifi is not completely stable. ESP Easy and NodeMCU fits my needs in every other aspect.

Misiu, 2019-11-13:
@TD-er any updates on the SD Card Value logger?
@Droscovi what errors do you get?

TD-er, 2019-11-13:
@Misiu I have already changed a lot of code to make it compile again, but there are still parts left which make it not compile.

TD-er, 2019-11-13:
I can now compile it, with SD card support, but it will not be useful.
Almost all RAM is used when `FEATURE_SD` is enabled.

I opened an issue about it: https://github.com/esp8266/Arduino/issues/6767

uzi18, 2019-11-13:
what happened if you just enable includes for sd card libraries without use it in runtime?

TD-er, 2019-11-13:
> what happeneds if you just enable includes for sd card libraries without use it in runtime?

Exactly the same as when I enable `FEATURE_SD`
Good idea by the way.

uzi18, 2019-11-13:
so something is in includes that break build, you can try to move these includes on the top

TD-er, 2019-11-13:
On the top where?

uzi18, 2019-11-13:
of all includes

TD-er, 2019-11-13:
So place it in ESPEasy.ino file at the top?

TD-er, 2019-11-13:
With only the include of `SD.h` at the top of ESPEasy.ino, the result is the same (slightly worse even):
```
DATA:    [==========]  95.4% (used 78148 bytes from 81920 bytes)
```

uzi18, 2019-11-13:
and any compiler warnings?

TD-er, 2019-11-13:
Nope

uzi18, 2019-11-13:
hope it is not one more time pio build issue. @clumsy-stefan maybe you can try this on arduino ide?

TD-er, 2019-11-13:
I will create a commit for the changes needed to actually compile it.

clumsy-stefan, 2019-11-13:
Can try the commits tomorrow.

uzi18, 2019-11-14:
@clumsy-stefan is it possible for you to paste here first and last 10 lines of build log from arduino ide?
(first you need to enable verbose build in preferences)
thanks

clumsy-stefan, 2019-11-14:
tried to compile with commit ff3cac9de4333ce5ac4c342f90cc4c050497a5fc but it fails to compile with:
```
/Users/stefan/git/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: address 0x3fffccd8 of /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_319071/ESPEasy.ino.elf section `.bss' is not within region `dram0_0_seg'

/Users/stefan/git/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: address 0x3fffccd8 of /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_319071/ESPEasy.ino.elf section `.bss' is not within region `dram0_0_seg'
collect2: error: ld returned 1 exit status
```

Did not further analy/test yet though...

TD-er, 2019-11-14:
The same build error, so that's good :)

clumsy-stefan, 2019-11-14:
> The same build error, so that's good :)

Thats's why I love IT... It's fails, so it's good.... That's why nobody else understands what we do.. Binary logic rocks 😃

TD-er, 2019-11-14:
Yep, there are only 10 kinds of people ;)

uzi18, 2019-11-14:
@clumsy-stefan and what about build log?

clumsy-stefan, 2019-11-14:
except the lines what it's compiling what I posted above is the only interesting part... more details only when sucessfully compiled..

uzi18, 2019-11-14:
so please disable Feature_sd to have successfull build log

clumsy-stefan, 2019-11-14:
hmm.. what information are you looking for? There is nothing in there except which files it compiles (~50-100 lines) and where it fails (which it doesn't if succesfully compiled).

```
/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/include -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/lwip2/include -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_279546/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_191024=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10810 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DUSE_CUSTOM_H -DPUYA_SUPPORT=0 -DCONFIG_FREERTOS_ASSERT_DISABLE -DCONFIG_LWIP_ESP_GRATUITOUS_ARP -DCONFIG_LWIP_GARP_TMR_INTERVAL=30 -DCORE_POST_2_5_0 -DCORE_POST_2_6_0 -DNDEBUG -DICACHE_FLASH -DWEBSERVER_RULES_DEBUG=0 -DVTABLES_IN_FLASH -mtarget-align -Werror=unused-variable -DBUILD_4M -DESP8266 -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/cores/esp8266 -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/variants/d1_mini -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/libraries/ESP8266WiFi/src -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/libraries/ESP8266WebServer/src -I/Applications/Arduino.app/Contents/Java/libraries/I2Cdevlib /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_279546/sketch/ESPEasy.ino.cpp -o /dev/null
/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/include -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/lwip2/include -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_279546/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_191024=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10810 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DUSE_CUSTOM_H -DPUYA_SUPPORT=0 -DCONFIG_FREERTOS_ASSERT_DISABLE -DCONFIG_LWIP_ESP_GRATUITOUS_ARP -DCONFIG_LWIP_GARP_TMR_INTERVAL=30 -DCORE_POST_2_5_0 -DCORE_POST_2_6_0 -DNDEBUG -DICACHE_FLASH -DWEBSERVER_RULES_DEBUG=0 -DVTABLES_IN_FLASH -mtarget-align -Werror=unused-variable -DBUILD_4M -DESP8266 -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/cores/esp8266 -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/variants/d1_mini -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/libraries/ESP8266WiFi/src -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/libraries/ESP8266WebServer/src -I/Applications/Arduino.app/Contents/Java/libraries/I2Cdevlib -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/libraries/Wire /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_279546/sketch/ESPEasy.ino.cpp -o /dev/null
/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/include -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/lwip2/include -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_279546/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_191024=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10810 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DUSE_CUSTOM_H -DPUYA_SUPPORT=0 -DCONFIG_FREERTOS_ASSERT_DISABLE -DCONFIG_LWIP_ESP_GRATUITOUS_ARP -DCONFIG_LWIP_GARP_TMR_INTERVAL=30 -DCORE_POST_2_5_0 -DCORE_POST_2_6_0 -DNDEBUG -DICACHE_FLASH -DWEBSERVER_RULES_DEBUG=0 -DVTABLES_IN_FLASH -mtarget-align -Werror=unused-variable -DBUILD_4M -DESP8266 -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/cores/esp8266 -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/variants/d1_mini -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/libraries/ESP8266WiFi/src -I/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/libraries/ESP8266WebServer/src -I/Applications/Arduino.app/Contents/Java/libraries/I2Cdevlib -I/Applications/Arduino.app/Con [... cut ...]
RODATA : 5764  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 38648 )         - zeroed variables      (global, static) in RAM/HEAP 
/Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-size -A /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_279546/ESPEasy.ino.elf
Sketch uses 841017 bytes (80%) of program storage space. Maximum is 1044464 bytes.
Global variables use 46228 bytes (56%) of dynamic memory, leaving 35692 bytes for local variables. Maximum is 81920 bytes.
```

uzi18, 2019-11-14:
need specialy first lines like these:
```
/home/users/linaja/arduino-1.8.3/arduino-builder -dump-prefs -logger=machine -hardware /home/users/linaja/arduino-1.8.3/hardware -hardware /home/users/linaja/.arduino15/packages -hardware /home/users/linaja/sketchbook/hardware -tools /home/users/linaja/arduino-1.8.3/tools-builder -tools /home/users/linaja/arduino-1.8.3/hardware/tools/avr -tools /home/users/linaja/.arduino15/packages -built-in-libraries /home/users/linaja/arduino-1.8.3/libraries -libraries /home/users/linaja/sketchbook/libraries -fqbn=arduino:avr:uno -ide-version=10803 -build-path /tmp/arduino_build_127720 -warnings=all -build-cache /tmp/arduino_cache_745775 -prefs=build.warn_data_percentage=75 -verbose /home/users/linaja/sketchbook/xx.ino
/home/users/linaja/arduino-1.8.3/arduino-builder -compile -logger=machine -hardware /home/users/linaja/arduino-1.8.3/hardware -hardware /home/users/linaja/.arduino15/packages -hardware /home/users/linaja/sketchbook/hardware -tools /home/users/linaja/arduino-1.8.3/tools-builder -tools /home/users/linaja/arduino-1.8.3/hardware/tools/avr -tools /home/users/linaja/.arduino15/packages -built-in-libraries /home/users/linaja/arduino-1.8.3/libraries -libraries /home/users/linaja/sketchbook/libraries -fqbn=arduino:avr:uno -ide-version=10803 -build-path /tmp/arduino_build_127720 -warnings=all -build-cache /tmp/arduino_cache_745775 -prefs=build.warn_data_percentage=75 -verbose /home/users/linaja/sketchbook/xx.ino                                                                                                                  
Using board 'uno' from platform in folder: /home/users/linaja/arduino-1.8.3/hardware/arduino/avr                                                                                                                                 
Using core 'arduino' from platform in folder: /home/users/linaja/arduino-1.8.3/hardware/arduino/avr                                                                                                                              

```

clumsy-stefan, 2019-11-14:
There you go:
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/stefan/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/stefan/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/stefan/git/libraries -fqbn=esp8266.com:esp8266:smy_4M:xtal=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=80,FlashMode=dio,eesz=4M1M,vt=flash,exception=disabled,led=2,ip=lm2f,lvl=None____,wipe=none,baud=115200 -ide-version=10810 -build-path /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_604396 -warnings=none -build-cache /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_cache_293311 -prefs=build.warn_data_percentage=75 -verbose /Users/stefan/git/ESPEasy/ESPEasy/ESPEasy.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/stefan/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/stefan/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/stefan/git/libraries -fqbn=esp8266.com:esp8266:smy_4M:xtal=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=80,FlashMode=dio,eesz=4M1M,vt=flash,exception=disabled,led=2,ip=lm2f,lvl=None____,wipe=none,baud=115200 -ide-version=10810 -build-path /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_build_604396 -warnings=none -build-cache /var/folders/yc/vn31kmqj2n3crsnwwtnpkfsc0000gm/T/arduino_cache_293311 -prefs=build.warn_data_percentage=75 -verbose /Users/stefan/git/ESPEasy/ESPEasy/ESPEasy.ino
Using board 'smy_4M' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266
Using core 'esp8266' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/esp8266.com/esp8266
```
But that's different for the various boards I use...

uzi18, 2019-11-14:
@clumsy-stefan thanks i will try to build with arduino-builder :)

TD-er, 2019-12-04:
I just tested it and the custom build does allow me to build a version with the `FEATURE_SD` defined.
I don't have a board that has an SD card, but it does seem to work.
All SD-card related items are visible and the unit does not crash when I try to view the files :)
I will add a "test_SD" build, so we will build the source more often and people may actually use the code :)

TD-er, 2019-12-05:
Last night I made a [ESPEasy_mega-20191130-17-PR_2798.zip](https://www.dropbox.com/s/oajoktd46u3643k/ESPEasy_mega-20191130-17-PR_2798.zip?dl=0) test build.
There is a `test_ESP8266_4M1M_VCC_MDNS_SD` build in there.

PR #2798 (2019-12-07): [FEATURE_SD] Add build in Platformio.ini with SD enabled (#2700 )
This does make sure we build more often with `FEATURE_SD` and `FEATURE_MDNS` enabled.

Fixes: #2700
commit 4e4e6b7f (2019-12-07): [FEATURE_SD] Add build in Platformio.ini with SD enabled (#2700 )

Fixes: #2700 

This does make sure we build more often with `FEATURE_SD` and `FEATURE_MDNS` enabled.
closed 2019-12-07 by TD-er