letscontrolit/ESPEasy #2704: Publishing a CO2 value to another device does not work anymore

micropet, 2019-11-02:
I have a device with a MH-Z19 Co2 sensor. It sends the Co2 value to another device with a Neopixel Stripe.

This does not work with the current version (20191028) anymore. My last working version is from 23.08.2019.

In the Rules I have, as TD-er said, the quotation marks.

At 540 PPM the device sends short "green" "NeoPixelAll, 0,250,0,0", then comes immediately "violet "NeopixelAll, 250,0,40,0"".

What amazes me: Co2 is not displayed in the log.

Here is an excerpt from the log:

 Command: event
245916: EVENT: violet
245963: ACT: Publish ESP-206 / cmd, 'NeopixelAll, 250,0,40,0'
245972: Command: publish
246060: EVENT: MH-Z19 # MHZ19Temp = 22.00
246129: EVENT: MH-Z19 # U = 0.00
272810: WD: Uptime 5 ConnectFailures 0 FreeMem 16344 WiFiStatus 3
275829: ACT: Publish ESP-206 / cmd, 'NeoPixelAll, 0,250,0,0'
275838: Command: publish
275885: ACT: event violet
275893: Command: event
275893: EVENT: violet
275940: ACT: Publish ESP-206 / cmd, 'NeopixelAll, 250,0,40,0'
275949: Command: publish
276013: EVENT: MH-Z19 # MHZ19Temp = 22.00
276078: EVENT: MH-Z19 # U = 0.00
278255: EVENT: Clock # Time = Sat, 10:18
302810: WD: Uptime 5 ConnectFailures 0 FreeMem 16328 WiFiStatus 3
305854: ACT: Publish ESP-206 / cmd, 'NeoPixelAll, 0,250,0,0'
305864: Command: publish
305908: ACT: event violet
305916: Command: event
305916: EVENT: violet
305965: ACT: Publish ESP-206 / cmd, 'NeopixelAll, 250,0,40,0'
305974: Command: publish
306036: EVENT: MH-Z19 # MHZ19Temp = 22.00
306101: EVENT: MH-Z19 # U = 0.00

TD-er, 2019-11-02:
I added a bit formatting to your post.

Could you add the following to your rules:

```
on MH-Z19#PPM do
    LogEntry,"MH-Z19 CO2: [MH-Z19#PPM]"
    if [MH-Z19#PPM]<700
....
```
This way we can see what the value actually is.
What is the set interval of the plugin?

TD-er, 2019-11-02:
By the way, I looked at the screenshot you posted:
[image]

Do you have Serial disabled in the advanced setting tab? (it should be)
Is there something else connected to the serial0 port? (e.g. on an USB to serial adapter)

micropet, 2019-11-02:
The first 2 or 3 measurements of MH-Z19 show mostly 000 ..
All following measurements are correct.
Serial is switched off.
The interval is 30 seconds.
On USB only one power supply is connected.

There is no PPM entry in the log with the current version:

194723: WD: Uptime 3 ConnectFailures 0 FreeMem 16408 WiFiStatus 3
198772: Command: publish
198805: ACT: event violet
198812: Command: event
198812: EVENT: violet
198845: ACT: Publish ESP-206 / cmd, 'NeopixelAll, 4,0,1,0'
198852: Command: publish
198898: EVENT: MH-Z19 # MHZ19Temp = 21.00
198940: EVENT: MH-Z19 # U = 0.00
199571: EVENT: Clock # Time = Sat, 8:18 pm
224723: WD: Uptime 4 ConnectFailures 0 FreeMem 16376 WiFiStatus 3
228768: ACT: Publish ESP-206 / cmd, 'NeoPixelAll, 0,3,0,0'
228775: Command: publish
228806: ACT: event violet
228812: Command: event
228813: EVENT: violet
228845: ACT: Publish ESP-206 / cmd, 'NeopixelAll, 4,0,1,0'
228852: Command: publish
228901: EVENT: MH-Z19 # MHZ19Temp = 21.00
228944: EVENT: MH-Z19 # U = 0.00

TD-er, 2019-11-02:
That log is output without the suggested line to add an extra logentry?
I will also have a look at the code of the plugin to see if I can explain why I do not see the logentry in your logs.
I also do not see the MH-Z19#PPM event, which would explain why the rules may not be working, since that's the trigger you use for your rules.

micropet, 2019-11-02:
No, LogEntry, "MH-Z19 CO2: [MH-Z19 # PPM]" is registered.
And yes, the Logeintag MH-Z19 # PPM event is in the old version, but it is missing here.

TD-er, 2019-11-02:
Hmm, so it seems there is some issue with a missing event being sent (or processed) 

I notice the spaces in your log with the event.
```
198940: EVENT: MH-Z19 # U = 0.00
```
Can you check if there is a space in the names of the plugin or variable name?

I checked in my test node:
```
72971187: EVENT: mem#ram=12384.00
```
As you can see, there is no space in there, surrounding the `#` or `=`

micropet, 2019-11-02:
The blanks were brought in by the Google translator, sorry. This is how it looks like:

4668884: EVENT: MH-Z19#MHZ19Temp=22.00
4668927: EVENT: MH-Z19#U=0.00

TD-er, 2019-11-02:
And not any event for the `MH-Z19#PPM` ?

micropet, 2019-11-02:
No

TD-er, 2019-11-02:
I just updated one of my nodes running a MH-Z19 to the same latest test build I made last night.

```
247748: EVENT: Clock#Time=Sat,22:35
248928: SYS :
253881: MHZ19: Raw PPM: 820 Filtered PPM value: 823 Temp/S/U values: 26/64/10819.00
253883: EVENT: co2#PPM=823.00
253914: ACT : LogEntry 'PPM: 823'
253922: Command: logentry
253923: PPM: 823
253925: EVENT: co2#Temperature=26.00
253931: EVENT: co2#U=10819.00
253958: Domoticz: Sensortype: 6 idx: 73 values: 823;26.00;10819.00
```

It is running this rule:
```
on co2#PPM do
  LogEntry "PPM: [co2#PPM]"
endon
```

[image]

micropet, 2019-11-02:
That's strange.
My node is working perfectly with an old version and not anymore with the current one.
Otherwise I have not changed anything ..

If I just take your rules, then it works for me too.
The problem must lie with the complete rule of mine.

on MH-Z19#PPM do
  LogEntry "PPM: [MH-Z19#PPM]"
endon

gives me:

12078571: EVENT: Clock#Time=Sat,23:36
12078714: MHZ19: PPM value: 554 Temp/S/U values: 26/0/0.00
12078716: EVENT: MH-Z19#PPM=624.00
12078735: ACT : LogEntry 'PPM: 624.00'
12078743: Command: logentry
12078743: PPM: 624.00
12078757: EVENT: MH-Z19#MHZ19Temp=22.00
12078785: EVENT: MH-Z19#U=0.00

TD-er, 2019-11-02:
Can you download and zip the rules file and send it to me?
So not cut/paste the content, but really download it from the filesystem

micropet, 2019-11-02:
When i only take this:

on MH-Z19#PPM do
    LogEntry,"MH-Z19 CO2: [MH-Z19#PPM]"
    if [MH-Z19#PPM]<700
        event gruen
    endif
    if [MH-Z19#PPM]>=700 and [MH-Z19#PPM]<=1100
        event gelb
    endif
    if [MH-Z19#PPM]>1100 and [MH-Z19#PPM]<1400
        event violet
    endif
    if [MH-Z19#PPM]>=1400
        event rot
    endif 
endon

i get:

12498742: MH-Z19 CO2: 592.00
12498747: ACT : event gruen
12498753: Command: event
12498753: EVENT: gruen
12498779: ACT : event violet
12498785: Command: event
12498786: EVENT: violet
12498834: EVENT: MH-Z19#MHZ19Temp=22.00
12498866: EVENT: MH-Z19#U=0.00
12501603: Command: publish

first "gruen" then "violet" ?

TD-er, 2019-11-02:
And after saving the rules, can you also try with renaming the task name to not having a minus sign in the name? (also change it in the rules)

micropet, 2019-11-02:
[Soft.zip](https://github.com/letscontrolit/ESPEasy/files/3801026/Soft.zip)

micropet, 2019-11-02:
I did.
Still sending "green" then "violet"

12978315: EVENT: MHZ19#MHZ19Temp=22.00
12978359: EVENT: MHZ19#U=0.00
12978571: EVENT: Clock#Time=Sat,23:51
13004723: WD : Uptime 217 ConnectFailures 0 FreeMem 16152 WiFiStatus 3
13008135: MHZ19: PPM value: 536 Temp/S/U values: 26/0/0.00
13008137: EVENT: MHZ19#PPM=606.00
13008157: ACT : LogEntry,'MHZ19 CO2: 606.00'
13008164: Command: logentry
13008164: MHZ19 CO2: 606.00
13008172: ACT : event gruen
13008179: Command: event
13008180: EVENT: gruen
13008209: ACT : Publish ESP-206/cmd,'NeoPixelAll,0,3,0,0'
13008217: Command: publish
13008249: ACT : event violet
13008256: Command: event
13008256: EVENT: violet
13008289: ACT : Publish ESP-206/cmd,'NeopixelAll,4,0,1,0'
13008296: Command: publish
13008364: EVENT: MHZ19#MHZ19Temp=22.00
13008406: EVENT: MHZ19#U=0.00

micropet, 2019-11-02:
For information:
MQTT regularly sends the correct PPM value.

micropet, 2019-11-02:
Gijs, I'm going to bed. Let's move on tomorrow.

micropet, 2019-11-03:
It works now. I think we have a problem with the rulesparser.
I have now instead of

     if [MHZ19 # PPM]> 1100 and [MHZ19 # PPM] <1400

added the equals sign
     if [MHZ19 # PPM]> = 1100 and [MHZ19 # PPM] <= 1400

`9103802: MHZ19: PPM value: 443 Temp/S/U values: 25/0/0.00
9103803: EVENT: MHZ19#PPM=513.00
9103824: ACT : LogEntry,'MHZ19 CO2: 513.00'
9103827: Command: logentry
9103827: MHZ19 CO2: 513.00
9103830: ACT : event gruen
9103832: Command: event
9103832: EVENT: gruen
9103842: ACT : Publish ESP-206/cmd,'NeoPixelAll,0,250,0,0'
9103845: Command: publish
9103895: EVENT: MHZ19#MHZ19Temp=21.00
9103934: EVENT: MHZ19#U=0.00`

TD-er, 2019-11-03:
So it does work incorrect for `>` or `<` so it seems.

That's strange, since I did not really change that part... I hope ;)

TD-er, 2019-11-03:
Hmm indeed, there is some bug here, which I can reproduce with my rules:
```
861224 : conditionMatchExtended: true_-53>-50 and -53<-1 and 9976.00 >10000
861225 : conditionMatchExtended: true && true_-53<-1 and 9976.00 >10000                        
861225 : conditionMatchExtended: true && true && false_9976.00 >10000
861226 : Lev.1: [if true && true && false]=false
861229 : Lev.1: [else]=true
861230 : ACT  : LogEntry,"ELSE"
861235 : Command: logentry
861236 : LogEntry,"ELSE"
```

It claims both of these are the same:
- `-53>-50 and -53<-1 and 9976.00 >10000` 
- `true && true && false`

Almost like the signs (or the variables) have been flipped in the compare.

TD-er, 2019-11-03:
Apparently there is a bug in what values are using to perform the match.

```
compareValues: _-52>=-50 and -52<=-1 and 12856.00 >=10000                       _ val1: -52.00 val2: -50.00 = false
7107 : conditionMatchExtended: false_-52>=-50 and -52<=-1 and 12856.00 >=10000
7108 : compareValues: _-52<=-1 and 12856.00 >=10000                       _ val1: -52.00 val2: 10000.00 = false
7108 : conditionMatchExtended: false && false_-52<=-1 and 12856.00 >=10000
7109 : compareValues: _12856.00 >=10000                       _ val1: 12856.00 val2: 10000.00 = true
7109 : conditionMatchExtended: false && false && true_12856.00 >=10000
7110 : Lev.1: [if false && false && true]=false
7113 : Lev.1: [else]=true
7115 : ACT  : LogEntry,"ELSE"
7120 : Command: logentry
7121 : LogEntry,"ELSE"
```

See this line:
```
compareValues: _-52<=-1 and 12856.00 >=10000                       _ val1: -52.00 val2: 10000.00 = false
```

TD-er, 2019-11-03:
It is indeed something I messed up when changing the `findCompareCondition` code.
I overlooked the way that code did look for the right compare.
It does scan for occurrences and must return the first it can match in the string.
In my changes it does return the first checked one it found in the string.

TD-er, 2019-11-03:
Changed the code and it does now report:
```
11565 : compareValues: _-51>=-50 and -51<=-1 and 13872.00 >=10000                       _ val1: -51.00 val2: -50.00 = false
11565 : conditionMatchExtended: false_-51>=-50 and -51<=-1 and 13872.00 >=10000
11566 : compareValues: _-51<=-1 and 13872.00 >=10000                       _ val1: -51.00 val2: -1.00 = true
11566 : conditionMatchExtended: false && true_-51<=-1 and 13872.00 >=10000
11567 : compareValues: _13872.00 >=10000                       _ val1: 13872.00 val2: 10000.00 = true
11568 : conditionMatchExtended: false && true && true_13872.00 >=10000
11568 : Lev.1: [if false && true && true]=false
11571 : Lev.1: [else]=true
11573 : ACT  : LogEntry,"ELSE"
11578 : Command: logentry
11578 : LogEntry,"ELSE"
```

Will make a commit for it and it will be included in tomorrow's build.

TD-er, 2019-11-03:
@micropet Thanks for being so persistent on this bug.

micropet, 2019-11-03:
I thank you !!
If something suddenly stops working what has been going on for months, that's stupid.
Then I'm looking forward to tomorrow's version.

closed 2019-11-03 by TD-er

micropet, 2019-11-03:
Works !! Thanks.

PR #2708 (2019-11-03): [OTA] Fix minimal OTA size computations (#2707) & Rules bug (#2704)
Fixes: #2707
The computations did not take into account the needed extra space for 4k block alignment for OTA.

Fixes: #2704
Multiple compares in a rules line would be wrongly computed.
commit 6a48118c (2019-11-03): [Rules] Fix (serious) bug in matching compares (#2704)

Fixes: #2704

This is quite a serious bug in comparing values.
The compare was using the wrong values on a compare line when multiple compares had to be made. (with multiple and/or)