letscontrolit/ESPEasy #2707: OTA Flash not possible

fly74, 2019-11-03:
Hi folks,

OTA from ESP_Easy_mega-20191028_minimal_core_242_ESP8285_1M_OTA.bin to ESP_Easy_mega-20191103_minimal_core_242_ESP8285_1M_OTA.bin with 2 way is not possible. Update page only shows:

URI: /update
Method: GET
Arguments: 0

with different browsers.

TD-er, 2019-11-03:
Have you also tested other flash sizes to perform an OTA?  (just to make sure the OTA is not broken)

TD-er, 2019-11-03:
I just tested with a 4M node I have here and the OTA itself does work.
Then it may be a matter of having the wrong size used, either in the detection code in the web interface, or in the build script.

Can you share a screenshot or cut/paste of the web page showing the build/sketch information?
I wonder what it reports as used sketch size.

Or do you update via another route and not using the update feature from the web page?

fly74, 2019-11-03:
4M Sizes are OK on other nodes. But the 1M Sonoff is the Problem.

`Firmware
--
Build:⋄ | 20104  - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20191028
Plugins:⋄ | 3 [Normal] [Minimal 1M OTA]
Build Md5: | 6dfb115cbbefd1d3bf93bdd04530a4
Md5 check: | passed.
Build Time:⋄ | Oct 28 2019 11:32:41
Binary Filename:⋄ | ESP_Easy_mega-20191028_minimal_core_242_ESP8285_1M_OTA.bin
`

[image]

[image]

TD-er, 2019-11-03:
Can you also include the reported sketch size? I see it is not included in the standard Github copy selection.

fly74, 2019-11-03:
`ESP Board
--
ESP Chip ID: | 1225965 (0x12B4ED)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP8285
Storage
Flash Chip ID: | Vendor: 0x5E Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 601 kB (272 kB free)
SPIFFS Size: | 110 kB (28 kB free)
Page size: | 256
Block size: | 4096
Number of blocks: | 27
Maximum open files: | 5
Maximum path length: | 32`

TD-er, 2019-11-03:
The size of the 2-step OTA binary is 276848 bytes (270.36 kB), so it should fit.
But I guess we're having an overlap here in sector boundaries.
I will compute the new max OTA size (which will be a bit lower) and also have a look at the code to determine whether the 2-step OTA is possible.

```c++
bool OTA_possible(uint32_t& maxSketchSize, bool& use2step) {
  maxSketchSize = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
  const bool otaPossible = maxSketchSize > SMALLEST_OTA_IMAGE;
  use2step = maxSketchSize < ESP.getSketchSize();
  if (use2step) {
    const uint32_t totalSketchSpace = ESP.getFreeSketchSpace() + ESP.getSketchSize();
    maxSketchSize = totalSketchSpace - SMALLEST_OTA_IMAGE;
  }
  if (maxSketchSize > MAX_SKETCH_SIZE) maxSketchSize = MAX_SKETCH_SIZE;
  return otaPossible;
}
```

What strikes me as odd, is that I do use a sector bound correction for the free sketch space, but I don't do it on the computations on this line:
```c++
    const uint32_t totalSketchSpace = ESP.getFreeSketchSpace() + ESP.getSketchSize();
```

TD-er, 2019-11-03:
If I take into account the 4k boundary needed for writing (a 4k block is minimal size to be erased), then the max. sketch size can be only 600k.
So then the max. must be set at 614400 bytes.

TD-er, 2019-11-03:
I created a PR for this, which does take into account the 4k block alignment requirements.
See #2708

PR #2708 (2019-11-03): [OTA] Fix minimal OTA size computations (#2707) & Rules bug (#2704)
Fixes: #2707
The computations did not take into account the needed extra space for 4k block alignment for OTA.

Fixes: #2704
Multiple compares in a rules line would be wrongly computed.
commit 78336b65 (2019-11-03): [OTA] Fix minimal OTA size computations (#2707)

Fixes: #2707

The computations did not take into account the needed extra space for 4k block alignment for OTA.
closed 2019-11-03 by TD-er