letscontrolit/ESPEasy #2709: last 2 version only publish month instead of complete date

Sasch600xt, 2019-11-03:
Hello all,

since the last 2 Version i get only the month back instead of the build date (Month, Day, Year)

So this rule i use (since years) :
`Publish,%sysname%/status/Build_date, %sysbuild_date%`

Have a great day

TD-er, 2019-11-03:
Use quotes for the last argument.
```
Publish,%sysname%/status/Build_date,"%sysbuild_date%"
```

Sasch600xt, 2019-11-03:
Thank you ! That helped :)

Sasch600xt, 2019-11-03:
just for understanding,

for IP it works still without quotes:
`Publish,%sysname%/status/Ip_Adresse, %ip%

So is it just for build date ?`

TD-er, 2019-11-03:
The biggest change is in the parsing of the parameters.
Some commands were parsing the parameters and some were parsing the last parameter as "now just use all until the end of the line"
A parameter separator is either a space or a comma (space-comma and comma-space are just a single separator).

When the build date is converted into a string, it does contain a space.
```
Publish,%sysname%/status/Build_date, %sysbuild_date%
```

Will be replaced into:
```
Publish,myhostname/status/Build_date, Nov 03 2019
```

As you can see, there is a space between `Nov` and `03`, which is then seen as a new parameter.

closed 2019-11-03 by TD-er