letscontrolit/ESPEasy #2711: Feature request, prevent Wifi to let ESPNOW work as option

FanOfHue, 2019-11-03:
Please add a rule command to prevent ESP Easy from connecting to Wifi or start an AP.

Currently it seems that normal Wifi and ESPNOW cannot be used simultaniously.
So we build an ESP Easy as serial gateway to a another ESP Easy.
The normal ESP Easy should be able to control the gateway on serial cross connection.
This scenario has been tested and verified to work on SmartNodeRules.
It should not be a fixed setting because you need to be able to escape it from serial command from another ESP.

This is how it's done with rules on SmartNodeRules: (some rule commands may be different or not exist yet on ESP Easy)

The gateway ESP:

```
on System#Config do
  Config,AutoConnect,0  // This prevents unit from starting STA or AP automatically
endon
on system#boot do
  timerset init,30
endon
on stop do
  timerset init,0
  wifiInit // rule based start of Wifi connect for maintenance
endon
on timer#init do
  // No stop received in 30 seconds, so start ESPNOW to listen to peer nodes
  espnowconfig,1234567890abcdef,1234567890abcdef,36:33:33:33:33:33,Receiver
  espnowAddPeer 1234567890abcdef,36:33:33:33:33:01,1
endon
```

The units starts without automatically connect to Wifi.
Then it waits for 30 seconds to get a 'stop' command from the normal ESP on serial.
If no stop is received, it will automatically start ESPNOW

The normal ESP can send a reboot on serial, wait few seconds, then send stop to make the gateway boot normal for OTA or other maintenance.
So to put the gateway into maintenance, from the normal ESP:

```
on maintenance do
  serialsend reboot
  delay 2000
  serialsend event stop
endon
```

Has been working for weeks without any issue.

ghtester, 2019-11-05:
I would also appreciate an option to disable AP mode or both AP and STA.

abhelou, 2021-04-11:
@FanOfHue any progess if ESPNOW needed, while also sending mqtt from mesh to controller via wifi or serial... ?