letscontrolit/ESPEasy #2712: Please reinstate plugin call UNCONDITIONAL

FanOfHue, 2019-11-03:
Don't know since when this is disabled in ESP Easy. My plugin P253 depends on it.

`// PluginCall(PLUGIN_UNCONDITIONAL_POLL, 0, dummyString);
`
When i remove the // it works again. Could you elaborate on why it's disabled?

TD-er, 2019-11-03:
It was disabled here: https://github.com/letscontrolit/ESPEasy/commit/7064e76c97ccf2bf9c90d0a0bb07d2913dec24b6
It was only used on switch plugins (P001, P009, P019) and that change was also about changing the gpio pin state monitoring.

It is now disabled, so it is effectively not being used anymore.
This means we should either:
- Use a new one for your purpose
- disable it for the "switch plugins"

What do you use it for?
Is the name for it really fitting for your purpose?

FanOfHue, 2019-11-04:
In addition to PLUGIN_UNCONDITIONAL, i have a question about PLUGIN_WRITE also.
It works for my plugins, but looking at the code, it seems to be a workaround that controls it:

from rules tab (skipped irrelevant code to make it more readable):

```
    if (!PluginCall(PLUGIN_WRITE, &TempEvent, tmpAction)) {
      ExecuteCommand(VALUE_SOURCE_SYSTEM, action.c_str());
    }
```

So it calls PLUGIN_WRITE and on return false, tries to execute as internal command

from plugin tab (skipped irrelevant code to make it more readable):
```
    case PLUGIN_WRITE:
      {
        for (taskIndex_t task = 0; task < TASKS_MAX; task++)
        {
          if (Settings.TaskDeviceEnabled[task] && validPluginID(Settings.TaskDeviceNumber[task]))
            bool retval = (Plugin_ptr[DeviceIndex](Function, &TempEvent, str));
        }

        // @FIXME TD-er: work-around as long as gpio command is still performed in P001_switch.
        for (deviceIndex_t deviceIndex = 0; deviceIndex < PLUGIN_MAX; deviceIndex++) {
          if (validPluginID(DeviceIndex_to_Plugin_id[deviceIndex])) {
            if (Plugin_ptr[deviceIndex](Function, event, str)) {
            }
          }
        }
      }
```

So if the plugin is used in a task, call PLUGIN_WRITE but there's also a workaround to call all plugins (so without a task active)
To summarize: PLUGIN_WRITE is called to all plugins, even without being used in a task, but it's considered a workaround.
What would be the required approach to let a plugin handle PLUGIN_WRITE without a task requirement?
I can image lots of plugins that add 'commands' without needing a task.
When the workaround is stopped, some of my plugins will stop working also.

Development Wiki says that PLUGIN_WRITE is not related to tasks:
https://www.letscontrolit.com/wiki/index.php?title=ESPEasyDevelopment#PLUGIN_WRITE

What i need in general:

1) PLUGIN_WRITE or something similar to add commands to ESP Easy without a task requirement
2) A ten-per-second call to the plugin, also without a task requirement, just to handle internal things like processing incoming data

Instead of using tasks, it would also be an option to use registration of callbacks functions to the framework.
This would not cause delays on the framework when the callback is a null pointer.
But for multiple plugins, we would need multiple callbacks (dynamic array of callback function list?)
Think of how the Webserver class handles callbacks to a dynamic list of functions, one for each page.

TD-er, 2019-11-04:
> To summarize: PLUGIN_WRITE is called to all plugins, even without being used in a task, but it's considered a workaround.

Nope, it is not a work-around.
We do have several commands.
Some are at a global level, like `LogEntry`, `Publish`, etc.
And some are specific for a plugin.
Commands can also be sent via several routes, like:
- Rules
- HTTP GET (in the URL)
- MQTT
- Via the command interface on the Tools tab in the web interface
- Via serial
- Via ESPEasy p2p

And maybe I forgot a few?
So there is a command receiving part, which also must decode it (via GET url, you have to URL-encode some characters) and validate it.
Then there is a part which must try to execute it:
- Global commands
- Plugin specific commands

The plugin commands may need some proper redesign though, since most of them do not support a task index  and thus not support multiple instances of a plugin.
Also if plugin commands would be a bit more recognizable, then we don't have to "search" for it by first trying to match it against internal commands and then try to run it on all active tasks to see whether it will stick somewhere.

The work-around for checking against all included plugins is also mentioned as a temp work-around in the FIXME comment.
That's how it was working in the past and was broken for some specific command which are not conforming to the above mentioned design.
The P001_switch plugin is really a bad designed one, or rather grew into this monster it is right now.
I really would like to throw it away and split its functionality over 4 or 5 new plugins. That would make it all so much more reliable and simple.
So if your concern is (slightly) related to the P001 plugin, then please park it for now, as that cludge of a plugin will only be here like it is now for a limited time.

As you may have seen, I have optimized the search for the internal commands quite a bit, so that's rather fast. But the search for plugin commands is really not efficient.

> What would be the required approach to let a plugin handle PLUGIN_WRITE without a task requirement?
> I can image lots of plugins that add 'commands' without needing a task.

Can you give some example, as the current design does not really make sense for "lots of plugins" adding commands that can be useful without being active.
If those use cases are about handling GPIO pins, then those issues will be addresses as soon as we split P001 (and P009, P019) by having a central GPIO manager.
              
I think your ESPnow tool should not be considered a plugin, but rather a controller.
Only thing is, the current controllers do mainly output data and yours will be more like input/output.
Some controllers (openHAB/Domoticz MQTT) do allow to receive commands, but those are rather plugin specific and also a bit of a hack.
The MQTT import is now a plugin, to overcome the missing link of the current controller approach and in a sense it is also a bit of a plugin, as it does yield sample data, like a sensor.

So if you need to have a command which is not tied to a plugin, then it is a global command.
If you need a ten/sec call to a plugin which is not an active task, then your choice of making a plugin is probably the wrong one.
MQTT import does also have some periodical poll to handle MQTT messages, but that's also a hack.
I think your situation is very similar to the MQTT import one.

So maybe your ESPnow layer should be split into two tightly coupled instances:
- a controller
- a plugin

I also have a question here.
I want to add WiFi mesh, which is also using ESPnow as a layer between nodes which are not the master. Maybe we should also check and see if those applications do bite eachother.

uzi18, 2019-11-04:
@TD-er what about add explicity address for command? like: 
Publish(1),args ->publish by controller 1
Publish("name") ->publish by controller name
or
"name",Publish -> publish by controller name
"taskname",command -> send command to taskname
1,command ->send command to task 1

FanOfHue, 2019-11-04:
Thanks for taking the time to explain all details!

I prefer using the powerful rule engine to do all the work so i would need configuration commands for all plugins that i use instead of gui based templates and the task engine.
This is how Smartnoderules works and i had hoped to get something similar done in ESP Easy by extending plugins with configuration and execute commands.

Just a sample:

  // Config part
  ESPNowConfig 1234567890abcdef,1234567890abcdef,02:01:02:03:04:01,Sender
  ESPNowAddPeer 1234567890abcdef,02:01:02:03:04:30,0

  // Execute part
  ESPNowSend LSDoorSensor/state=%eventvalue%

Using the rule engine is great, configurations are just text files and it's very convenient to just copy/paste parts between modules.
And store them as notepad files instead of binary dat files that are useless to humans.

I'm aware that current approach is not really great for performance.
But would still be nice if a plugin (whatever type/name) could register commands with callback to the framework in a way that it does not suffer poor performance.
Pseudo code sample:

 registerCommand(F("MyCustomCmd"), &MyCustomFunction)

or something like that.

Could be that i'm completely on the wrong track, trying to expand/customize ESP Easy without touching the core framework, but solely by using plugins.

TD-er, 2019-11-04:
> registerCommand(F("MyCustomCmd"), &MyCustomFunction)

Something like a macro which can be used in a rule?
We already have something similar.
You can create any event name you like and call that event as a command.
Such event can also be given from another node (currently only via HTTP or p2p), so I guess your link between ESPEasy and your SmartNodeRules could also be extended to give commands between nodes.
So it is really just an event, with maybe some parameters.
This event will then trigger something in a rule (which may also call other commands).

I don't know yet if any more simple approach is also possible, but I guess considering remote commands as events is quite flexible.

So to have a link between these nodes, we either need to extend the p2p controller to also send/receive data via ESPnow, next to UDP, or we need a new controller which under the hood does share a lot of code with p2p controller.

@uzi18 The Publish command (and some other commands) could use a lot of extra options.
That's one of the reasons I was cleaning up that part of the code and also no longer allowed to consider "parameter 2 - till end of line" as a valid parameter structure.
This does allow to add some parameters later like indeed addressing what controller to use.

uzi18, 2019-11-04:
idea with prepend taks/controller name is stright forward, when set DeviceIndex or TaskIndex we can cut it and interpret like other commands

TD-er, 2019-11-04:
DeviceIndex is something internal, depending on the included plugins per build.
PluginID or taskIndex should be possible, but pluginID is a bit redundant when it is a plugin-specific command.
Maybe task name is also making sense when addressing a task.

uzi18, 2019-11-04:
it is possible also to hardcode TASK1,command or Controler1.command when address without name

TD-er, 2019-11-04:
Hmm that's maybe an even better approach, since you don't have to change the commands in the plugins.

FanOfHue, 2019-11-27:
To solve future performance issue in ESPCoreRules, i've implemented a way to enable/disable plugins at runtime so the core does not have to check for unneeded commands all the time. Similar to tasks.

And to handle fast calls to a plugin like needed for network processing, i added following code to ESPCoreRules. Maybe something similar could be added to ESPEasy. It should not affect performance when it's not in use by any plugin:
 
```
globals.h:
==========
void (*corePluginCall_ptr[PLUGIN_FASTCALL_MAX])(void);

setup:
======
for (byte x = 0; x < PLUGIN_FASTCALL_MAX; x++)
    corePluginCall_ptr[x]=0;

main loop:
==========
// handle plugins that reqistered a fast loop call
for(byte x=0; x < PLUGIN_FASTCALL_MAX; x++)
  if(corePluginCall_ptr[x] != 0)
    corePluginCall_ptr[x]();

misc:
=====
boolean registerFastPluginCall(void (function(void))){
  for(byte x=0; x < PLUGIN_FASTCALL_MAX; x++){
    if(corePluginCall_ptr[x] == 0){
      corePluginCall_ptr[x] = *function;
      return true;
    }
  }
  return false;
}

plugin sample:
==============
case PLUGIN_WRITE:
{
  if (cmd.equalsIgnoreCase(F("MSGBUSinit")))
    registerFastPluginCall(&P100_MSGBusReceive);
}
```

TD-er, 2019-11-27:
Hmm that's an interesting approach, to only have a vector of active plugins.
Does make the code simpler, faster etc.

FanOfHue, 2019-12-21:
Have to re-enable this feature after every github update (until it will disappear forever?)

Any new on this matter. I would prefer to have it enabled again by default or please provide an alternate mechanism like the "registerFastPluginCall"

FanOfHue, 2020-02-12:
No longer required

closed 2020-02-12 by FanOfHue

uzi18, 2020-02-12:
@FanOfHue how did you resolved it?