letscontrolit/ESPEasy #2713: Average readings from sensor before sending result to controller

Misiu, 2019-11-05:
I have a very simple setup: Wemos D1 with DS18B20.
I successfully configured the Home Assistant MQTT controller and values are correctly send to my HA instance.
The problem is that sometimes I get inaccurate readings - the reading changes the value in one reading but in second or third it gets back to normal.

Is there a way to configure ESPEasy to read the value every X seconds, then average the results and then send the value to the controller?

There are many methods of averaging results, but a simple sum of readings divided by the number of readings would be great (more advanced, like removing extremes before calculating the average would be even better)

So for example, make a reading every 5 seconds, then after having 12 readings (one minute) calculate the average and then send the result to controller.

I have no idea if this is currently possible, maybe using a rule.
I wasn't able to find a solution, so this issue is a feature request.
If this is currently possible please close this issue.

TD-er, 2019-11-05:
It can be done and is also described in the documentation.
See [rules documentation](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html) and search for "average" on the page.

Misiu, 2019-11-05:
@TD-er it looks like I didn't search hard enough. Thank You for the link. I'll try to build something and post my code here. Maybe someone will find it useful.

Misiu, 2019-11-05:
Rules work great. I had no idea that this gives so much power 🚀

I'll describe steps I took to get rules working (maybe someone will find this useful)

1. Enable rules - Tools->Advanced->Rules
[image]

2. Create a generic dummy device for each sensor you want to calculate average
[image]

3. Create rule in Rules tab

```
On Kitchen#Temperature Do
 Let,10,[VAR#9]
 Let,9,[VAR#8]
 Let,8,[VAR#7]
 Let,7,[VAR#6]
 Let,6,[VAR#5]
 Let,5,[VAR#4]
 Let,4,[VAR#3]
 Let,3,[VAR#2]
 Let,2,[VAR#1]
 Let,1,[Kitchen#Temperature]
 TaskValueSet,3,1,([VAR#1]+[VAR#2]+[VAR#3]+[VAR#4]+[VAR#5]+[VAR#6]+[VAR#7]+[VAR#8]+[VAR#9]+[VAR#10])/10
EndOn
```

This works great however I found two issues that I think are bugs.
1. After powering up my board the value from my device is immediately send to controller, even I set the interval to 60 seconds. Shouldn't ESPEasy wait 60 seconds before sending the first value? I calculate the average from 10 values, so the result is wrong because the rule takes first value and nine zeros. Is this intended? Can this be changed?

2. The dummy device has a display bug.
I configured the dummy device to output a single value, but after rebooting the device shows 4 values, when I edit the device and hit submit 3 unneeded values disappear:
[image]

I use ESP_Easy_mega-20191003_normal_ESP8266_4M1M.bin if that matters.

One additional question about rules:
when creating events or reading values we call device and value by name, for example, `Kitchen#Temperature` but when storing value we must call the device and the value by the number, for example, `TaskValueSet,3,1`
Can this be unified? Ideally, we would be able to call `TaskValueSet,KitchenFix#Temperature,Value`. This will be much more readable. What do you think?

TD-er, 2019-11-05:
You should set the interval to 0 if you don't want it to send out data.
The interval timer will be scheduled at init of the plugin, which is part of this issue: https://github.com/letscontrolit/ESPEasy/issues/1202#issuecomment-550014020

TD-er, 2019-11-05:
> The dummy device has a display bug.
> I configured the dummy device to output a single value, but after rebooting the device shows 4 values, when I edit the device and hit submit 3 unneeded values disappear:

That's a known issue. See https://github.com/letscontrolit/ESPEasy/issues/2675

TD-er, 2019-11-05:
> One additional question about rules:
> when creating events or reading values we call device and value by name, for example, `Kitchen#Temperature` but when storing value we must call the device and [...]

That's also an existing issue: https://github.com/letscontrolit/ESPEasy/issues/1676

Misiu, 2019-11-05:
@TD-er than you for the links.

> You should set the interval to 0 if you don't want it to send out data.
> The interval timer will be scheduled at init of the plugin, which is part of this issue: #1202 (comment)

I want to send the data, but after the interval. By default, the device sends the data after boot. I'd like to wait the interval before sending the first value and then send it normally.

TD-er, 2019-11-05:
You can send the data by calling RunTask from the rules for example.

Misiu, 2019-11-05:
I'll try that, I have 2 more free rules sets. Ideally, there should be an option to disable the first send to the controller after boot.

TD-er, 2019-11-05:
> I'll try that, I have 2 more free rules sets. Ideally, there should be an option to disable the first send to the controller after boot.

That's what https://github.com/letscontrolit/ESPEasy/issues/1202 is about.
Please add the suggestion there, so I will not forget it.