letscontrolit/ESPEasy #2715: No IDE-OTA,  cannot use core 2.4.2 any more

s0170071, 2019-11-06:
when I select core 2.4.2 in the Arduino IDE boards manager, 
I have to put a #define ARDUINO_ESP8266_RELEASE_2_4_2 in the ESPEasy.ino file. 

Then I get 

```
/home/john/Arduino/scetchbooks/libraries/ESPEasySerial/ESPeasySerial_ESP8266.cpp:277:19: error: 'class HardwareSerial' has no member named 'hasRxError'
   return getHW()->hasRxError();

```

Background: just examining, why OTA is not working. May be core related. 
I can upload a bin file from the web interface and its working fine. 
I can upload a sketch from the Arduino IDE, no error shown, yet the binary seems not uploaded.

TD-er, 2019-11-06:
That's a library I wrote, so apparently I have not tested it well with core 2.4.2?
Which is a bit strange, since if I'm not mistaken, that core version is also part of the builds....?

clumsy-stefan, 2019-11-07:
>Background: just examining, why OTA is not working. May be core related.
>I can upload a bin file from the web interface and its working fine.
>I can upload a sketch from the Arduino IDE, no error shown, yet the binary seems not uploaded.

I experienced the same when using `espota.py` sketch is beeing uploaded without errors, node rebooted, but still the old version runs.

Interestingly it works on ESP32 nodes, there I can use `espota.py` to upload and the new version is active after reboot.

s0170071, 2019-11-07:
its the same for core 2.4.2. I will check the plain OTA example this evening.

s0170071, 2019-11-07:
Nailed it. 
The basic OTA example works for core 2.4.2 and 2.5.2. Problem comes from serialPrintln() lines in  ArduinoOTA.onEnd() 

```
  ArduinoOTA.onEnd([]() {
      serialPrintln(F("\nOTA  : End"));
      //"dangerous": if you reset during flash you have to reflash via serial
      //so dont touch device until restart is complete
       serialPrintln(F("\nOTA  : DO NOT RESET OR POWER OFF UNTIL BOOT+FLASH IS COMPLETE."));
      delay(100);
      reboot();
  });  

```
I would recommend using the normal Serial.println() since a reboot is imminent anyway. Not sure however what exactly goes wrong after serialPrintln() gets called.

clumsy-stefan, 2019-11-08:
Hmm.. did ar recompile without these two lines and tried it on two different esp8266 nodes. Still can't use ota then, same thing happens, it uploads the new binary, reboots, but starts the old one. So in my environment I can't really confirm that this solves the problem...

I'll also do a few more tests on this...

clumsy-stefan, 2019-11-08:
Did a few more tests as @s0170071 pointed to the right direction. For me it works without issues if I just remove the `delay();`and `reboot();` functions at the end of `ArduinoOTA.onRnd()` function.

Also comparing to examples out there the reboot is never "forced" in the `eonEnd` call. It seems the node is rebooting anyway after succesfully flashing! So this could be some kind of race condition if the reboot occurs too early or that after OTA the node is not dooing a "normal" reboot but some kind of special calls to actually update the flash...

@s0170071 could you try that?

s0170071, 2019-11-08:
Did some more tests and it turned out that the serialPrintln statements are not the only thing that prevents flashing. Removing them improves the situation but sometimes flashing still fails. 

It is way more reliable if I just remove all proprietary callback functions - which also includes the reboot. I would guess that the actual flash or switch to the new firmware happens after the onEnd function returns or the delay triggers sufficient yielding. Which does not happen if the reboot happens before that.

clumsy-stefan, 2019-11-08:
Agree! I have a reproducable stable situation after removing the sleep and reboot functions... So I guess chainging the `SerialPrintLn()` to standard `Serial.PrintLn()` and removing the `delay()` and `reboot()` function would probably solve the issue completetly!

s0170071, 2019-11-08:
ok. 
what good does the printLn anyway ? I mean, its OTA. There is rarely any serial attached.

clumsy-stefan, 2019-11-08:
> what good does the printLn anyway ? I mean, its OTA. There is rarely any serial attached.

I thiink not a lot besides debugging... But the serial output shouldn't interfere with the OTA, as all the examples I found also output progress and end to the serial port....

TD-er, 2019-11-08:
Are the node and the computer sending the OTA connected to the same WiFi AP? (both connected via WiFi)
I noticed update issues when both were connected to the same WiFi AP. When connecting from another (wired connected) computer it was working better.
But maybe it has nothing to do with this issue, as it was already failing during upload of the image and not rebooting at all when the upload failed.

clumsy-stefan, 2019-11-08:
I tried different variants, but could reliably update when not doing the reboot manually (eg, removing the `reboot()` line..

TD-er, 2019-11-08:
Can you make a PR for it?

clumsy-stefan, 2019-11-08:
sure, but I'm a bit unsure, about the `SerialPrintLn()`, as fpr  me it works with them, so I think we should wait for @s0170071 feedback?

I can start a a PR with just excluding the delay and reboot and add the change of serial output later on, so at least you or others could test it also...

TD-er, 2019-11-08:
> I can start a a PR with just excluding the delay and reboot and add the change of serial output later on, so at least you or others could test it also...

Please do!

clumsy-stefan, 2019-11-08:
done, see #2722

clumsy-stefan, 2019-11-08:
BTW: did you see, that some of the latest update/commit in the core saves around 20k bin. size!! I guess it's the one you mentioned about includeing LittleFS... 1M Builds are far below the max size again!

TD-er, 2019-11-08:
Yep, 20k was due to the issue I reported where SPIFFS and LittleFS code were both included in the binary while only one of them was really needed.
I will also change the used NONOS to 20191024 revision, to make sure it uses the latest WiFi code from Espressif. Not sure what that does to the binary size though.

clumsy-stefan, 2019-11-08:
About OTA (especially 2 step), would it make sense to make a new minimal OTA uploader binary? Probbaly with the new core it also gets smaller?

Also would it make sense to make a ArduinoOTA-Only uploader? Could that also save some space so bigger binaries are possible?

TD-er, 2019-11-08:
Nope, the OTA image is using core 2.3.x or maybe even older.
I have not found any image that's smaller.

At the esp8266/Arduino repo there is some debate about whether it is possible to flash a compressed image for OTA and let the ESP decompress it during the upgrade.
This would be the same compression as used for flashing via serial using esptool.
On a typical ESPeasy build that would save about 30% of the size.

clumsy-stefan, 2019-11-08:
But instead of compiling a webserver in (in the minimal image) just using the minimal Arduino OTA could be smaller?

TD-er, 2019-11-08:
I don't know what is taking more space, a web client (pull from an URL), or web server or OTA service.

clumsy-stefan, 2019-11-08:
the basic OTA example is about 30k less than the ESPEasyUploaderMega, but obviously it does not use the ESPEasy storage for SSID etc... it's compiled in static...

s0170071, 2019-11-09:
@TD-er a bit off-topic but do you happen to know how I do suspend any wifi activity for 50ms without interrupting the connection ? 
I have wifi interference with a humidity sensor and do not want to use delay()

TD-er, 2019-11-09:
There is currently not a user accessible function to disable WiFi activity.
You could set the node to "eco mode", but that's just increasing the chance there is no WiFi activity during your measurement. It is no guarantee.

But I would rather look into the cause of this interference, which cannot be WiFi related, but must be power related.
Maybe you can add a 100 nF capacitor **as close as possible** to the humidity sensor's power line? (between Vcc and GND)

s0170071, 2019-11-09:
I then just stall the code for with delayMicroseconds(). I hoped for a non blocking solution.
Never mind about the cause. It is a known effect with capacitive humidity sensors. They use a high impedance oscillator with the capacitance being humidity sensitive. It is prone to interference by design.
But: If you have not already considered this, you may want to prevent wifi traffic during measurements for other humidity sensors. It stabilizes readings a lot.

TD-er, 2019-11-09:
That's good to know.
Recently we also did the other way around for the ADC plugin, to not perform ADC readings when the WiFi was attempting to make a connection.

closed 2023-09-03 by TD-er