letscontrolit/ESPEasy #2716: Amending ESPeasy INA219 plugin - Error compiling for board Generic ESP8285 Module.

Massivenrg, 2019-11-07:
Hi. Total noob here

I try to make changes in ESPeasy to get a new plugin from this post:
https://github.com/letscontrolit/ESPEasy/issues/132

I'm using ArduinoIDE - I have replaced source code in INA219 tab and try to compile - after instaling loads of missing libraries, changing ESP8266 board plugin to different versions (currently 2.4.2) I get to the point where I'm stuck. 
Please help
Error message:

> Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8285 Module, 80 MHz, Flash, ck, 26 MHz, 1M (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
In file included from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip_addr.h:43:0,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include/user_interface.h:30,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:29,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\ESPEasy_with_modif_ina219.ino:215:

C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\Networking.ino: In function 'void SSDP_send(byte)':

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:199:54: error: request for member 'addr' in 'ip', which is of non-class type 'uint32_t {aka unsigned int}'

 #define ip4_addr1(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[0])

                                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:205:38: note: in expansion of macro 'ip4_addr1'

 #define ip4_addr1_16(ipaddr) ((u16_t)ip4_addr1(ipaddr))

                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/arch/cc.h:87:24: note: in expansion of macro 'ip4_addr1_16'

 #define IP2STR(ipaddr) ip4_addr1_16(ipaddr), \

                        ^

C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\Networking.ino:531:22: note: in expansion of macro 'IP2STR'

                      IP2STR(&ip)

                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:200:54: error: request for member 'addr' in 'ip', which is of non-class type 'uint32_t {aka unsigned int}'

 #define ip4_addr2(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[1])

                                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:206:38: note: in expansion of macro 'ip4_addr2'

 #define ip4_addr2_16(ipaddr) ((u16_t)ip4_addr2(ipaddr))

                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/arch/cc.h:88:5: note: in expansion of macro 'ip4_addr2_16'

     ip4_addr2_16(ipaddr), \

     ^

C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\Networking.ino:531:22: note: in expansion of macro 'IP2STR'

                      IP2STR(&ip)

                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:201:54: error: request for member 'addr' in 'ip', which is of non-class type 'uint32_t {aka unsigned int}'

 #define ip4_addr3(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[2])

                                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:207:38: note: in expansion of macro 'ip4_addr3'

 #define ip4_addr3_16(ipaddr) ((u16_t)ip4_addr3(ipaddr))

                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/arch/cc.h:89:5: note: in expansion of macro 'ip4_addr3_16'

     ip4_addr3_16(ipaddr), \

     ^

C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\Networking.ino:531:22: note: in expansion of macro 'IP2STR'

                      IP2STR(&ip)

                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:202:54: error: request for member 'addr' in 'ip', which is of non-class type 'uint32_t {aka unsigned int}'

 #define ip4_addr4(ipaddr) (((const u8_t*)(&(ipaddr)->addr))[3])

                                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:208:38: note: in expansion of macro 'ip4_addr4'

 #define ip4_addr4_16(ipaddr) ((u16_t)ip4_addr4(ipaddr))

                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include/arch/cc.h:90:5: note: in expansion of macro 'ip4_addr4_16'

     ip4_addr4_16(ipaddr)

     ^

C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\Networking.ino:531:22: note: in expansion of macro 'IP2STR'

                      IP2STR(&ip)

                      ^

exit status 1
Error compiling for board Generic ESP8285 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

TD-er, 2019-11-07:
Why do you want to use core 2.4.2?
The code is calling members of objects which were not present on core 2.4.2.
Apparently I missed a few of these when compiling for core 2.4.2, since there was an issue yesterday for similar compile issues on ESPEasy Software Serial and core 2.4.2.

Core 2.5.2 is out there for a while and core 2.6.0 is right around the corner.

Massivenrg, 2019-11-07:
Because when I'm using core 2.5.2 Im getting error like below, where I have take a look for this error I found this thread
https://github.com/SandboxElectronics/LoRaGoDOCK-Gateway/issues/7

Erro (with 2.5.2 core included)r:

Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8285 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), ck, 26 MHz, 1M (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
In file included from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip_addr.h:43:0,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/include/ipv4_addr.h:38,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/arch/cc.h:93,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/arch.h:48,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/debug.h:40,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwipopts.h:3581,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/opt.h:51,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/init.h:40,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/IPAddress.h:27,

                 from C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,

                 from C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\ESPEasy_with_modif_ina219.ino:215:

C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\Networking.ino: In function 'void SSDP_send(byte)':

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:176:67: error: request for member 'addr' in 'ip', which is of non-class type 'uint32_t {aka unsigned int}'

 #define ip4_addr_get_byte(ipaddr, idx) (((const u8_t*)(&(ipaddr)->addr))[idx])

                                                                   ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:177:27: note: in expansion of macro 'ip4_addr_get_byte'

 #define ip4_addr1(ipaddr) ip4_addr_get_byte(ipaddr, 0)

                           ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:190:38: note: in expansion of macro 'ip4_addr1'

 #define ip4_addr1_16(ipaddr) ((u16_t)ip4_addr1(ipaddr))

                                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/arch/cc.h:88:24: note: in expansion of macro 'ip4_addr1_16'

 #define IP2STR(ipaddr) ip4_addr1_16(ipaddr), \

                        ^

C:\Users\total.noob\Desktop\SmartHome\ESPEasy_R120\Source\ESPEasy_with_modif_ina219\Networking.ino:531:22: note: in expansion of macro 'IP2STR'

                      IP2STR(&ip)

                      ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:176:67: error: request for member 'addr' in 'ip', which is of non-class type 'uint32_t {aka unsigned int}'

 #define ip4_addr_get_byte(ipaddr, idx) (((const u8_t*)(&(ipaddr)->addr))[idx])

                                                                   ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:178:27: note: in expansion of macro 'ip4_addr_get_byte'

[... 141 lines cut ...]

         String & operator +=(float num) {

                  ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.h:152:18: note: String& String::operator+=(double)

         String & operator +=(double num) {

                  ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.h:156:18: note: String& String::operator+=(const __FlashStringHelper*) <near match>

         String & operator += (const __FlashStringHelper *str){

                  ^

C:\Users\total.noob\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/WString.h:156:18: note:   no known conversion for argument 1 from 'IPAddress' to 'const __FlashStringHelper*'

exit status 1
ambiguous overload for 'operator+=' (operand types are 'String' and 'IPAddress')

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

TD-er, 2019-11-07:
Ah, from the start-post I now see you are using the sources from R120?
At least the path suggests it.
That's an old code base.
Please use a more recent one from the mega branch of our Github repo.

Massivenrg, 2019-11-07:
Yes, I have taken R120. But now, after your advice, I tried with mega. 
But it is still showing errors with missing libraries, then I installing missing one but keep showing still new and so on. I'm frustrated now because I get to one that I can't find anywhere. 
"OLED_SSD1306_SH1106_images.h"
Is there no zip file with whole libraries needed for ESPeasy?
Many thanks

TD-er, 2019-11-07:
There is an entire directory with all the libraries we use, it is in the libs directory.
Just copy the content of that directory into your libraries dir of ArduinoIDE, or maybe switch to PlatformIO + VS-code.

Massivenrg, 2019-11-07:
Could you point, please where to look for this library directory?

TD-er, 2019-11-07:
https://github.com/letscontrolit/ESPEasy/tree/mega/lib

It is a directory called `lib` in the ESPEasy repo.

uzi18, 2019-11-07:
we already have ina219 support in mega builds

Massivenrg, 2019-11-07:
TD-er. Many thanks for that. Finally I have been able to compile sketch with success! 
uzi18 - jest I know, but this plugin is amended and is showing three separate values and have shunt resistor factor which is most important for me as I want to measure higher current than genuine ina219

But this not the end of my problems. I'm facing witch problem after fleshing the ESP8266.
I have exported .bin file in AdruinoIDE and then load this firmware to the board using the same setting as always. The firmware upload is finished with success, but no AP is appearing on my WIFI network. 
What Im doing wrong?

TD-er, 2019-11-07:
Try first with a nightly build, so you can make sure it creates a filesystem and default settings.

Massivenrg, 2019-11-07:
I will probably get a lynch for this, please forgive but I'm total newbie in this. I have been using espeasy, tasmota and others but in ready not customized .bin files. I'm good in electronics and wiring but the biggest issue with me, that, unfortunately, I can't program and I suffer from this. 
Could you explain in simplest way what you mean by ' nightly build'
I have already loaded to the same wemos board genuine ESPeasy mega 1M with success it was working perfectly.

TD-er, 2019-11-07:
Well, lynching users is -sadly- no longer allowed by Github.
It gives them too much paper-work, and some other administrative work to delete their accounts and such.

So let's just help you out here ;)

This is what I mean by the "nightly builds": https://github.com/letscontrolit/ESPEasy/releases

Massivenrg, 2019-11-11:
Thanks again. I have made some uploads of different versions and all seems to work fine. 
When trying to upload my modified .bin file looks like I'm doing something wrong as after flashing there is no communication through com port (when flesher trying set wifi ssid and password)

When compiling - getting back warnings:
WARNING: Category 'LED' in library CHT16K33 LED Matrix Library is not valid. Setting to 'Uncategorized'
WARNING: Category 'Input' in library MechInputs is not valid. Setting to 'Uncategorized'
WARNING: Category 'Input' in library SerialSensors is not valid. Setting to 'Uncategorized'

Not sure if this is important. 

What I'm doing wrong?? :)

@uni18 - maybe you could contact me / help on priv (WhatsApp +447946649268) bylbym wdzieczny bo widze ze jestes dosc gleboko w temacie :) pozdro

uzi18, 2019-11-11:
you can find me on gitter.im, messanger or hangouts

tonhuisman, 2023-09-03:
Compiling the ESPEasy code using Arduino IDE is not endorsed, better use VSCode/PlatformIO as [documented](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html).
This issue can be closed.

closed 2023-09-03 by TD-er