letscontrolit/ESPEasy #2717: P048 plugin status

TungstenE2, 2019-11-07:
Hi all,

can you please move P048 plugin from Status 'Testing' to 'Normal' in order to make it availabe for more users?

https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P048_Motorshield_v2.ino

https://www.letscontrolit.com/wiki/index.php/Official_plugin_list

thx

TD-er, 2019-11-07:
We must make sure to make the builds smaller, or else the `normal` builds will no longer fit, at least on 1M nodes.

TungstenE2, 2019-11-07:
so you suffer your own success... ;-)

Kick other plugins out, or have a subset of plugins for 1M nodes and a full set for 4M nodes. 
I assume 1M nodes will not be controlling motor shields, right?
But to leave them in testing is not the right way.

TD-er, 2019-11-07:
@TungstenE2 See https://github.com/letscontrolit/ESPEasy/issues/2590
Remarks like those you just made would be really useful there :)

tonhuisman, 2022-04-18:
Because of the changed build configurations, this issue can a) be solved by creating a Custom build, or b) use a test build configuration that includes the desired plugin(s). Issue can probably be closed.

closed 2022-04-18 by TD-er