letscontrolit/ESPEasy #2719: openhab MQTT issues

jimmys01, 2019-11-08:
After updating to the latest I have an issue.
Openhab MQTT does not work on some nodes, and on others works but not completly

Example: I have the rule:

```
On Rules#Timer=1 do
Publish ESP_Easy/%sysname%/tele/STATE,{"Time":"%lcltime%","Uptime":"%uptime%","Vcc":%vcc%,"IP":"%ip%","Wifi":{"AP":1,"SSId":"%ssid%","RSSI":%rssi%,"APMac":"%bssid%"}}
timerSet,1,600
endon
```

And it was puplishing fine the following
`{"Time":"1970-00-00 00:00:00","Uptime":"0","IP":"172.17.1.210","Wifi":{"AP":1,"SSId":"Resort office","RSSI":-33,"APMac":"64:D1:54:A9:05:AA"}}`

Now it puplishes `{"Time":"1970-00-00 00:00:00"` (notice that it cuts it before the ',', perhaps this is related)

For devices puplishes nothing.

TD-er, 2019-11-08:
Try to wrap the arguments in quotes.
If you need the " in JSON, you can use the ' to wrap the argument.

For example:

```
On Rules#Timer=1 do
Publish ESP_Easy/%sysname%/tele/STATE,'{"Time":"%lcltime%","Uptime":"%uptime%","Vcc":%vcc%,"IP":"%ip%","Wifi":{"AP":1,"SSId":"%ssid%","RSSI":%rssi%,"APMac":"%bssid%"}}'
timerSet,1,600
endon
```

jimmys01, 2019-11-09:
I will resort in using my patch for my application since I don't want to edit the rules to a lot of nodes

closed 2019-11-09 by jimmys01

TD-er, 2019-11-09:
How many is "a lot"?
Are the rules the same for all nodes?

jimmys01, 2019-11-09:
A lot is 200+ nodes. Yes the rules are the same in 99% of them

TD-er, 2019-11-09:
So if I could make a command to fetch a file from an URL, to the flash, then you would be able to update all rules?

jimmys01, 2019-11-09:
That's OK, I will get around on doing it sometime

TD-er, 2019-11-10:
Is it also a good "temp solution" to have a fallback option (checkbox) to parse to the end of the line for the commands
- LogEntry
- publish

This still needs an update of the settings, and may result in other issues later on when we may add extra parameters to these commands.