letscontrolit/ESPEasy #2724: Updated Rules parameter parsing (2019/11)

TD-er, 2019-11-09:
This will be a sticky issue to explain once and for all how parameters are parsed in rules and how they should be parsed.

A parameter separator is typically a comma (`,`), but for convenience the separator can also be accompanied by a space or be a single space.

So these can be used to separate a parameter:
- comma
- comma space
- space comma
- space
- space space

This does impose some restrictions on the use of a comma or a space in a parameter.
Especially sending JSON to a MQTT controller can become next to impossible with these limitations.

In order to allow the comma or space in a parameter, you can wrap the parameter in quotes.
- `'`
- '"'

There are multiple quotes available for this, to be able to use "the other quote" in your parameter.
For example in JSON, you need the double quote for string like values or keys.
```
Publish domoticz/in,{"idx":26,"nvalue":0,"svalue":"[AQ#TVOC]"}
```

This can be fixed by wrapping the last parameter with single quotes like this:
```
Publish domoticz/in,'{"idx":26,"nvalue":0,"svalue":"[AQ#TVOC]"}'
```

Just make sure to use the same quote as start and end of your parameter and don't use that character in the parameter itself.
N.B. these extra quotes are removed from the parameter when used, as well as trailing or leading spaces.

The reason this behavior was changed from before 2019/11 was that the old implementation could lead to unpredictable results.
Also there was no really strict definition for what a parameter would look like.
Meaning a parameter for one command would be parsed different from another command.

Now we have a single method for parsing a parameter, but on the downside some rules using the `Publish` command may no longer work when sending JSON.
These can be fixed very easily by adding the single quotes wrapping the parameter.

This is a breaking change, but it had to be like this or else we would have to deal with lots and lots of very odd reported issues where seemingly similar rules do work on one node and not on the other.
The old behavior was not only buggy (e.g. the last quote could still be present in the text), but also depending on the order of variables (wrapped in `[` & `]`).
Also it was impossible to add more parameters to existing commands which did parse a parameter to the end of a line.

Misiu, 2019-11-15:
@TD-er does this fixes the `TaskValueSet` [issue](https://github.com/letscontrolit/ESPEasy/issues/1676)? I'm still very new to ESPEasy and rules (btw they are AWESOME!!!)

TD-er, 2019-11-15:
Nope, that feature request is not yet implemented.
This issue I added mainly to have some reference to link when people ran into trouble because of the recent changes regarding more strict parsing of parameters.

Domosapiens, 2019-11-28:
Using release mega-20191116
Seen in log: Too many arguments for cmd=SendTo

Seen this:
https://github.com/letscontrolit/ESPEasy/issues/2749#issuecomment-554771702
And documentation:
[https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#some-working-examples](url)
Section: SendTo and Publish
Needs an update ??
> SendTo: SendTo <unit>,<command>

Imagine you have two ESP Easy modules, ESP#1 and ESP#2 In the Rules section of ESP#1 you have this:

on demoEvent do
  sendTo,2,event,startwatering //(to use the previous example.)
endon
?? Must be **sendTo,2,"event,startwatering" **

And ESP#2 has the rules according to the previous example (givemesomewater)

If you then enter this with the correct IP address in the URL of your browser:

http://<ESP#1-ip >/control?cmd=event,demoEvent
?? Must be **http://<ESP#1-ip >/control?cmd="event,demoEvent" **

Then ESP#1 shall send the event ‘startwatering ‘ to ESP#2.

It is also possible to directly order GPIO changes, like:

on demoEvent do
  sendTo,2,GPIO,2,1
endon
?? Must be **sendTo,2,"GPIO,2,1"**

TD-er, 2019-11-28:
I updated your comment to use the correct Markdown syntax, as it is almost completely a form of documentation and not really a question, right?

The thing with commands embedded in the GET url is that you probably have to escape some of the characters, so it is best to cut/paste your command in the command dialog (tools page) first and then see what the URL has become.

Domosapiens, 2019-11-28:
@TD-er  ...yes I found out some aspects, but was asking for confirmation.

Am I the only one ??
-testing a version after 12 Nov
-running into trouble,
-looking in the error log
-seen something on github
I think this breaking change needs attention on the forum.

But how about this one?

> On BoilerStat#Boiler_Stat do
SendToHTTP 192.168.1.81,80,/control?cmd=taskvalueset,10,4,[BoilerStat#Boiler_Stat]
EndOn

Error Log:

> Too many arguments: cmd=SendToHTTP arg1=192.168.1.81 arg2=80 arg3=/control?cmd=taskvalueset extraArg4=10 lineLength=82	
> _SendToHTTP 192.168.1.81,80,/control?cmd=taskvalueset,10,4,0

Where to escape ?

SendToHTTP 192.168.1.81,80,"/control?cmd=taskvalueset,10,4,[BoilerStat#Boiler_Stat]"
Is that correct ??
or
SendToHTTP 192.168.1.81,80,/control?cmd="taskvalueset,10,4,[BoilerStat#Boiler_Stat]"
??

Both are Ok in the command dialog (thanks for the hint!)
(sorry, running production machine and therefore only 1 test/day possible ...)

TD-er, 2019-11-28:
> `SendToHTTP 192.168.1.81,80,/control?cmd=taskvalueset,10,4,[BoilerStat#Boiler_Stat]`

`SendToHTTP` has 3 parameters, so you should wrap a parameter in quotes when using space or comma in the parameter (or when in doubt :) )

The command + parameters are:
```
SendToHTTP,<IP address>,<Portnumber>,<url>
```
So the `<url>` should be wrapped in quotes like this.
```
SendToHTTP 192.168.1.81,80,"/control?cmd=taskvalueset,10,4,[BoilerStat#Boiler_Stat]"
```

As a reference, see the [command reference page](https://espeasy.readthedocs.io/en/latest/Reference/Command.html).
Or all commands there is a short description in the form I just gave you, with the , (comma) separator for the parameters.
When a parameter needs a comma or a space, then that one should be wrapped in quotes.
If you need double quote in the parameter (for example in JSON text), then use single quote or back quote to wrap the parameter. (just use the same for start/end of the parameter)

So these are all correct:
```
SendToHTTP 192.168.1.81,80,"/control?cmd=taskvalueset,10,4,[BoilerStat#Boiler_Stat]"
SendToHTTP 192.168.1.81,80,'/control?cmd=taskvalueset,10,4,[BoilerStat#Boiler_Stat]'
SendToHTTP 192.168.1.81,80,`/control?cmd=taskvalueset,10,4,[BoilerStat#Boiler_Stat]`
```

TD-er, 2019-12-04:
I just added a  flag in the Advanced settings to allow the rules parser to be a bit more tolerant in parsing the last parameter of some commands like Publish and sendToHttp.

Also the log is a bit more elaborate now in helping to fix the issue at hand.

With the tolerant flag enabled:
[image]

The default setting, with the flag disabled:
[image]

The rules being executed here:
```
 logentry,This is a test
 logentry,"This is a test"
```

jimmys01, 2020-01-03:
`IRSENDAC,'{"protocol":"GREE","power":"Off","mode":"Cool","temp":15,"swingv":"Off","swingh":"Off","light":"Off"}'` is not being parsed anymore

Without the ' it works fine again
IRSENDAC,{"protocol":"GREE","power":"Off","mode":"Cool","temp":15,"swingv":"Off","swingh":"Off","light":"Off"}

TD-er, 2020-01-03:
What build was working and what build isn't working anymore?

jimmys01, 2020-01-03:
That I do not know since I just updated everything and not I do not remember what it was on there.

TD-er, 2020-01-03:
Is it the last code we have on Github, or is it a nightly build?
Can you check if there is a flag set to be less strict on the rules parsing? (Tools => Advanced)

jimmys01, 2020-01-03:
It is the latest git hub code. The issue is present with the flag enabled and disabled

grzegorz189, 2020-04-11:
Hello, why can't you open rules set 2

TD-er, 2020-04-12:
@jimmys01 I noticed your last post again and was wondering if there is still an issue I need to look into.

Prutsium, 2020-05-28:
I am struggling to get the following rule to the right "New" format.
Could use a bit of help :)

On Living_Room_Temp#Temp_1 do
SendTo,20,TaskValueSet 5,2,[Living_Room_Temp#Temp_1]
endon

As it's not sending the value anymore and the error lead me to here :)

uzi18, 2020-05-28:
SendTo,20,"TaskValueSet 5,2,[Living_Room_Temp#Temp_1]"

Prutsium, 2020-05-28:
> 
> 
> `SendTo,20,"TaskValueSet 5,2,[Living_Room_Temp#Temp_1]"`

Thank you so much! Works again!

uzi18, 2020-05-28:
sendto needs 2 arguments, so second one you need to quote

jimmys01, 2020-11-30:
> 
> 
> @jimmys01 I noticed your last post again and was wondering if there is still an issue I need to look into.

IRSENDAC,**'**{"Protocol":"COOLIX","Power":true,"Opmode":"auto","Degrees":24,"Fanspeed":"auto","Swingv":"min","light":"true"}**'** does not work
IRSENDAC,**"**{"Protocol":"COOLIX","Power":true,"Opmode":"auto","Degrees":24,"Fanspeed":"auto","Swingv":"min","light":"true"}**"** does not work
IRSENDAC,{"Protocol":"COOLIX","Power":true,"Opmode":"auto","Degrees":24,"Fanspeed":"auto","Swingv":"min","light":"true"} works.

Tested with the latest commits

TD-er, 2020-11-30:
> `IRSENDAC,"{"Protocol":"COOLIX","Power":true,"Opmode":"auto","Degrees":24,"Fanspeed":"auto","Swingv":"min","light":"true"}"` does not work

This one is no surprise, as it cannot see any difference between end of parameter, or a quote in the parameter.

> `IRSENDAC,'{"Protocol":"COOLIX","Power":true,"Opmode":"auto","Degrees":24,"Fanspeed":"auto","Swingv":"min","light":"true"}'` does not work

This one is surprising.
That could only happen if the command handling does not use the ESPEasy command parsing.
Will have a look at it.

> `IRSENDAC,{"Protocol":"COOLIX","Power":true,"Opmode":"auto","Degrees":24,"Fanspeed":"auto","Swingv":"min","light":"true"}` works.

OK, good to know something is still working :)

jimmys01, 2020-12-03:
just to be sure the, command 
`IRSEND,RAW2,something,38,something` 
must be transformed to 
`IRSEND,'RAW2','something','38','something'` ?

AloisKlingler, 2020-12-16:
just as a hint: the documentation regarding this is deceptive.
https://espeasy.readthedocs.io/en/latest/Rules/Rules.html and also https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#SendTo_and_Publish need to be adapted to either have the necessary quotes at the examples or at least have a hint for the "Tolerant last Parameter" Flag. :-)

Best regards
Alois

budenn777, 2023-05-17:
Faced with the problem of saving the rules! Save button doesn't always work
Especially if multiple devices are added. In a home network, it does not work on a PC, but it works on a smartphone!

budenn777, 2023-05-18:
On System#Boot do
play,0016
endon

On KEY1#State=1 do

LCD,4,1,KEY1_PRESSED
play,0022
if [MEM#Dummy1]<4
TaskValueSet 10,1,[MEM#Dummy1]+1
endif
if [MEM#Dummy1]=0
LCD,2,1,Lampa_CVETY_spalni
endif
if [MEM#Dummy1]=1
LCD,2,1,Lampa_dvor_1
endif
if [MEM#Dummy1]=2
LCD,2,1,Ventilator_spalni
endif
if [MEM#Dummy1]=3
LCD,2,1,Ventilator_podval
endif
if [MEM#Dummy1]=4
LCD,2,1,Lampa_zady
endif

endon

On KEY2#State=1 do

LCD,4,1,KEY2_PRESSED
play,0021
if [MEM#Dummy1]>0
TaskValueSet 10,1,[MEM#Dummy1]-1
endif
if [MEM#Dummy1]=0
LCD,2,1,Lampa_CVETY_spalni
endif
if [MEM#Dummy1]=1
LCD,2,1,Lampa_dvor_1
endif
if [MEM#Dummy1]=2
LCD,2,1,Ventilator_spalni
endif
if [MEM#Dummy1]=3
LCD,2,1,Ventilator_podval
endif
if [MEM#Dummy1]=4
LCD,2,1,Lampa_zady
endif

endon

On KEY3#State=1 do
play,0020
if [MEM#Dummy1]=0
SendToHTTP 192.168.1.50,80,/relay1=1
endif
if [MEM#Dummy1]=1
SendToHTTP 192.168.1.114,80,/relay2=1
endif
if [MEM#Dummy1]=2
SendToHTTP 192.168.1.6,80,/switch?id=2^&on=true
SendToHTTP 192.168.1.6,80,/switch?id=3^&on=false
endif
if [MEM#Dummy1]=3
SendToHTTP 192.168.1.100,80,/switch?id=1^&on=true
endif
if [MEM#Dummy1]=4
SendToHTTP 192.168.1.104,80,/switch?id=0^&on=true
endif
endon

On KEY4#State=1 do
play,0019
if [MEM#Dummy1]=0
SendToHTTP 192.168.1.50,80,/relay1=0
endif
if [MEM#Dummy1]=1
SendToHTTP 192.168.1.114,80,/relay2=0
endif
if [MEM#Dummy1]=2
SendToHTTP 192.168.1.6,80,/switch?id=2^&on=false
SendToHTTP 192.168.1.6,80,/switch?id=3^&on=false
endif
if [MEM#Dummy1]=3
SendToHTTP 192.168.1.100,80,/switch?id=1^&on=false
endif
if [MEM#Dummy1]=4
SendToHTTP 192.168.1.104,80,/switch?id=0^&on=false
endif
endon

budenn777, 2023-05-18:
ESP_Easy_mega_20230508_collection_F_ESP32_4M316k
I use Win7(x32) + Google Chrome

budenn777, 2023-05-18:
[image]

tonhuisman, 2023-05-18:
It's quite some time ago that I worked with Win7, not sure if Chrome still (fully) supports that OS, supplying updates etc. (IOW: You'd better not use Win 7 anymore).
It might be some weird interaction there. Can you try using Internet Explorer, to see if that does save the rules more reliable?

budenn777, 2023-05-18:
Yes, I also assumed that win7 was outdated and checked on a new PC + win10, the problem did not go away...

tonhuisman, 2023-05-18:
Well, if it works from a mobile phone, it isn't really an ESPEasy problem, I assume, it must be something on the PC-side, maybe your Antivirus software is blocking something because the ESP is accessed via HTTP? (Some AV software is really annoying, and they _never_ give a clue in what is being blocked...)

TD-er, 2023-05-18:
> Ok, I'll look... :-) While temporarily solved the problem of loading through Tools + File browser

That's exactly what the JavaScript does when updating the rules.
It generates a HTTP POST to upload the text from the edit window and sends it to the upload URL of ESPEasy.

So if your browser blocks Javascript, or is unable to load the JavaScript (those files are loaded from a CDN server), then the save will not work.

See the documentation: https://espeasy.readthedocs.io/en/latest/Reference/ExternalHostedStaticFiles.html?highlight=CDN#external-hosted-static-files
You might want to save the rules_save.js to the file system of ESPEasy if your browser can't download this file from the CDN servers.

budenn777, 2023-05-18:
Thank you! I will try

budenn777, 2023-05-19:
Uploaded the files from the link https://github.com/letscontrolit/ESPEasy/tree/mega/static
But the button for loading rules didn't work. I opened the page code, there are errors, but I don't understand them. Here are three screenshots...

budenn777, 2023-05-19:
[image]

budenn777, 2023-05-19:
[image]

closed 2023-05-19 by TD-er

budenn777, 2023-05-19:
Yes, sorry for being off topic.