letscontrolit/ESPEasy #2726: Running task with Dummy Sensor does not generates events

mackowiakp, 2019-11-10:
mega-20191108

When runing task defined as Dummy sensor, no events are generated. Formerly - for sure up to release mega-20190928 - calling task with such sensor from rule, generated events, returning all values written in to such sensor. As I look to source, there is "event" statement in PLUGIN_READ section. But it reads and generate events only when sensor is read by controller interval setup feature.
Dont know is it intended or issue?

TD-er, 2019-11-10:
> Dont know is it intended or issue?

To answer as a real programmer: "Yes" ;)
It is partly intended, but it may be with the wrong assumptions.
I consider it a bug.
If the `TaskRun` command is given, it should generate an event as it is something you deliberately will do.
I don't think the periodical read of the dummy plugin is useful. Is that useful? Should that also generate an event?

mackowiakp, 2019-11-10:
>  Should that also generate an event?

I think - Yes. I use it instead periodical reading/generating events via "Interval" defined in controller setup. Jut need it on-demand.
And there is another, general problem. It is not possible to set "0" as "Interval" for sensor to disable periodical reading. So to practically disable such kind of reading it is necessary to set it to 65535.

TD-er, 2019-11-10:
> And there is another, general problem. It is not possible to set "0" as "Interval" for sensor to disable periodical reading. So to practically disable such kind of reading it is necessary to set it to 65535.

See also:
- https://github.com/letscontrolit/ESPEasy/issues/791
- https://github.com/letscontrolit/ESPEasy/issues/1202

Budman1758, 2019-11-10:
> I don't think the periodical read of the dummy plugin is useful. Is that useful? 

I personally dont think it"s useful. Just one reporters opinion.  :)

mackowiakp, 2019-11-11:
Aftre bit of investigation, I think that the problem is more common than discussion about "is periodical read of the dummy plugin is useful or not".
For example look at screenshots below. If "TaskRun,<number>"  in use in rule, it always returns "failed", does not matter the <number> is use.
I just finished test version of FM tuner driver (will publish as playground) and TaskRun feature is essential for use in rules. As I wrote, it worked properly up to ver mega-20190928.

[image]

[image]

TD-er, 2019-11-11:
> Thats my opinion. But I am very old....

I think the word you're looking for is "experienced" :)

uzi18, 2019-11-11:
@mackowiakp and how did you want to use taskrun with this plugin?

mackowiakp, 2019-11-11:
If You mean about Radi-FM plugin, TaskRun returns RDS info to logs (if available). At this moment I have no conception how to pass such long string of characters to any of available display drivers in universal form. But for sure running this task returns actual frequency and volume as event to display it  even on 7 segment display, on demand from rule.

TD-er, 2019-11-11:
I found the bug in the parsing of task and var nr.
For some commands you only need to supply the task nr and not the var nr. So having a second value of 0 is perfectly fine, but was rejected.

mackowiakp, 2019-11-11:
"No bugs, no software. And no funny...."

PR #2725 (2019-11-12): Docs/rules parameter parsing
Fixes: #2726
Fixes: #2710
closed 2019-11-12 by TD-er