letscontrolit/ESPEasy #2727: Latest version on ESP8266 is unable to connect to the MQTT broker with user pass authentication

hamed-ta, 2019-11-10:
as it can be seen in the pictures the latest version of ESPEasy is unable to connect to the broker with username and password athentication enabled but with the same version and same settings on ESP32 it works well.
[image]
[image]

TD-er, 2019-11-10:
I know you also needed to have the hostname defined a certain way.
Is that setup correct on the ESP8266 too?

hamed-ta, 2019-11-10:
> Is that setup correct on the ESP8266 too? yes

ghtester, 2019-11-10:
I am also using Username and Password authentication to  MQTT broker (despite it's sent as plain text) but I did not have any related issue even with latest ESP_Easy version. Perhaps it could be an issue at broker's side, especially if you are using the integrated openHAB's MQTT broker Moquette - it's unstable and AFAIK currently unusable (tested latest available from openHAB 2.5 and had to stop using it). If that's your case, try another MQTT broker if possible, I can recommend mosquitto from Centos 7 which works fine.

hamed-ta, 2019-11-11:
> I am also using Username and Password authentication to MQTT broker (despite it's sent as plain text) but I did not have any authentication related issue even with the latest ESP_Easy version.
> Perhaps it could be an issue at broker's side, especially if you are using the integrated openHAB's MQTT broker Moquette - it's unstable and AFAIK currently unusable (tested latest available from openHAB 2.5 and had to stop using it). If that's your case, try another MQTT broker if possible, I can recommend mosquitto from Centos 7 which works fine.

I’m using VerneMQ as the broker and it is using ClientId as a credential as well as User and Pass so the things can be a little different also as I said I am using the same setting with the same broker in ESP32 successfully so I don’t know who the things is different in ESP8266!

ghtester, 2019-11-11:
OK, I don't know the VerneMQ broker but with Moquette I have experienced many strange things so that's why I "ignored" your statement about ESP32 working with the same configuration. ;-) If the VerneMQ's log does not tell you more, try to reboot both ESP32 and ESP8266, try to reboot VerneMQ service and if it's still the same, try to subscribe the same topics as ESP8266 and ESP32 with a test utility (like as mosquitto_sub) to see what's the difference.

closed 2023-09-03 by TD-er