letscontrolit/ESPEasy #2729: Rules not working

fly74, 2019-11-10:
Hi folks,

Rules not working in ESP_Easy_mega-20191108_minimal_core_242_ESP8266_1M_OTA.bin

after save it looks like an UTF8 binary stream:

[image]

TD-er, 2019-11-10:
What flash brand does the unit have?

fly74, 2019-11-10:
Gosund ver. 1.1

TD-er, 2019-11-10:
Is that a brand of flash chips?
You can see the flash info on the Tools => info page.

fly74, 2019-11-10:
ESP Board
--
ESP Chip ID: | 10801447 (0xA4D127)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP01_1M
Storage
Flash Chip ID: | Vendor: 0x51 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 589 kB (284 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | true
OTA 2-step Needed: | true
SPIFFS Size: | 110 kB (28 kB free)
Page size: | 256
Block size: | 4096
Number of blocks: | 27
Maximum open files: | 5
Maximum path length: | 32

fly74, 2019-11-10:
[Amazon](https://www.amazon.de/Gosund-erforderlich-Stromverbrauch-Fernsteurung-Kompatibel/dp/B07PRDZ7F3)

fly74, 2019-11-10:
[Circuit](https://forum.iobroker.net/assets/uploads/files/1573049936752-img_9596.jpg)

TD-er, 2019-11-10:
Was it working before on this node?

fly74, 2019-11-11:
It's brand new and the first release I've tested. Everything works fine:

-toggle the relay GPIO 15 via HTTP
-toggle the 2 LEDs GPIO 0/2 via HTTP
-see the Button-switch state change in devices GPIO 13

So I thought, I can use a rule like in SONOFF to toggle the relay from the Bottun-Switch.

TD-er, 2019-11-11:
Well, it should work, but I wonder why it gets mangled like that.
There are some flash chips which do not behave like other flash chips do.
The most notorious ones are Puya chips, but maybe there are others too.

Can you try another browser and cut/paste some rules text into the edit window?  (just about any text will do)
How big was your rules text? In the past we had some issues saving larger rules but at least the text was not mangled but just cut short.

fly74, 2019-11-11:
Only a short rule switch to relay. Chrome Firefox edge edge beta. All the same.

gavtas, 2019-11-12:
The same situation with ?Mirabella? WiFi light bulbs. Details below.
Workaround: Edit rules on a laptop and upload using ESPEasy Tools->File browser.
[image]

TD-er, 2019-11-12:
> Workaround: Edit rules on a laptop and upload using ESPEasy Tools->File browser.

Thanks for this hint.
I did change something regarding the rules saving routine, but the strange thing is that it does work here on my nodes.

Can you try a build of +/- 2 months ago to see if it then works on these nodes?
Also what core version do you use? I know the last few builds only have the core 242 builds for minimal OTA included, but maybe you managed to have a smaller build yourself?

gavtas, 2019-11-12:
I am using core 242 from last build. 

Firmware
--
Build:⋄ | 20104 - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20191108
Plugins:⋄ | 3 [Normal] [Minimal 1M OTA]
Build Md5: | fa43e57c1ae24120906fee6176d46f9c
Md5 check: | passed.
Build Time:⋄ | Nov 8 2019 03:22:54
Binary Filename:⋄ | ESP_Easy_mega-20191108_minimal_core_242_ESP8266_1M_OTA.bin

fly74, 2019-11-12:
File upload don't work for me. it only creates a new file with the fullpath name as name and coexists with rules1.txt without replacing.

TD-er, 2019-11-12:
Huh? How is that possible?
What browser do you use to create this?

fly74, 2019-11-12:
Edge - But Edge Beta is working.

TD-er, 2019-11-12:
> Edge - But Edge Beta is working.

As in, the beta version is working fine with this?
This behavior does look like a browser bug as the path should not be included in the POST, but only the filename.

fly74, 2019-11-12:
not important, edge beta becomes final in Q1/2020

TD-er, 2019-11-12:
OK, so on another browser it is working well?

fly74, 2019-11-12:
as I said edge beta with chrome core ist working well and it will be the soon the default in windows.

TD-er, 2019-11-12:
> as I said edge beta with chrome core ist working well and it will be the soon the default in windows.

Well, given this answer, I can indeed conclude that's what you may have meant.
But your sentences were really short and can have multiple interpretations.

So in short, with the current version of Edge, these kind of things happen?
No wonder nobody is using Edge.

fly74, 2019-11-12:
:) but not all windows users installing alternative browsers, but the browser war ist over when edge-chrome starts.

gavtas, 2019-11-13:
Some quick tests show rules saving correctly on the version below. I hope this helps. Thanks.

Firmware
--
Build:⋄ | 20103 - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20190805
Plugins:⋄ | 3 [Normal] [Minimal 1M OTA]
Build Time:⋄ | Aug 5 2019 02:08:02

TD-er, 2019-11-13:
Thanks, will lok into the changes I made when saving the rules.
The reason I made the changes in the first place is that in some situations the ESP was crashing when saving the rules. So I split the saving into multiple chunks to allow some short delay between chunks to prevent the watchdog to kick in.
It looked more of a problem on highly fragmented filesystems.

fly74, 2019-11-13:
Fragmentation can't be the reason, because I flash a brand new device with zeros then with the binary and first change was the rule.

fblanclaas, 2019-11-28:
hello
I have the same problem
can the problem come from SPI flash mode?
[image]
but with some board it works
do not works with this board
Build:⋄ | 20104 - Mega32
System Libraries:⋄ | ESP32 SDK v3.2.3-14-gd3e562907
Git Build:⋄ | mega-20191123
Plugins:⋄ | 75 [Normal] [TEST ESP32]
Build Time:⋄ | Nov 23 2019 03:45:09
ESP Chip ID: | (0xACC221A4AE30)
ESP Chip Frequency: | 240 MHz
ESP Board Name: | Espressif ESP32 Dev Module
Flash Chip ID: |  
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash Writes: | 1 daily / 2 boot
SPIFFS Size: | 283 kB (127 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 35

but works with this board

Build:⋄ | 20104  - Mega
System Libraries:⋄ | ESP82xx Core 2_6_1, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20191123
Plugins:⋄ | 79 [Normal] [Testing]
Build Md5: | bfc7f9b8f5382468cec5c412c57e26
Md5 check: | passed.
Build Time:⋄ | Nov 23 2019 03:46:39
Binary Filename:⋄ | ESP_Easy_mega-20191123_test_ESP8266_4M_VCC.bin
ESP Chip ID: | 2705325 (0x2947AD)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Flash Chip ID: | Vendor: 0xC8 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO

Drehwaehler, 2019-12-05:
Same here with this version, it worked before on the same hardware.

Firmware
Build:⋄ | 20104  - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20191130
Plugins:⋄ | 3 [Normal] [Minimal 1M OTA]
Build Md5: | cfcf1892f18a9eba62cac0a8206a3d14
Md5 check: | passed.
Build Time:⋄ | Nov 29 2019 23:50:39
Binary Filename:⋄ | ESP_Easy_mega-20191130_minimal_core_242_ESP8285_1M_OTA.bin
System Status
Syslog Log Level: | None
Serial Log Level: | Info
Web Log Level: | None
ESP Board
ESP Chip ID: | 9129007 (0x8B4C2F)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP8285
Storage
Flash Chip ID: | Vendor: 0x51 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 7 daily / 8 boot
Sketch Size: | 586 kB (288 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | true
OTA 2-step Needed: | true
SPIFFS Size: | 110 kB (28 kB free)
Page size: | 256
Block size: | 4096
Number of blocks: | 27
Maximum open files: | 5
Maximum path length: | 32

Drehwaehler, 2019-12-05:
If i upload the rules with the file browser, it works as expected. I have to delete the old rulesx.txt before.

TD-er, 2019-12-05:
Yesterday I had something similar.
After reducing the rules to something below 1550 bytes, it would save them.
But not always completely. Once I had only like 1220 bytes saved while the rules I tried to save were about 1500 bytes.

Drehwaehler, 2019-12-05:
My rules are quite short: 86 Bytes. And the garbling happens even when i type in 1 single Letter and save the rule. I tried both, Chrome and Firefox. On 3 different hardware platforms: Sonoff S2, Sonoff 4CH and some China 30A relay. I will also rollback  to the version mega-20190805 tomorrow and report, if that changes the problem.

TD-er, 2019-12-06:
Please also try a more recent core version (2.6.x) to see if it is related to something that has been fixed now.

TD-er, 2019-12-06:
I just tested on a node myself running the last core version available and I noticed the parameters of the form field are not received in full.
Free memory is very low on this unit and the parameter for the rules is completely lost (in some tests only containing about 600 bytes)

```
80397786 : Info  : EVENT: mem#ram=6632.00
80397900 : Info  : EVENT: mem#=1340.00
80398011 : Info  : EVENT: mem#=1340.00
80398118 : Info  : EVENT: mem#=1340.00
80400601 : Info  : SaveToFile: free stack: 3408
80400610 : Info  : FILE : Saved rules1.txt
80400612 : Info  : SaveToFile: free stack after: 3408
80400618 : Info  : Rules : Save rulesSet: 1 currentSet: 1 rules.length(): 0
80400620 : Info  :  Webserver args: 0: 'plain' length: 2343
```
The amount of data I tried to save was about 1500 bytes.
But not even the `set` and `rules` parameter were present when received by the ESP.

So I'm thinking this may also be a memory issue.
Maybe we can somehow add the expected amount of data in the rules field, so we reject saving it if the rules field is not received in full?

Also I wonder what this `plain` field is, as reported in the logs.
It looks like it is also a lot and maybe not needed?

TD-er, 2019-12-06:
I've been looking into the code parsing the arguments and made an issue for it on the esp8266/Arduino repo: https://github.com/esp8266/Arduino/issues/6884

closed 2020-03-23 by TD-er

dersparkiie, 2020-09-08:
> 
> Gosund ver. 1.1
> […](#)

Which Gosund do you use?
Gosund SP111 v1.1?

I'm still searching a working Espeasy Version for this one...

fly74, 2020-09-08:
Release mega-20200829 is working well, except this issue https://github.com/letscontrolit/ESPEasy/issues/3075

dersparkiie, 2020-09-08:
With or without Power Measurement?

fly74, 2020-09-08:
without, but I would try it with, if you guide me

PR #2921 (2020-03-23): Rules save fix
Now the rules are uploaded as a (virtual) file instead of form data. The previous way resulted in corrupted rules files if the unit didn't get the full POST body.

Should fix #2896
Should fix: #2819
Should fix: #2729
Should fix: #1296