letscontrolit/ESPEasy #2731: ESPEasy p2p protocol not working

s0170071, 2019-11-11:
my 1 year old node shows all other node on the main screen, the new nodes (1 week old) only display themselves on the main screen list of nodes.

TD-er, 2019-11-11:
Have you set the UDP port?
Has the new unit a unit nr other than 0? (and unique)
Did you restart the unit after setting the UDP port?

s0170071, 2019-11-11:
Yes, yes and yes. 
All my units show up on the **old** node so all must be sending. 
Only the very recent ones do not receive or display other nodes than themselves.

TD-er, 2019-11-11:
ECO mode is off?
Gratuitous ARP is on?
Same subnet?

uzi18, 2019-11-11:
You can check also with loglevel Debug-more if something is sent/received by node

s0170071, 2019-11-12:
just did another restart, now its working.

closed 2019-11-12 by s0170071