letscontrolit/ESPEasy #2734: Incorrect value transfer by the _C11 Generic HTTP Advanced controller

NOV62, 2019-11-12:
Hello! I installed my own EspEasy build on the ESP-1M module. The assembly was performed by editing the custom.h file and compiled in the Arduino IDE v.1.8.10 environment, kernel 2.6.0. When the module was launched, it turned out that the Generic HTTP advanced controller (_C11) was transferring data from the device incorrectly.
example:
Controller Settings:
Host: open-monitoring.online
Controller Publish: get? ____ & key = ____ & p1 =%value%

Instead of the numeric value of the parameter, the controller passes "% value%".

Here is the log:

195187: EVENT: T # Temp = 22.00
195210: HTTP before parsing:
195210: GET / get? ____ & key = ____ & p1 =%value% HTTP / 1.1 ^^ Host: open-monitoring.online ^^ User-Agent: ESP Easy / 20104 / Nov 12 2019 09: 20: 42 ^
195211: HTTP after parsing:
195211: GET / get? Cid = ____ & key = ____ & p1 =%value% HTTP / 1.1 ^^ Host: open-monitoring.online ^^ User-Agent: ESP Easy / 20104 / Nov 12 2019 09: 20: 42 ^
195213: HTTP after replacements:
195213: GET / get? Cid = ____ & key = ____ & p1 =%value% HTTP / 1.1 ^^ Host: open-monitoring.online ^^ User-Agent: ESP Easy / 20104 / Nov 12 2019 09: 20: 42 ^
195257: HTTP: C011 Success! HTTP / 1.1 200 OK ^

When using the Generic HTTP controller (_С008) everything is fine, the value is transmitted. How to fix it?
thank

TD-er, 2019-11-12:
Have you tried using `%val1%` etc?

The supported variables for these are:

- `%id%`
- `%val1%`
- `%val2%`
- `%val3%`
- `%val4%`
- `%tskname%`
- `%vname1%`
- `%vname2%`
- `%vname3%`
- `%vname4%`

NOV62, 2019-11-13:
Thank. With% val1% everything worked

closed 2019-11-13 by NOV62